OpenCPN/OpenCPN #1929: Question: wxWidgets version used plugin current builds #1876

rgleason, 2020-05-26:
We need an listing of  what wxWidget versions used with which environments and details how these environments should be built. It is not as straightforward as one might think with multiple versions of wxWidgets being used.

We also need a listing of the terms that are to be used for "ARCH"

Such that things can move forward (without going backwards), or. indeed, grinding to a halt.  Thanks very much.

bdbcat, 2020-05-27:
These are the accepted ARCH values:
arm64
armhf
i386
x86_64

All lower case, enforced at runtime filename checks.

bdbcat, 2020-05-27:
Rick...
Nothing has changed with respect to wxWidgets build versions.  The CMAKE scripts will find the correct development version, unless you have multiple versions installed, which is not a common thing.

closed 2020-05-27 by rgleason