OpenCPN/OpenCPN #1930: Chart Downloader: Large chart catalog appears to hang Windows

transmitterdan, 2020-05-27:
When a large number of charts is added to the downloader chart list then Windows itself appears to hang.  It is not hung and if one waits long enough it will recover.  This happens when several hundred charts appear in one catalog such as all US charts catalog (>1000 charts).

I believe the root cause is that the chart list panel is created from several individual items all derived from wxWindow.  Each one has a hWnd handle that Desktop Window Manager (DWM) has to handle.  There are at least 5 hWnds per chart entry so if there are 1,000 charts then there are 5,000 hWnd handles.  That number may be off a few thousand.  I think this exposes some resource limitation buried deep in WDM.  At one point Windows used to limit the number of handles to 10,000.  I don't know if that limit exists in Windows 10 still.  But I think it does for XP and Windows 7.

I have devised a possible optimization by reducing the number of unique window handles to 3,000 with some improvement but not yet happy with it.

transmitterdan, 2020-05-27:
There is also a bug in wxWidgets <3.1.3 that shows up when the list of charts exceeds about 600.  This is another reason to find a longer term solution.  In wxWidgets 3.1.3 this bug is fixed and it can handle 1,000 charts with out problems.  Then the DWM issue becomes the concern.

transmitterdan, 2020-05-27:
See optimization in my fork:  https://github.com/transmitterdan/OpenCPN/tree/chartListOpt

transmitterdan, 2020-05-27:
Link to installer based on wxWidgets 3.1.3.
https://1drv.ms/u/s!Ahh4-68oWChagb42-28NjHeYS7vSfw?e=5JTDjf

bdbcat, 2020-05-27:
Dan...
Great detective work!
IIUC, there are two separate bugs?
1. hWND limit
2. Something else in wxWidgets, fixed by 3.1.3 ( info/link?)

I do not think we should move OCPN 5.2 to wxWidgets 3.1.3 just for this bug.  There will be other bugs, we can be sure.
I think if we are creative, we can reduce the hWND to just two per chart panel.  This could be done by implementing OnPaint() for the panel.  So we have one hWND for the panel, and another for the check control.  The rest is manual paint.

Dave

transmitterdan, 2020-05-27:
Dave,

We could do that by repainting the panel as the scroll bar moves which would limit the number of hWnds.  I still don't understand why this overwhelms the window manager though.  Also, I want to do some tests on GTK3 with existing system.  I think on Windows now it isn't that bad.

I want to do some other optimizations though for the context menu selections.  There is no reason to "fill from file" just for selecting updated or new charts.  It is a simple thing to just update the check boxes without rebuilding the entire panel.

I don't think the wxWidgets bug will hurt us much.  It's been there for years and no one has noticed it.  Who in their right mind will scroll through a thousand charts hand selecting which ones to download anyway?

bdbcat, 2020-05-27:
Dan
"Who in their right mind .."  Indeed.
If you are on it, I'll wait for your PR.

Do you have a Widgets bug track link, so I can learn about the correction in 3.1.3?
Thanks a lot.
Dave

transmitterdan, 2020-05-27:
I haven't figured out what the bug is in wx 3.1.2 and below.  I just know that when I build with those versions the chart list gets wonky after 500-600 charts.  But this has always been true I think so not a big issue I suspect.  It only affects the auto painting of the panel list.  It doesn't affect the ability to update new or updated charts.

transmitterdan, 2020-05-27:
Dave,

What version of wx will 5.2 use for release on Windows?

Dan

bdbcat, 2020-05-27:
3.1.2, bundled with the app.

transmitterdan, 2020-05-27:
@bdbcat 

Dave, please test this version to see if you can live with it for this one case of unusually large chart lists.  It freezes for a reasonably short time and displays all the charts correctly.  Also, it no longer rebuilds the list when selecting various checkbox configurations (Updated, New, etc.).  So overall I think it is ok for V5.2 release.  There is some room for further optimization which I can work but wanted to see if you can live with this state for now.  This was build by the appveyor CI process in the latest git master so it uses all the library versions same as in next release.

https://dl.cloudsmith.io/public/dan-dickey/opencpn-unstable-testing/raw/files/opencpn-msvc_5.1.518+176.deb4d44_setup.exe

bdbcat, 2020-05-27:
Dan...
This will do for O5.2.
We must remember to instruct users who complain about this, to simply define smaller download packages (by states, or regions, etc.).  Indeed, who really needs all 1007 of NOAA RNC's in real life situations?
Accepting your draft PR for next Beta.

Dave

closed 2020-05-28 by transmitterdan