OpenCPN/OpenCPN #1931: PluginHandler: rfind() simple error.

leamas, 2020-05-27:
4c902d25e adds some lines like 

    if (plugin_os.rfind("ubuntu", 0) == 0){                       // starts_with()

The intent is to test if the string (here plugin_os) starts with "ubuntu". This will fail if string contains the searched string (here "ubuntu") at some other position, for example as part of the build info. In this case rfind will return a positive number > 0 even if string actually starts with "ubuntu".

Possible fixes includes using `std::string::find()` instead of `rfind()` or using `ocpn::startswith()` for readability.

bdbcat, 2020-05-27:
Suggestion accepted/implemented

leamas, 2020-05-27:
Right, in 00cc35e74. Closing

closed 2020-05-27 by leamas