OpenCPN/OpenCPN #1939: Demo plugin not compiling

campbellhennessy, 2020-06-03:
I had an issue compiling the demo plugin in Linux.  I have no issues compiling without the demo being enabled.  Please see the error message(s) below.  

The first error message, when running cmake (I am running with flags to build the demo and a flag to force GTK3):
I believe it is due to the PRIVATE keyword at plugins/demo_pi_sample/CMakeLists.txt:141.  I removed that line and the cmake error cleared.  I notice that in other plugins for the target link library to opencpn that they do not include the PRIVATE keyword.  

The second error message, when running make:
Not sure what is happening here, a problem at the linking stage it seems.

All the best!

```
-- *** Staging to build demo_pi ***
-- SKIP_VERSION_CONFIG: 
-- translate directory: /mnt/CamLinData/ECS/OpenCPN/build/plugins/demo_pi_sample/CMakeFiles
-- INCLUDE_DIRECTORIES: /mnt/CamLinData/ECS/OpenCPN/build/include;/usr/include;/usr/lib/wx/include/gtk3-unicode-3.0;/usr/include/wx-3.0;/usr/include;/usr/include;/mnt/CamLinData/ECS/OpenCPN/build/plugins/demo_pi_sample/CMakeFiles/include;/mnt/CamLinData/ECS/OpenCPN/plugins/demo_pi_sample/include;/mnt/CamLinData/ECS/OpenCPN/plugins/demo_pi_sample/src
-- Found OpenGL...
--     Lib: /usr/lib/libGL.so/usr/lib/libGLU.so
--     Include: /usr/include
-- Found OpenGL...
--     Lib: /usr/lib/libGL.so/usr/lib/libGLU.so
--     Include: /usr/include
-- Found wxWidgets: -pthread;;;-lwx_gtk3u_xrc-3.0;-lwx_gtk3u_webview-3.0;-lwx_gtk3u_html-3.0;-lwx_gtk3u_qa-3.0;-lwx_gtk3u_adv-3.0;-lwx_gtk3u_core-3.0;-lwx_baseu_xml-3.0;-lwx_baseu_net-3.0;-lwx_baseu-3.0 (found version "3.0.5") 
CMake Error at plugins/demo_pi_sample/cmake/PluginInstall.cmake:8 (TARGET_LINK_LIBRARIES):
  The keyword signature for target_link_libraries has already been used with
  the target "demo_pi".  All uses of target_link_libraries with a target must
  be either all-keyword or all-plain.

  The uses of the keyword signature are here:

   * plugins/demo_pi_sample/CMakeLists.txt:141 (target_link_libraries)

Call Stack (most recent call first):
  plugins/demo_pi_sample/CMakeLists.txt:143 (INCLUDE)

-- Starting POTFILE generation
CMake Deprecation Warning at plugins/demo_pi_sample/cmake/PluginPackage.cmake:11 (CMAKE_POLICY):
  The OLD behavior for policy CMP0002 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  plugins/demo_pi_sample/CMakeLists.txt:145 (INCLUDE)

-- *** Added sample plugin: /mnt/CamLinData/ECS/OpenCPN/plugins/demo_pi_sample
-- *** Package will NOT include tide and current data ***
-- Git status: clean
-- Configuring incomplete, errors occurred!
See also "/mnt/CamLinData/ECS/OpenCPN/build/CMakeFiles/CMakeOutput.log".
make: *** [Makefile:4942: cmake_check_build_system] Error 1

```

```
[ 97%] Linking CXX shared library libdemo_pi.so
cd /mnt/CamLinData/ECS/OpenCPN/build/plugins/demo_pi_sample && /usr/bin/cmake -E cmake_link_script CMakeFiles/demo_pi.dir/link.txt --verbose=1
/usr/bin/c++ -fPIC  -pthread -pthread -pthread -O2 -g -DNDEBUG  -Wl,-Bsymbolic -Wl,-Bsymbolic -shared -Wl,-soname,libdemo_pi.so -o libdemo_pi.so CMakeFiles/demo_pi.dir/src/demo_pi.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/latlong.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/long.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/nmea0183.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/response.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/rmb.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/sentence.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/talkerid.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/rmc.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/hexvalue.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/lat.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/expid.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/wpl.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/rte.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/hdt.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/hdg.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/hdm.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/gll.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/vtg.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/gga.cpp.o CMakeFiles/demo_pi.dir/src/nmea0183/gsv.cpp.o  -pthread -lwx_gtk3u_xrc-3.0 -lwx_gtk3u_webview-3.0 -lwx_gtk3u_html-3.0 -lwx_gtk3u_qa-3.0 -lwx_gtk3u_adv-3.0 -lwx_gtk3u_core-3.0 -lwx_baseu_xml-3.0 -lwx_baseu_net-3.0 -lwx_baseu-3.0 -lbz2 -lz ../../libs/garmin/libGARMINHOST.a ../../libs/nmea0183/libNMEA0183.a ../../libs/easywsclient/libEASYWSCLIENT.a ../../libs/wxservdisc/libWXSERVDISC.a 
make[2]: Entering directory '/mnt/CamLinData/ECS/OpenCPN/build'
make[2]: Nothing to be done for 'plugins/grib_pi/CMakeFiles/grib_pi.dir/build'.
make[2]: Leaving directory '/mnt/CamLinData/ECS/OpenCPN/build'
make[2]: Entering directory '/mnt/CamLinData/ECS/OpenCPN/build'
make[2]: Nothing to be done for 'plugins/dashboard_pi/CMakeFiles/dashboard_pi.dir/build'.
make[2]: Leaving directory '/mnt/CamLinData/ECS/OpenCPN/build'
[ 97%] Built target grib_pi
[ 97%] Built target dashboard_pi
/usr/bin/ld: ../../libs/nmea0183/libNMEA0183.a(GPwpl.cpp.o): relocation R_X86_64_PC32 against symbol `_ZN8wxString4nposE@@WXU_3.0' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [plugins/demo_pi_sample/CMakeFiles/demo_pi.dir/build.make:413: plugins/demo_pi_sample/libdemo_pi.so] Error 1
make[2]: Leaving directory '/mnt/CamLinData/ECS/OpenCPN/build'
make[1]: *** [CMakeFiles/Makefile2:1983: plugins/demo_pi_sample/CMakeFiles/demo_pi.dir/all] Error 2
make[1]: Leaving directory '/mnt/CamLinData/ECS/OpenCPN/build'
make: *** [Makefile:175: all] Error 2

```

campbellhennessy, 2020-06-03:
So I managed to get it to compile but I feel like there must be some way to solve it more cleanly (using shared libraries) but I am really not experienced with CMake.

Problem 1 was solved by removing the keyword PRIVATE in the target link library
Problem 2 was solved by copying  GPwpl.hpp, GPwpl.cpp, apb.hpp, apb.cpp, xte.hpp and xte.cpp from /src/nmea0183 to the plugin source directory for nmea0183 and then updating the CMakeList for the plugin to have those files as well.

I am still really confused by the duplication of the nmea source files in the plugin, I would appreciate anyone who could help me understand this.  Also very curious if there is a cleaner way to resolve this compilation problem.

leamas, 2020-06-04:
For better or worse, the official policy is to bundle dependencies in the plugins. The only exception is libraries which contains singletons which just cannot be duplicated. So copying the missing files is, for now, the way to go.

The demo plugin build failure is probably a left-over after some major updates in the cmake build system this cycle.

As for the linkage problem we should IMHO consider to find a solution which uses the new target_link_libraries() signature which includes the PRIVATE keyword to avoid leaking dependencies from the plugin into the main code. The trick is that if used, it must be used everywhere -- seems that a lot of invocations in cmake/PluginInstall.cmake needs to updated.

campbellhennessy, 2020-06-04:
Thank you for the information leamas.  Do you think it is best if I leave it as is or submit a pull request for the changes I made to get it working (as an interim measure until someone has the time to refactor the CMake to use keywords)?

leamas, 2020-06-04:
My personal opinion is that it's better to try  to make a proper patch. I might be able to do it if you don't have the time. 

In any case I'd advise against making a quick fix. It's not for me to judge though, I'm not the maintainer...

campbellhennessy, 2020-06-04:
I think I will give it a try over the next week.  If I understand correctly it would be going through the cmakes for the plugin and ensuring that all the target_link_libraries have a sane value for the interface... which I suppose as it is a plugin should all be private?

campbellhennessy, 2020-06-04:
That went faster than I anticipated.  It now compiles for me without complain in ArchLinux.  Here is a link to the compare of what I did.  As suggested it now makes all linked libraries as private to the plugin.  I'm still not super familiar with CMake so I would appreciate if someone looks at the changes (there really aren't many) before I submit a pull request.  https://github.com/OpenCPN/OpenCPN/compare/master...campbellhennessy:demo-plugin-cmake-keywords

bdbcat, 2020-06-04:
@campbellhennessy 
Have not looked at your code, yet.  But the real test is to build a plugin for Windows.  In Windows, there is no visibility to libraries as there is under linux.  Thus, the duplication of code is required.  Also, this model enforces a simpler API to OCPN core, thus further de-coupling the programming, testing, and release efforts of plugins.
I Look forward to your PR fixing the demo sample!
Good Luck
Dave

leamas, 2020-06-05:
@campbellhennessy
At a glance:  your changes looks fine.

campbellhennessy, 2020-06-08:
@bdbcat 
You are certainly right that things are more difficult in Windows :).

I am now looking for some advice.  Some of the NMEA library code that is included in the demo project calls on globals that are defined in chart1.cpp (for example g_TalkerIdText used in response.cpp).... This seems to compile fine in linux but when it is packaged in Windows it throws errors about being unable to resolve the symbols.  For the time being I am just going to comment out the references to the globals to see if the rest compiles fine.  I notice that in the dashboard plugin the response.cpp file does not contain references to g_TalkerIdText... I am not sure if this is just because the source file was copied from an earlier version of the nmea library or if those references were stripped intentionally to avoid this compile problem.

I would appreciate any thoughts :)

campbellhennessy, 2020-06-08:
> @bdbcat
> You are certainly right that things are more difficult in Windows :).
>
> I am now looking for some advice. Some of the NMEA library code that is included in the demo project calls on globals [...]

Works in windows if I remove references to g_TalkerIdText and remove all the apb files and references to them (they use global g_NMEAAPBPrecision from chart1.cpp)

rgleason, 2021-04-23:
Move to "Discussions" topic  Q&A?   or close?

closed 2021-04-23 by campbellhennessy