OpenCPN/OpenCPN #1941: Plugin Manager - rejecting valid plugins with no reasons given

jongough, 2020-06-06:
I have just installed OCPN 5.1.520+50d0e0b on Ubuntu focal (clean install), started it up and tried to install ODraw 1.8.0 through plugin manager after updating to the latest master catalog. I cannot see ODraw even though it is in the ocpn-plugins.xml with both gtk3 & without gtk3. If I download and try and install the tar.gz file through the 'Import plugin...' button I get the error shown in the dialog box below.
[image]

If I download the non-gtk3 deb file from cloudsmith and install it through the Ubuntu installation process, apt, and then run OCPN it works quite happily. The deb and tar.gz files are from the same circleci build process with only the package file being different. They are both generated in the same job and uploaded to the cloudsmith repository at the same time.

If the plugin is OK being installed from a deb file why is it being marked as incompatible by Plugin Manager? There is no indication in the log of what is wrong. I have included the log file for a run that has the deb version installed and then asked Plugin Manager to cache the tar.gz file.
[log of ubuntu focal -20.04 - plugin manager rejecting plugin.txt](https://github.com/OpenCPN/OpenCPN/files/4739347/log.of.ubuntu.focal.-20.04.-.plugin.manager.rejecting.plugin.txt)

If I try to use the master catalog, after refreshing it, the plugin does not show up and there is no information provided to the user.

bdbcat, 2020-06-06:
Jon...
Improper metadata.
The correct tags for a Ubuntu 20.04 build using x86_64 is:

```
    <target>ubuntu-gtk3-x86_64</target>
    <target-version>20.04</target-version>
    <target-arch>x86_64</target-arch>

```

jongough, 2020-06-06:
When did the meta data change? Why is the target arch now also in the target? Why is ocpn-metadata generator not picking these types of changes up? 

Each change like this, whilst easy for the person that did it, is a nightmare for those of us trying to keep up and generate all the other plugins. There was no notification this was happening even through the group email. This makes it frustrating at best!

Also, think of the users, there is no indication of what is going on, the developers have it all working and then it doesn't with no indication of what is happening. You had to point out that there is a change in this issue, isn't that a bit late?

rgleason, 2020-06-06:
I believe I am having some of these problems with weatherfax too. I tried  to import a recent weatherfax_with "amd" extension and the plugin just disappeared and Opencpn closed. So for Ubuntu bionic I want ARCH="x86-64"?   Will try that one next.

jongough, 2020-06-06:
It is a change which I am addressing at the moment in testplugin. I hope to have it going soon. 

The issues is that plugin manager keeps changing its requirements but not documenting or communicating them so it makes it very hard to make workable plugins.

jongough, 2020-06-06:
We need a definitive, managed, list of environments that are valid for use. I think I have just found another difference where amd64 and x86_64 seem to be used interchangeably, or are changing over time.

rgleason, 2020-06-06:
I am not finding in the CS repositories any weatherfax  ubuntu 18.04  with ARCh "x86-64"
so I can't test it on ubuntu I guess.

bdbcat, 2020-06-06:
jon...
In your email inbox, 22 May, 2020, you should see:

```
Hello all...
In some last minute testing, I have found that we need a few changes in
the plugin metadata contents.

1.  We need a new field , <target-arch>, to clearly specify the CPU
architecture for which a plugin has been built.

2.  For linux plugins, we need to expand the plugin naming convention to
ensure unique names across various architectures.  This will give us
package names like:

oesenc_pi-4.2.8-0_ubuntu-x86_64-16.04.tar.gz

Note the added "x86_64".

All of these changes are present in the oesenc_pi github now, as "Fontend 1"
Two files to change:
1. oesenc-plugin.xml.in     (add <target-arch> to template)
2. PluginSetup.cmake        (expand linux plugin naming conventions)

Please take a look, adapt your plugin CI workflow to conform, and submit
PRs to the master catalog when ready.

```

That mail thread continues on to discussion of x86_64 vs amd64, and other things.
Evidently some plugin devs got the word...

See also https://github.com/OpenCPN/OpenCPN/issues/1921

Dave

bdbcat, 2020-06-06:
Rick...
Please, what is the state of WeatherFax and Weather Routing?  Seems some platforms are missing?

bdbcat, 2020-06-06:
Jon...
"I think I have just found another difference where amd64 and x86_64 seem to be used interchangeably, or are changing over time."

And where would this be?

bdbcat, 2020-06-06:
jon...
"Also, think of the users, there is no indication of what is going on"
Actually, if the developers test their plugins, (as you have just done), and discover/fix this kind of big, then users will never see such failures to load.  Really, OCPN core cannot anticipate all ways in which plugin metadata may be simply wrong, and produce useful messages.  Some things just have to be right...

bdbcat, 2020-06-06:
jon...
"Why is gtk3 not an XML element like it should be? Why don't we have a list of all the dependencies in the XML like deb files do?"

We have talked about this over and over.  Pluginmanager is not a  dependency tracking app manager like found on linux as dpkg, and friends.  It never will be.  If used correctly, plugin manager works well for its intended purpose.

No further changes are being made for O5.2
Dave

jongough, 2020-06-07:
Unfortunately you have made plugin manager the default installation manager for OCPN. With this comes the need to be more cognisant of the environments and help the user if things go wrong. You are now suggesting that all plugin developers have all supported environments available for testing, and worse than that you need the developers to have multiple machines of the same type with different software types installed just to check that things work. 

Why do you thing we moved away from installing zip files to package managers? They helped with this process and notify the user of what has failed and why. None of this is done by plugin manager. I have raised this before a couple of times but have been ignored. I have suggested a small change to plugin manager to use the installation packages available rather than the zip files, but this was ignored.

So, publish the xml/xsd, in there make it so that ALL valid values for ARCH etc. are documented fully, put proper and meaningful error messages into the OCPN log, make ocpn-metadata validate properly the xml BEFORE we attempt to upload 'incorrect' plugins. I could go on, but I will leave it there.

I still don't know if I should use amd64 or x86_64 as they appear to be used interchangeably in oesenc_pi. Can you please publish the full list of acceptable values for platform specific xml tags. Looking through oesenc_pi is not easy as it is all over the place.

bdbcat, 2020-06-07:
Jon...
I will also leave it where it is, until we get some more volunteer dev resources...

Proper value is x86_64.
You may inspect oesenc_pi metadata .XML file for all possible values.
I will repeat here:
```

<?xml version="1.0" ?>
<plugins>
  <version>0.0.0</version>
  <date>2020-06-05 21:12</date>

  <plugin version="1">
    <name> oeSENC </name>
    <target>ubuntu-gtk3-x86_64</target>
    <target-version>20.04</target-version>
    <target-arch>x86_64</target-arch>
  </plugin>

  <plugin version="1">
    <name> oeSENC </name>
    <target>debian-x86_64</target>
    <target-version>10</target-version>
    <target-arch>x86_64</target-arch>
  </plugin>
  
  <plugin version="1">
    <name> oeSENC </name>
    <target>darwin</target>
    <target-version>10.13.6</target-version>
    <target-arch>x86_64</target-arch>
  </plugin>

  <plugin version="1">
    <name> oeSENC </name>
    <target>raspbian-armhf</target>
    <target-version>9.4</target-version>
    <target-arch>armhf</target-arch>
  </plugin>
  
  <plugin version="1">
    <name> oeSENC </name>
    <target>fedora-x86_64</target>
    <target-version>29</target-version>
    <target-arch>x86_64</target-arch>
  </plugin>
  
  <plugin version="1">
    <name> oeSENC </name>
    <target>msvc</target>
    <target-version>10.0.14393</target-version>
    <target-arch>x86_64</target-arch>
  </plugin>
  
  <plugin version="1">
    <name> oeSENC </name>
    <target>ubuntu-gtk3-x86_64</target>
    <target-version>18.04</target-version>
    <target-arch>x86_64</target-arch>
 </plugin>
 
  <plugin version="1">
    <name> oeSENC </name>
    <target>ubuntu-x86_64</target>
    <target-version>18.04</target-version>
    <target-arch>x86_64</target-arch>
  </plugin>
  
  <plugin version="1">
    <name> oeSENC </name>
    <target>ubuntu-arm64</target>
    <target-version>18.04</target-version>
    <target-arch>arm64</target-arch>
  </plugin>
  
  <plugin version="1">
    <name> oeSENC </name>
    <target>ubuntu-x86_64</target>
    <target-version>16.04</target-version>
    <target-arch>x86_64</target-arch>
  </plugin>
  
  <plugin version="1">
    <name> oeSENC </name>
    <target>mingw</target>
    <target-version>10</target-version>
    <target-arch>x86_64</target-arch>
  </plugin>
  
  <plugin version="1">
    <name> oeSENC </name>
    <target>flatpak</target>
    <target-version>18.08</target-version>
    <target-arch>x86_64</target-arch>
  </plugin>

  <plugin version="1">
    <name> oeSENC </name>
    <target>raspbian-armhf</target>
    <target-version>10</target-version>
    <target-arch>armhf</target-arch>
  </plugin>
  
</plugins>

```

bdbcat, 2020-06-07:
jon...
"You are now suggesting that all plugin developers have all supported environments available for testing, and worse than that you need the developers to have multiple machines of the same type with different software types installed just to check that things work.'

Not at all, although this would be useful.  You will understand how wxWidgets does not perform exactly the same across all "supported" platforms....
I am suggesting that when a problem arises, let us calmly take the Occam's razor approach, ask simple questions, learn things, and then fix the problem in the simplest way possible.  Without resetting the whole design, and re-opening the laundry list of potential deficiencies...

bdbcat, 2020-06-07:
Jon...
btw, there is a difference between "ignored" and "not adopted".  I have read every one of your posts....

jongough, 2020-06-07:
Whilst I can go through oesenc_pi ([https://github.com/OpenCPN/OpenCPN/issues/1941#issuecomment-640148826]) to find out what has been done recently it is not really the way that xml is supposed to work. There should be a definition of exactly what is being required and what works. There should also be versioning or it gets difficult to report issues. Your opencpn/plugins should probably contain all the information that a developer needs without having to look through another plugin to work out what is going on.

I am finding it very frustrating trying to get plugins to work with the new plugin manager and I have not been making changes/pull requests to try and help due to the number of changes to OCPN I did that got ignored for long enough that I had to redo them. Each change to the mainline, particularly when git is 'flattened' to hide the number of commits needed, has caused me major issues where I have to delete and re-create the local git repository and then re-apply or re-work the change to fit in, only to find that after the push another pull request was accepted that completely destroyed mine. I am also having this issue frequently with the plugins repository trying to update the xml file.

In [https://github.com/OpenCPN/OpenCPN/issues/1941#issuecomment-640149280](https://github.com/OpenCPN/OpenCPN/issues/1941#issuecomment-640149280) I have been trying to raise issues that are sensible, user centric and with all the information needed to replicate so that things can be improved. I have done quite a bit of testing on multiple platforms and have raised issues as they occurred. Unfortunately many of the issues, particularly around the user experience have yet to be addressed.

I am now in the process of releasing an updated version of OD that has the 'new' format target. I still think the way that this has been implemented is wrong and makes things more complicated, but.....

jongough, 2020-06-07:
I have just tried installing oesenc_pi both non-gtk3 & gtk3 from the tarball via "Import plugin..." and it fails with "Incompatible import plugin detected." This was `oesenc_pi-4.2.8-0_ubuntu-x86_64-20.04.tar.gz` and `oesenc_pi-4.2.8-0_ubuntu-gtk3-x86_64-20.04.tar.gz` . The version of focal is:
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04 LTS"
```
If I let it install from a refreshed Master Catalog it installs OK. when I look in the cache the file there is:
`oesenc_pi-4.2.8-0_ubuntu-gtk3-x86_64-18.04.tar.gz`
but interestingly the file format does not seem to be correct internally, i.e. the directory that is shown when you open with archive manager is `.`.

If I download the 18.04 version and install it from the 'Import plugin...' it works. The 20.04 version does not. Why is it picking an 18.04 version for a 20.04 Ubuntu?

jongough, 2020-06-07:
It would appear that 'Plugin import...' is broken on Ubuntu 18.04, 20.04, Linux Mint 19.3 .

leamas, 2020-06-07:
> publish the xml/xsd, in there make it so that ALL valid values for ARCH etc. are documented.

The xsd file is published in the plugins project.

The arch is currently just an arbitrary token. It might be useful to make it an xml  'choice' to narrow the supported values. Same for 'target-os'. The drawback is limiting the flexibility, but it might still be worth it. I have filed https://github.com/OpenCPN/plugins/issues/112

As of now, this would just be helpful for developers, there is no xsd validation done in runtime.

rgleason, 2020-06-07:
Jon, has testplugin been revised? If so, I will try using that to fix weatherfax the next version I was trying to bring current to testplugin   on[ branch master-update2](https://github.com/rgleason/weatherfax_pi/tree/master-update2)  I had gotten it building locally the tarball with metadata.xml inside, and it works in Windows, I was just starting to test it in Ubuntu and was getting rejected because I was not able to pick the right ARCH.  So I think if there is a fix in Testplugin, that I can make those changes and weatherfax will work.

@bdbcat  the current weatherfax on master should work, but I have been trying to make weatherfax current and then will start on weather_routing, then hopefully I will be able to get to some others.  I am moving all Sean's plugins  Cloudsmith Repositories from my personnal one to OpenCPN organization. That gives all organization team members access to those plugin repos, so that fixing and deployment is easier.

bdbcat, 2020-06-07:
Jon...
"It would appear that 'Plugin import...' is broken on Ubuntu 18.04, 20.04, Linux Mint 19.3 ."
Works fine for me, if you will select the correct tarball. 
From master catalog, the current oesenc tarball for focal is:
https://dl.cloudsmith.io/public/david-register/ocpn-plugins-unstable/raw/names/oesenc-ubuntu-gtk3-x86_64-20.04-tarball/versions/4.2.8.0.adf78a4/oesenc_pi-4.2.8-0_ubuntu-gtk3-x86_64-20.04.tar.gz

bdbcat, 2020-06-07:
@rgleason ...
I'd be happy to help.  Shall I look at weatherRouting?  I think that I have push authorization for that one...

jongough, 2020-06-07:
I have been using the file you point at in [https://github.com/OpenCPN/OpenCPN/issues/1941#issuecomment-640227361](https://github.com/OpenCPN/OpenCPN/issues/1941#issuecomment-640227361) loaded through the 'Import plugin...' button and I get the error dialog.
[image]
This gives the dialog shown in [https://github.com/OpenCPN/OpenCPN/issues/1941#issue-632192893](https://github.com/OpenCPN/OpenCPN/issues/1941#issue-632192893)

jongough, 2020-06-07:
I have found the error. The cloudsmith repository [https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/?q=](https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/?q=) does not have a `focal` build available so I had installed a `bionic` build. When I then installed the `bionic` plugin it works, when I try to install the `focal` it fails. There really shouldn't be this distinction as both versions should be valid.

When can we expect a focal version of OCPN in the cloudsmith repository?

bdbcat, 2020-06-08:
Ready now.
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn-focal-gtk3_5.1.605+1.8869917-1_amd64.deb

bdbcat, 2020-06-08:
Looking for PR to add OCPN Draw to master catalog...

jongough, 2020-06-08:
I have to get the flatpak upload working again. I seem to have 'moved' something and only noticed in the recent build. I will have a new version ready soo.

rgleason, 2020-06-08:
Dave wrote:
```
@rgleason ...
I'd be happy to help. Shall I look at weatherRouting? I think that I have push authorization for that one...
```
Dave, That would be a great help.  I was having difficulty because it worked in Windows, but not Ubuntu 20 I believe, so I removed all Ubuntu versions. 

Jon has updated weather_routing about six days ago to the latest frontend2 and build for arm64, but I don't believe the earlier issue is fixed.  Also I believe that there are further updates to frontend2 from testplugin_pi that are needed.  

Would you like us to incorporate all the frontend2 updates and then perhaps you can look at it to fix the Ubuntu issue?

BTW I have recently added Cloudsmith Repositories:  [opencpn  / weather-routing-(prod, beta, alpha)](https://cloudsmith.io/~opencpn/repos/) 
and the most recent build is[ weather-routing-beta](https://cloudsmith.io/~opencpn/repos/weather-routing-beta/packages/)

rgleason, 2020-06-08:
Also, with an intel processor + Ubuntu 18.04 which plugin version should be used? I gather things are changing again.

leamas, 2020-06-08:
I have filed https://github.com/OpenCPN/plugins/pull/116, which basically tries to nail down allowed values for target and target-arch into a formal xsd description. Here is also a an xsd file which can be used to validate a single metadata file in a plugin workflow.

jongough, 2020-06-08:
Pull request generated as requested in [https://github.com/OpenCPN/OpenCPN/issues/1941#issuecomment-640322782](https://github.com/OpenCPN/OpenCPN/issues/1941#issuecomment-640322782)

jongough, 2020-06-08:
Ubuntu focal gtk3 installs OD from the master catalog and runs now. Updating two other plugins now and will add them to the master catalog when I get them into the 'prod' cloudsmith repositories

rgleason, 2020-06-08:
To use these plugins should we be doing: git pull upstream master of Opencpn and then build?
Will the internal plugins be automatically rebuilt such that they work properly with Plugin Installer?
Will older plugins be problematical?
I tried compiling and using the current Beta Opencpn in Ubuntu 18.04 and had great difficulty with freeze ups and had to "kill" Opencpn in the middle of Plugin Manager operations.

rgleason, 2020-06-09:
Alec's helpful contribution
https://github.com/OpenCPN/plugins/pull/116

bdbcat, 2020-06-09:
Rick...
Best if you download the latest CI build from cloudsmith, and install directly.  That way, we avoid any possible contamination from your local build setup.
For example, you can find the Ubuntu 18.04 build here:
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn-bionic_5.1.605+1.7dd75c1-1_amd64.deb

To install, download the .deb file.  Then:
sudo dpkg --install {deb file}

rgleason, 2020-06-09:
Ok, thanks, will do that.

rgleason, 2020-06-10:
Wouldn't it be simpler, clearer and more flexible if there were two xml tags 
target-os  and target-arch  and then test for the critical target-os tags at the end?

bdbcat, 2020-06-11:
Rick...
What we have now is certainly functional, if a little redundant on some platforms.
What scenario are you imagining that will stress the current form?
Also, please note that any new platforms that are suspected to be coming (e.g. Windows for ARM, or wx3.2 for Ubuntu/debian) will cause major ABI breakage anyway, and require a new OCPN version to do anything at all.  So we will have future opportunities for refine the process.
Also, my experience is that once the tooling is done correctly, it tends to quickly fade from interest.  I think we are at that point, or very close...

bdbcat, 2020-06-11:
Weather outing coming soon?

rgleason, 2020-06-11:
I understand your point, and Jon and I certainly have been iterating quite enough at this point,  but I think separating the os and arch tags, would in the long run make things easier. If it is not done now, then there will be dozens of PI that will need to be changed. I am not a programmer so perhaps I have this wrong. I hope you'll consider this small change the xsd for a moment.

rgleason, 2020-06-11:
Weatherrrouting works in Windows and Ubuntu 18.04 now, tested.
I can make a PR to plugiins, but will pause for you to consider the above.
I am going to move on to watchdog in the interim. Thanks.

bdbcat, 2020-06-11:
"would in the long run make things easier"
But how, exactly?  And for whom?

bdbcat, 2020-06-11:
See email re/VDR_PI....

rgleason, 2020-06-11:
I have this thought that separating the OS and ARCH in the xsd would give greater flexibility in making program changes and adjustments inside Opencpn and PI, needed for adding environments without affecting the other plugins.  The necessary checks for ARCH against the ARCH Tag could occur at the very end as needed. Couldn't a new environment be added once Opencpn was set up for checking the xml for it, and then a certain plugin add the ci script providing both the OS tag and the ARCH tag, with no logic needed.  The OpenCPN would do the checking as necessary. There would be no logic or code in the pluginconfigure.cmake required to configure the individual environment tags to the xsd, other than meeting the format for the OS tag and the ARCH tag which is simple.  OpenCPN would then do all the checking in its code.

rgleason, 2020-06-11:
Sorry, I have too much email VDR_pi?   date?

bdbcat, 2020-06-11:
VDR_PI in master catalog is broken on Windows.  Recent change, I suppose.  Where is the github source?

bdbcat, 2020-06-11:
Rick...
I guess you are talking about simplifying pluginconfigure.cmake, or maybe pluginsetup.cmake.
And I say again, those are done, and working, and not terribly complex, for the current environments.
Any new environment (i.e. one not currently supported) will require significant adaptation of OCPN core, and all plugins.  This is a given.  It is more than a matter of simply changing the XML.

rgleason, 2020-06-11:
Is this likly due to the changes in Opencpn and xsd checking?
I guess that frontend should be fixed to conform to the xsd too.
Watchdog is also probably the same deal.
This is why I think we should really get this right!
1. All plugins provide in xml an OS tag and an ARCH tag.
2 Pluginconfigure.cmake does not have to do small changes to a <target> value that has OS and ARCH or no ARCH value.
3. OpenCPN logic does the necessary checking on the OS tag value and ARCH tag values as required.

rgleason, 2020-06-11:
https://github.com/rgleason/vdr_pi/tree/master
I could update tomorrow morning perhaps.

bdbcat, 2020-06-11:
Rick...
The problem with VDR has nothing to do with metadata or XML.  It is some internal problem.
I will look at it...

bdbcat, 2020-06-11:
" Pluginconfigure.cmake does not have to do small changes to a value that has OS and ARCH or no ARCH value."
This is exactly 3 line of code in pluginconfigure, and it is done.
And really, think about it:
The chance of ever having an MSVC or Mac version of OCPN that runs on any processor other than x86_64 is practically zero.  And if it happens, then we add something like "MSVC-ARM64", or whatever.  This will be the least of our problems with ARM64 Windows.  And Apple supporting ARM?  Never happen....

rgleason, 2020-06-11:
Dave, I bow to your judgment and experience. I will make sure you have collaboration for my vdr_pi

rgleason, 2020-06-11:
You have access to make a pr and merge it.

bdbcat, 2020-06-11:
OK, I'm on it.

rgleason, 2020-06-11:
Will download wxroute metadata and make a plugins pr.

bdbcat, 2020-06-11:
Rick...
VDR_PI fixed.  Simple typo.  Imports and runs fine now.
Ready for rebuild/PR as you like.

leamas, 2020-06-11:
One a side note:

> And Apple supporting ARM? Never happen....

Rumour is that they are moving to ARM: https://arstechnica.com/gadgets/2020/06/apple-plans-to-announce-arm-transition-for-all-macs-at-wwdc-2020/

>  then we add something like "MSVC-ARM64"

Not likely, agreed. But at some point there might be  a need for msvc-win64.

Again:  just a side note.

bdbcat, 2020-06-11:
https://www.cruisersforum.com/forums/f134/opencpn-version-5-1-529-beta-test-234928-29.html#post3161273

rgleason, 2020-06-11:
So, just to understand more.  Whenever any new environment is introduced OpenCPN and all plugins have to be rewritten and changed?  What happens to the old plugins, do they still run?  Or is this just in certain instances when the xsd list of acceptable plugins has to change?

bdbcat, 2020-06-11:
Rick...
Depends on the degree of environment change.  Huge changes like Win10/ARM64, would require everything, including core and all plugins, to be rebuilt. Small change like adding Ubuntu-ARM64 for 20.04 (focal) requires only building the new plugins (using current Frontends) and adding the possibility to XSD.  OCPN core will already handle it.
Understand that the XSD does not control the build in any way.  It simply validates the syntax of the metadata files.  Future changes to XSD will most often be simply additive.
(Aside:  I am trying to imagine what would be involved in getting wxWidgets/Win10 for ARM64 built.   Or imagine wxWidgets on MacOS for ARM64.  Scary, and won't happen overnight...)

[17 more replies]

closed 2021-04-27 by bdbcat