OpenCPN/OpenCPN #1942: Plugin Manager - 'Import plugin...' not remembering the directory used

jongough, 2020-06-07:
Using OCPN 5.1.529+50d0e0b on Ubuntu 20.04 using 'Import plugin...' button on Plugins page does not remember the last directory used. It always resets to a default location. The directory should be remembered within a single session if not over multiple sessions. With the current behaviour it is a frustrating experience to load up multiple plugins.

Hakansv, 2021-04-25:
Implemented by https://github.com/OpenCPN/OpenCPN/pull/2195

bdbcat, 2021-04-27:
Please test/comment/close against current github master HEAD

jongough, 2021-04-27:
Commit: de5ee16d seems to have a directory remembered but does not update the directory if the plugin import failed. If the import succeeds the location is updated to where the successful imported file came from. This works across restarts.

bdbcat, 2021-04-29:
Please close after test.

jongough, 2021-04-30:
Works as expected

closed 2021-04-30 by jongough