OpenCPN/OpenCPN #1943: Plugin Manager - Rejecting plugin running on Linux Mint

jongough, 2020-06-07:
OCPN 5.1.605 compiled on Linux Mint importing testplugin_pi 1.0.110 for bionic (available in cloudsmith here: [https://dl.cloudsmith.io/public/jon-gough/testplugin_pi-alpha/raw/names/testplugin_pi-1.0.110.0-ubuntu-arm64_18.04-bionic-armh64-tarball/versions/1.0.110.0+1.a2cdd46/testplugin_pi-1.0.110.0_ubuntu-arm64_18.04-bionic-armh64.tar.gz](https://dl.cloudsmith.io/public/jon-gough/testplugin_pi-alpha/raw/names/testplugin_pi-1.0.110.0-ubuntu-arm64_18.04-bionic-armh64-tarball/versions/1.0.110.0+1.a2cdd46/testplugin_pi-1.0.110.0_ubuntu-arm64_18.04-bionic-armh64.tar.gz) is marked as incompatible. The plugin manager is not recognising compatible versions of linux.

bdbcat, 2020-06-07:
Jon...
And that is a bug.
Corrected in github now.
Thanks for the catch.
Dave

bdbcat, 2021-04-29:
Closing, duplicate.

closed 2021-04-29 by bdbcat