OpenCPN/OpenCPN #1947: Route Manager fails to import GPX planned speed

Rasbats, 2020-06-08:
Export GPX - the planned speed is inserted in the GPX.
Import GPX - planned speed in the GPX does not appear in route manager.

[https://github.com/OpenCPN/OpenCPN/blob/master/src/NavObjectCollection.cpp](url)

Line 546

```
else
                     if( ext_name == _T ( "opencpn:rte_properties" ) ) {
                        pTentRoute->m_PlannedSpeed = atof( ext_child.first_child().value() );
                     }
```
 "opencpn:rte_properties" should read  "opencpn:planned_speed" to allow the planned speed in a GPX to be imported successfully.

rgleason, 2020-06-08:
Thanks Mike, this means the SAR plugin's gpx files and routes will work properly doesn't it?

bdbcat, 2020-06-09:
Done, in github now.
Close?

Rasbats, 2020-06-09:
Many thanks Dave. That will open up some other possibilities.

closed 2020-06-09 by Rasbats