OpenCPN/OpenCPN #1948: infrastructure: Illegal download.opencpn.org certificate.

leamas, 2020-06-11:
Using the site https://download.opencpn.org thows an illegal certificate error. On firefox, the core message is  _Firefox does not trust this site because it uses a certificate that is not valid for download.opencpn.org. The certificate is only valid for the following names: *.netdna-ssl.com, netdna-ssl.com_

Error is easily reproduced by clicking link above.

The walk-around is to use http://download.opencpn.org.  However, using plain http is insecure and frowned upon in most larger communities these days. This bug is filed after trying to validate the appstream metadata, which fails. However, the overall issue is IMHO more general

This is, for the moment, a lower priority issue.

rgleason, 2020-06-11:
I believe opencpn.org should have a valid certificate because that is the way the site was set up by bdbcat. 
https://www.sslshopper.com/ssl-checker.html#hostname=opencpn.org
Using https://www.digicert.com/help/  and entering download.opencpn.org it doesn't pass because the name is not the same.   ie "download" is added so the certificate is not trusted.

I wonder if another certificate for "download.opencpn.org" could be added?

rgleason, 2020-06-11:
https://www.sslchecker.com/sslchecker   using opencpn.org shows that root1 information is missing everything else is ok.

https://www.sslchecker.com/sslchecker  using download.opencpn.org shows 3 items in the certificate chain are missing and it is not trusted.

leamas, 2020-06-11:
This is about download.opencpn.org, nothing else. opencpn.org is a different host with another address.

rgleason, 2020-06-11:
"I wonder if another certificate for "download.opencpn.org" could be added?"

leamas, 2020-06-11:
Of course, but the only entity which can do this is whoever runs download.opencpn.org -- I presume that it's not we.

rgleason, 2020-06-11:
Well, Dave is the owner and he and K are webmasters. When setup the website had a bonified good SSL certificate for opencpn.org.   With increased security and standards, this appears to be no longer accepted for urls like downloads.opencpn.org.  

Also I provided some links to SSL checkers above and it appears that one element of the chain is not good for the opencpn.org certificate  (validation of address etc) and for downloads.opencpn.org there are 3 links of the chain missing.  Hope this helps.

leamas, 2020-06-11:
```
$ host download.opencpn.org
download.opencpn.org is an alias for download.opencpn.netdna-cdn.com.
download.opencpn.netdna-cdn.com has address 151.139.239.32
$ host opencpn.org
opencpn.org has address 192.138.189.133

```
i. e., download.opencpn.org is another machine.

@bdbcat: could you possibly sort this out? Have you root access to download.opencpn.org and is thus able to install  a certificate there? 

This is not to say that messing with this is the top priority right now.

rgleason, 2020-06-11:
That's interesting. Good point about IP's.

Alec I notice that flatpak is building now for plugins. Don't know how you did it.

nohal, 2020-06-11:
The downloads are served by StackPath CDN, our free sponsored opensource account can't use custom TLS certificate.

nohal, 2020-06-11:
OpenCPN downloads with valid TLS are served through https://download-opencpn.netdna-ssl.com

leamas, 2020-06-11:
@nohal: Thanks for chiming in! 

I figured something like that. I'll leave the bug open since it *is* a minor problem.  That said, let's live with it, we have more pressing issues to deal with.

leamas, 2020-06-11:
oops, missed your last comment. Does this mean that the simple fix would be to change the dns alias

```
$ host download.opencpn.org
download.opencpn.org is an alias for download.opencpn.netdna-cdn.com.

```
so that download.opencpn.org -> download-opencpn.netdna-ssl.com  ? Or would this break http://download.opencpn.org (doesn't look so...)

nohal, 2020-06-11:
No, changing the download.opencpn.org CNAME record would not fix anything (And it would actually probably break the current HTTP downloads).
Hosting download.opencpn.org on a different webserver and making it do a HTTP 302 redirect to the CDN would somewhat do the job.
I have also just looked at the CDN and there is now a way to have a custom certificate and an [API](https://docs.maxcdn.com) for it, but I don't have time to implement the toolchain that would be required to use it with Letsencrypt right now and doing it manually every <3 months will certainly not make anybody happy.

leamas, 2020-06-11:
> doing it manually every <3 months will certainly not make anybody happy.

Indeed. 

I suggest  we leave this bug open trying to live with the unencrypted downloads for now before we spend too much time on it.

That said, for the record: do you have a link with the CDN certificate setup info?

leamas, 2020-06-11:
BTW: good night...

nohal, 2020-06-11:
Link to the respective API docs is in my previous comment. It became available for our account sometimes during the past year or so since I last visited the admin portal.

leamas, 2020-06-11:
I should  sleep, but I'm missing something:

>No, changing the download.opencpn.org CNAME record would not fix anything (And it would actually possibly break the current HTTP downloads).

Using https on download.opencpn.org obviously doesn't work:
```

wget http://download.opencpn.com/screenshots/82434783-624dea00-9a61-11ea-847b-347d91b8f0fd.png -- OK
wget https://download.opencpn.com/screenshots/82434783-624dea00-9a61-11ea-847b-347d91b8f0fd.png  -- Connection timeout (as expected)

```
However, using download-opencpn.netdna-ssl.com/ everything actually looks good:

```
wget https://download-opencpn.netdna-ssl.com/screenshots/82434783-624dea00-9a61-11ea-847b-347d91b8f0fd.png  -- OK

wget http://download-opencpn.netdna-ssl.com/screenshots/82434783-624dea00-9a61-11ea-847b-347d91b8f0fd.png  -- OK

```

Also, the existing certificate and download-opencpn.netdna-ssl.com looks perfectly fine.
So, why does not a simple CNAME change fix it?

nohal, 2020-06-11:
Because changing the CNAME DNS record will not magically create a valid certificate for `download.opencpn.org` name.
Even if we had a valid certificate for download.opencpn.org, which we don't now, and changed the CNAME, that certificate still must be installed to and used by the CDN.

leamas, 2020-06-11:
Right, I'm rusty.  The certificate is checked against the request headers.

The simple fix for the issue which made me file this is to use download-opencpn.netdna-ssl.com.  This is not user-visible, so no problem in this case. 

The more general case is still a problem, but we could IMHO live with it.

leamas, 2020-06-14:
After a night's sleep: this bug is resolved.  While https://opencp.download.org does not work, https://download-opencpn.netdna-ssl.com does. And since the need for secured URLs mainly are in various packaging contexts, this solution is good enough.

Closing.

@nohal:   Thanks!

closed 2020-06-14 by leamas