OpenCPN/OpenCPN #1950: API - Problem using "OCPN_downloadFileBackground" on MacOSX and Linux

Rasbats, 2020-06-13:
This may have been overlooked.

I am unable to compile plugins using this function with MacOSX or Linux without editing the 'ocpn_plugin.h' file. 

Existing:
(Line 41)
```
#ifdef __WXMSW__
#ifdef MAKING_PLUGIN
#  define DECL_IMP     __declspec(dllimport)
#endif
#endif

```
Needed to add this, below the above:

```
#ifndef DECL_IMP
#define DECL_IMP
#endif
```

leamas, 2020-06-13:
This looks extremely strange... The` #ifdef __WXMSW__` should be undefined on linux/macos. And thus, the DECL_IMP should not be expanded.

So, something strange. What happens if you add a  patch like

    -- a/include/ocpn_plugin.h 
    +++ b/include/ocpn_plugin.h
    @@ -39,6 +39,7 @@
     #endif
 
     #ifdef __WXMSW__
    +#error WXMSW is defined (!)
     #ifdef MAKING_PLUGIN     
     #  define DECL_IMP     __declspec(dllimport)
     #endif

instead?

Rasbats, 2020-06-13:
We are still left with using 'DECL_IMP' here:
Line 1287
```
#ifdef MAKING_PLUGIN
extern   DECL_IMP wxEventType wxEVT_DOWNLOAD_EVENT;
#else
extern   DECL_EXP wxEventType wxEVT_DOWNLOAD_EVENT;
#endif
```
`__declspec(dllimport)` is not valid for MacOSX (or Linux).
The plugin code defines 'MAKING_PLUGIN'.
This causes an error with MacOSX and Linux.

leamas, 2020-06-13:
hm... indeed. The culprit seems to be around line 44. What about this patch, which seems more symmetrical to the rest:

    diff --git a/include/ocpn_plugin.h b/include/ocpn_plugin.h
    index fe4deb89c..23876bc6a 100644
    --- a/include/ocpn_plugin.h
    +++ b/include/ocpn_plugin.h
    @@ -38,10 +38,10 @@
     #endif
     #endif
     
    -#ifdef __WXMSW__
    -#ifdef MAKING_PLUGIN
    +#if defined(__WXMSW__) && defined(MAKING_PLUGIN)
     #  define DECL_IMP     __declspec(dllimport)
    -#endif
    +#else 
    +#  define DECL_IMP
     #endif

Rasbats, 2020-06-13:
Yes, this works on MacOSX. Thanks. Much neater. 

My 'SailawayNMEA_pi' plugin uses MAKING_PLUGIN. Not sure what others you could use to test Linux. I will try a Linux compilation at some point.

```
#if defined (__WXMSW__) && defined(MAKING_PLUGIN)
#  define DECL_IMP     __declspec(dllimport)
# else
#  define DECL_IMP
#endif
```

Rasbats, 2020-06-13:
Sorry... I will close when tested with the updated master code.

leamas, 2020-06-13:
Now, I'm certainly not Dave, so we need to make a PR to actually the code. I suggest that you make it.

 You could add a _Closes:  #1950_ to the PR description to make it automatically close this issue when merged

Rasbats, 2020-06-13:
Alec...
My pull requests invariably go wrong! Not keeping up with the master code? Could you make it for me please?

leamas, 2020-06-13:
OK. I'll do it, but later (preparing dinner now.)

Rasbats, 2020-06-13:
Thanks. That's great.

bdbcat, 2020-06-13:
Patch committed.
Dave

rgleason, 2020-06-14:
I would appreciate a clear explanation of what the problem is, what solution is required, if it affects  plugin code, and if all other plugins in the master require yet another change, this time for MACOS,  and if so, what that change should be. At this point I am getting dizzy and overloaded. Thank you.

bdbcat, 2020-06-14:
Rick...
Does not relate to any of your managed plugins, since you do not use the affected function..  You would have known if it was a problem for you, because your plugins would not have compiled for Mac if that was the case.
Focus now.  AFAIK, everything is stable at this point.  I'll let you know if that situation changes.

leamas, 2020-06-14:
Just for the record: Dave's fix is in 433b8ae17af8 (thanks!). I suggest that this bug is closed

Rasbats, 2020-06-14:
Many thanks to Dave and Alec for sorting this out. Now closing.

closed 2020-06-14 by Rasbats

leamas, 2020-06-14:
Thanks for reporting!