OpenCPN/OpenCPN #1952: Chart database update will allways refresh all chart directories

douwefokkema, 2020-06-15:
Problem updating the charts database. When you want to update one group of charts it automatically starts to update all charts. This could take a long time. I added a few charts and could not use O for more than an hour, you do not want this at sea.

Douwe Fokkema

bdbcat, 2020-06-15:
What do you mean by:
"update one group of charts"
Dave

douwefokkema, 2020-06-15:
Sorry, to be more precise, I added one new chart directory, then O started to update all directories. Took a very long time. Just like P_Dub described on the forum.

douwefokkema, 2020-06-16:
I tried to reproduce. Removed the last added chart directory, closed option window with OK button. Added same directory again. O did scan the oeSenc directories again, but only a few seconds. No problem. I reported this as I saw another user having the same issue. But for now, nothing to do, one can't solve a problem that does not reproduce.

closed 2020-06-16 by douwefokkema