OpenCPN/OpenCPN #1956: x86_64 wrong architecture for Ubuntu

tai-fun, 2020-06-19:
I saw this problem when trying to compile the latest code for the squidio plugin.
As I understand it the new Plugin Manager uses the name x86_64, but this is not accepted by Ubuntu.
When trying to install the package I get a message about wrong architecture. Changing back to the traditional "amd64" takes care of the problem.
This problem needs to be nipped in the bud, before is spreads to other plugins.
The same problem may exist for other distros as well. Why change from amd64 to x86_64 all over cmake, when it's only the tar-ball used for the new plugin manager that needs it? As far as I understand it.... 
Thomas

rgleason, 2020-06-19:
Thomas, there are specific environments for armh, armh and armh64.  You will find them in the typical 13 enviroments in the Plugin Manager. It should be automatically selected.

bdbcat, 2020-06-19:
Thomas....
I concur that Ubuntu 20.04 correctly installs Squiddio 1.3.4.7.
So, what exactly were your steps, and what detailed error messages did you see?

rgleason, 2020-06-20:
Is it possible that the flatpak versions are intended to be used?  --Just wondering.

tai-fun, 2020-06-20:
Dave

This is exactly what I did, and just repeated..."

$git clone https://github.com/mauroc/squiddio_pi.git
$cd squiddio_pi
$mkdir build
$cd build
$cmake ..
$make
$sudo make package

producing package -> "squiddio_pi-1.3.6.0-ubuntu-x86_64-20.04.deb"

$sudo dpkg -i squiddio_pi

output->
"$ sudo dpkg -i squiddio_pi-1.3.6.0-ubuntu-x86_64-20.04.deb
dpkg: warning: parsing file '/var/lib/dpkg/tmp.ci/control' near line 1:
 'x86_64' is not a valid architecture name in 'Architecture' field:
character '_' not allowed (only letters, digits and characters '-')
dpkg: error processing archive squiddio_pi-1.3.6.0-ubuntu-x86_64-20.04.deb
(--install):
 package architecture (x86_64) does not match system (amd64)
Errors were encountered while processing:
 squiddio_pi-1.3.6.0-ubuntu-x86_64-20.04.deb"

leamas, 2020-06-20:
... which basically boils down to the question _Do we support using the .deb package for managed plugins?_

If we do, a hack changing x86_64 -> amd64 could be motivated for  the .deb package. Most environments uses x86_64 -- I would describe Debian's use of amd64 as an anomaly. 

If we don't, the proper solution would be to drop the .deb package. 

In any case, the current situation reported is IMHO clearly a bug which should be fixed. @tai-fun: good catch!

rgleason, 2020-06-20:
I believe the x86_64 target should be separate, as suggested earlier by Jon and I, without having known about this issue.  Maybe it's time to fix it once and for all, even though everything has to be rewritten and propagated again, but I am certainly not a program guru.

bdbcat, 2020-06-20:
This is not a show stopper, or reason for reboot.
I think the simple thing to do is modify squiddio (and other plugins as convenient) to use AMD64 when building a .deb installer.
I wonder if the x85_64 tag would be accepted by a vanilla debian installation? @leamas , have you tried that?

leamas, 2020-06-20:
Some remarks:
  - This is about the plugins, the code in main opencpn don't do anything wrong.
  - Yes, we could patch x86_64 -> amd64 in the code creating the .deb package. However, we need to describe the use case for the .deb package (what's the point with that package, really?)
  - No need to test x86_64 on vanilla debian, The official architecture names can be found at https://wiki.debian.org/SupportedArchitectures -- here is no x86_64.

leamas, 2020-06-20:
Expanding on above: the .deb package is created by code I updated a year ago. Back then, it was extremely important to just add the new tarball without breaking the existing .deb package.  Simply stated, it was about playing it safe.

Now we have a quite different situation, focusing on the tarballs and the new plugin loader. Until enlightened, I don't see the role for the .deb package in this context. Why not just drop it?

bdbcat, 2020-06-21:
Alec...
Dropping functionality always ends up biting back, someday.  Hard to predict where.  But saying "Sorry, no more .deb packages for plugins" will provoke dismay from somewhere...

On the other hand, the solution is simple, at least for Froned2 (this plugin, and others)

In PluginPackage.cmake, add this patch around line 94:
```

    if(ARCH MATCHES "x86_64")
      set(CPACK_DEBIAN_PACKAGE_ARCHITECTURE "amd64")
    else()
      set(CPACK_DEBIAN_PACKAGE_ARCHITECTURE ${ARCH})
    endif()  

```
Low risk, I think.  
Dave

leamas, 2020-06-21:
> Dropping functionality always ends up biting back, someday.

Indeed. Still,  IMHO we need to do such things occasionally to avoid unmaintained parts of little user value, and also to not block future development.

However, if we do it it should be done at the right time. I agree that now isn't the proper time to drop the .deb package from plugins, the first time would be next release (5.4?), after a full cycle with the new tarballs.

So, basically I agree with Dave: let's fix the .deb  package generation for now.

@tai-fun: does applying this patch solve your problem?

    diff --git a/cmake/PluginPackage.cmake b/cmake/PluginPackage.cmake
    index 569acd5..f5f9f13 100755
    --- a/cmake/PluginPackage.cmake
    +++ b/cmake/PluginPackage.cmake
    @@ -91,7 +91,11 @@ if(UNIX AND NOT APPLE)
     
         set(CPACK_DEBIAN_PACKAGE_DEPENDS ${PACKAGE_DEPS})
         set(CPACK_DEBIAN_PACKAGE_RECOMMENDS ${PACKAGE_RECS})
    -    set(CPACK_DEBIAN_PACKAGE_ARCHITECTURE ${ARCH})
    +    if (ARCH STREQUAL "x86_64")
    +      set(CPACK_DEBIAN_PACKAGE_ARCHITECTURE "amd64")
    +    else ()
    +      set(CPACK_DEBIAN_PACKAGE_ARCHITECTURE ${ARCH})
    +    endif ()
         set(CPACK_DEBIAN_PACKAGE_VERSION "${CPACK_PACKAGE_VERSION}")
         set(CPACK_DEBIAN_PACKAGE_SECTION "misc")
         set(CPACK_DEBIAN_COMPRESSION_TYPE "xz") # requires my patches to cmake

rgleason, 2020-06-21:
> No need to test x86_64 on vanilla debian, The official architecture names can be found at https://wiki.debian.org/SupportedArchitectures -- here is no x86_64.

Yes, intel is listed as  "i386" which is what Jon had sometime ago.

tai-fun, 2020-06-21:
@Alec Leamas <leamas.alec@gmail.com>
 Yes your patch does the job.

leamas, 2020-06-21:
If I understand this correctly, Dave and I agrees that the proper way to handle this is to update the plugins with the patch in https://github.com/OpenCPN/OpenCPN/issues/1956#issuecomment-647095584 (basically is Dave's fix, just packaged as a "real" patch)

@rgleason :
> Dave, Why this? I don't understand [...]

This is the very problem: Debian uses amd64 as the architecture name for the 64-bit instruction set used by most  Intel and Amd desktop cpus. Most other contexts, including us,  uses x86_64  as the name for the same architecture. So, when we generate a Debian package for 64-bit Intel/Amd cpus we need to use the amd64 name, not x86_64.

This is a Debian-only "feature" so there is no point making overall changes, it's just about how to make Debian packages for this specific architecture.

rgleason, 2020-06-21:
Thanks,  so the "if (ARCH STREQUAL "x86_64")"   is not actually checking the processor, it is just checking to see if ARCH is set to x86_64 in the code.  I now understand Debian uses  "amd64" for 64 bit  intel/amd.

bdbcat, 2020-06-21:
One reason to use "x86_64" instead of "AMD64":
I often answer questions like:
"Oh, I have an intel laptop, and I am moving to linux.  Ubuntu seems nice.  Can I use the "amd64" flavor of OpenCPN?"

rgleason, 2020-06-21:
So we can add those changes to PluginPackage.cmake.   I may wait for Jon to do that, then propagate the change to 4-5 plugins.

tai-fun, 2020-06-21:
I'm closing this now as everything seems sorted out.

rgleason, 2020-06-22:
Does Jon know about this?  I don't think so. I would not assume it. I think this should remain open until the change has been made and request that it be reopened.  There is a plan, but nothing has happened.

jongough, 2020-06-24:
I have a fix which I am testing at the moment.

jongough, 2020-06-25:
Frontend2 (testplugin_pi 1.0.120.0) contains a fix for this issue. It generates DEB files with the architecture set to 'amd64' whilst the Plugin Manager files all have the 'x86_64' (where appropriate) designation.

rgleason, 2020-06-25:
Thanks Jon, will make this change to wxfax wxroute & watchdog.  Dave, do you want it changed for vdr?  Should I follow oesenc?

bdbcat, 2020-06-26:
Rick...
Sure, update VDR if you have the time.  Follow the code in oesenc_pi.  See PluginPackage.cmake, line 107.
Thanks
Dave

rgleason, 2020-06-26:
I will have more time starting June 30 for 2 weeks.

rgleason, 2020-06-26:
I have added at approx line 112
 ```
    if(ARCH MATCHES "x86_64")
      set(CPACK_DEBIAN_PACKAGE_ARCHITECTURE "amd64")
    else()
      set(CPACK_DEBIAN_PACKAGE_ARCHITECTURE ${ARCH})
    endif()  

```
and will try pushing it.

jongough, 2020-06-26:
Rick,
   That process is handled by Frontend2 (testplugin_pi 1.0.120.0) and it is done in a options file for cpack. Not sure why you have to add any code.

Jon

rgleason, 2020-06-26:
Jon, VDR uses frontend1 code, but thanks for keeping an eye on it!
Tomorrow, I'll start on frontend2 pi's.

bdbcat, 2020-07-05:
close now?

closed 2020-07-05 by tai-fun