OpenCPN/OpenCPN #1957: Cannot upload route to garmin

leamas, 2020-06-24:
Last summer opencpn could upload routes to my very old garmin. However,  recent versions fails with message _Error on route upload: Garmin GPS not connected_.
. 
The message is wrong -- using gpsbabel I can first save the route as a GPX file and then upload it to my garmin. The garmin complains about an illegal waypoint, but still accepts the uploaded route.

Relevant part of log: 
```

22:59:40.183 MESSAGE multiplexer.cpp:493 Sent Route to Garmin GPS on port: /dev/nmea Unit identifies as: S 10     2.02
22:59:40.186 MESSAGE multiplexer.cpp:515 Error Sending Route to Garmin GPS on port: /dev/nmea LastGarminError is:  Error Code is 0
22:59:41.699 MESSAGE chart1.cpp:7406    ***GPS Watchdog timeout at Lat:65.5617   Lon: 22.1405

```

Tested version from flatpak nightly builds:

```
$ flatpak list
Name                     Application ID                            Version            Branch    Origin             Installation Freedesktop Platform     
OpenCPN                  org.opencpn.OpenCPN                       5.1.621+d85121a    master    opencpn2-origin    user

```

Sort of lost here. Any clue out there?

leamas, 2020-06-25:
Looking into libs/garmin/jeeps/gpsserial.c the following lines looks broken:

```
void GPS_Serial_Error(const char *mb, ...)
{
      va_list argp;
      va_start(argp, mb);

      sprintf(last_error, mb, argp);

      va_end(argp);

//      GPS_Error(mb);
/*dsr
        va_list ap;
        char msg[200];
        char *s;
        int b;

        va_start(ap, mb);
        b = vsnprintf(msg, sizeof(msg), mb, ap);
        s = msg + b;
        *s++ = ':';
        *s++ = ' ';
        *s++ = '\0';
/      FormatMessage( FORMAT_MESSAGE_FROM_SYSTEM, 0,
//                      GetLastError(), 0, s, sizeof(msg) - b - 2, 0 );
        strcat(msg, strerror(errno));
        GPS_Error(msg);
*/
```

the sprintf statement cannot work, valists needs to be handled by vsnprintf/vsprintf. This is not the root, but could explain the strange error message. 

Also, misusing  the valist macros  could cause all sorts of havoc.

leamas, 2020-06-25:
Attached patch fixes the error message, which now says `11:34:51.710 MESSAGE multiplexer.cpp:480 Garmin GPS could not be initialized on port: /dev/nmea Error Code is -4 LastGarminError is: SERIAL: tcsetattr error`

[0001-libs-garmin-Fix-error-message-handling-1957.patch.gz](https://github.com/OpenCPN/OpenCPN/files/4830370/0001-libs-garmin-Fix-error-message-handling-1957.patch.gz)

bdbcat, 2020-06-25:
Patch merged in master.

bdbcat, 2020-06-25:
So, then, one needs to step thru the code in GPS_Serial_Open(), figure out what is not right about the tcsetattr() call.

leamas, 2020-06-25:
Seems more complicated... I'm on it, but unsure if I can figure it out.

BTW: There is more of the same kind, wrong handling of va_list. Stay tuned.

leamas, 2020-06-25:
>  So, then, one needs to step thru the code [...]

One thing is that the logging from the garmin module is a mess, bad va_list usage and much lower level logging is just lost. This is a shame, communication problems are not that unusual and it should IMHO be possible to get better logs without debugger or recompilation. I have some ideas for this part.

As for the original problem, opencpn cannot upload route while gpsbabel can: no progress.

leamas, 2020-06-25:
The first  `tc_set_attr `error was me doing stupid things (multiple instances).

Doing it right, I get an error when trying to set up the link. In particular, opencpn  expects DLE (which is what?) without getting it.  Attaching  a log retrieved after enabling all available loggiing and routing it to the log file. 

[garmin.log.gz](https://github.com/OpenCPN/OpenCPN/files/4833989/garmin.log.gz)

Needless to say, gpsbabel does the right thing with my Garmin. 

Thoughts?

bdbcat, 2020-06-26:
Alec...
I need to find my old Garmin handheld GPS, install new batteries, find adapter cable, etc...
Then I will try this locally.  Please stand by.
Dave

bdbcat, 2020-06-27:
Set up my ancient GPS38, recompiled O with some debug turned ON (Ubuntu 16.04), successfully sent out a route, and got this log output:

```
Tx Data:10 fe 00 02 10 03 : ...(UNKNOWN )
Rx Data:10 06 02 fe 00 fa 10 03  � (SESACK  )
Rx Data:10 ff 13 29 00 2f 01 47 50 53 20 33 38 20 20 20 33 2e 30 33 20 00 dc 10 03  )/GPS 38   3.03  (UNKNOWN )
Tx Data:10 06 02 ff 00 f9 10 03 : .....(SESACK  )

```
1.  The DLE is hex value 0x10, the first character of any valid packet in this protocol.
2,  The first Tx is a "wakeup" call
3.  Succeeding 2 Rx are ACK for session, and device ID report.  If this does not happen, nothing else will work.

I am thinking that somehow you are missing the first byte from the port, which would be the 0x10.  I see in your log the first two bytes are 0x06, 0x02.  Same as my 2nd and 3rd bytes.

I could try this on FlatPak.
Any way you could test on native linux?

Dave

bdbcat, 2020-06-27:
And where is the current Flatpak daily build located?

leamas, 2020-06-27:
sailing, slow connection. the nightly builds: flatpak/README.md. Will try native linux when time + bandwidth available

leamas, 2020-06-27:
Still sailing, but better circumstances.
  
  - Logging in  flatpak is enabled using `flatpak run org.opencpn.OpenCPN --loglevel debug`
  - flatpak debugging can be done using gdb inside the sandbox (after starting a sandbox shell using e. g. `flatpak run --command=sh --devel org.openpcn.OpenCPN`). See https://docs.flatpak.org/en/latest/debugging.html
  - Testing native build: need some time, back later (some hours).

leamas, 2020-06-27:
Trying a native build, not really sorting things out.
  - Occasionally, I have been able to upload. Couldn't repeat it, though
  - Tried resetting both garmin and my linux pc. No joy.
  - I have a stable symlink called /dev/nmea which points to the current USB device, typically /dev/ttyUSB0 or /dev/ttyUSB1. I can enter this filename as the devie to use for uploading, but I then get error log about the 'ea' device.  For now, I use /dev/USB0 or /dev/ttyUSB1 from the pull-down list.
  - Have seen the same logs from the native build after getting  Garmin device not connected.
  - Is this a about physical problems, and gpsbabel being better to handle error conditions?
  - Need to make dinner, my partner has strong opinions on this point.

leamas, 2020-06-27:
Having reasonable stable status running the native version. I cannot say that it differs from the flatpak'd one, but using the native install is one thing less to care about.

Enclosing a logfile with the two different errors I get (which one seems to be random). At the end, here is also debug output from running gpsbabel, which works.

"confused"... thoughts?

[garmin2.log.gz](https://github.com/OpenCPN/OpenCPN/files/4841212/garmin2.log.gz)

Also, if I choose "Upload to new GPS" and writes the perfectly sane name /dev/nmea (symlinked to /dev/ttyUSB0) I get this log:

```
22:39:22.954 MESSAGE multiplexer.cpp:480 Garmin GPS could not be initialized on port: ea Error Code is -4
 LastGarminError is: SERIAL: tcsetattr error
 
```

Seems to be some other error, probably related to line  466 in multiplexer.c (short_com is the logged name)

`wxString short_com = com_name.Mid(7);`

But then again, this is sidetracking the main problem.

bdbcat, 2020-06-28:
Alec...
It got late, not much time to look at this...
However, take a look at

static int32 GPS_A000(const char *port)

found in file gpsapps.c.
Especially consider the comment around line: 288
```

      // It seems that sometimes the first byte received after port open is lost...
        //      On error, Flush() and try once more....
  
```

Hmmmmm....

bdbcat, 2020-06-28:
Alec...
Consider this:
https://stackoverflow.com/questions/13013387/clearing-the-serial-ports-buffer

Maybe adding a delay after opening the port, before the first flush() will help?

leamas, 2020-06-28:
Tried add some delays according to above. Getting these logs. Looks better, but still not there. Current diff: 

```
diff --git a/libs/garmin/jeeps/gpsserial.c b/libs/garmin/jeeps/gpsserial.c
index 3ab01a541..7f11ee1cb 100644
--- a/libs/garmin/jeeps/gpsserial.c
+++ b/libs/garmin/jeeps/gpsserial.c
@@ -326,7 +326,7 @@ int32 GPS_Serial_Open(gpsdevh *dh, const char *port)
        gps_errno = SERIAL_ERROR;
        return 0;
     }
-
+usleep(50000);
     if(tcgetattr(psd->fd,&psd->gps_ttysave)==-1)
     {
        gps_errno = HARDWARE_ERROR;
@@ -432,7 +432,7 @@ int32 GPS_Serial_Write(gpsdevh *dh, const void *obuf, int size)
 int32 GPS_Serial_Flush(gpsdevh *fd)
 {
     posix_serial_data *psd = (posix_serial_data *)fd;
-
+sleep(1);
     if(tcflush(psd->fd,TCIOFLUSH))
     { GPS_Serial_Error("SERIAL: tcflush error: %s", strerror(errno));

```

Logs: 
[garmin3.log.gz](https://github.com/OpenCPN/OpenCPN/files/4841995/garmin3.log.gz)
[garmin4.log.gz](https://github.com/OpenCPN/OpenCPN/files/4841996/garmin4.log.gz)

leamas, 2020-06-28:
Also have this snippe tin the log sometimes. Significant?!

```
11:29:55.682 MESSAGE gps_wx_logging.h:30 Unit:  GPS 120     2.02 
ID:     47
Version:        2.02
11:29:55.862 WARNING gps_wx_logging.h:22 A001 protocol not supported
11:29:55.863 MESSAGE multiplexer.cpp:493 Sent Route to Garmin GPS on port: /dev/ttyUSB0
 Unit identifies as: GPS 120     2.02 
```

bdbcat, 2020-06-28:
Alec....
Looks like a "simple" dropped byte.

The good GPSBabel log:

```
Tx Data:10 fe 00 02 10 03 : ...(PRDREQ  )
Rx Data:10 06 02 fe 00 fa 10 03  .. (ACK     )
Rx Data:10 ff 16 2f 00 ca 00 47 50 53 20 31 32 30 20 20 20 20 20 32 2e 30 32 20 00 d3 10 03  ....GPS.120.....2.02.. (PRDDAT  )

```

And the OCPN log, only the third packet, first few bytes...

```
11:20:31.805   DEBUG gps_wx_logging.h:38 Rx Data:
11:20:31.806   DEBUG gps_wx_logging.h:38 10 
11:20:31.807   DEBUG gps_wx_logging.h:38 ff 
11:20:31.809   DEBUG gps_wx_logging.h:38 2f 
11:20:31.810   DEBUG gps_wx_logging.h:38 00 
11:20:31.811   DEBUG gps_wx_logging.h:38 ca 
11:20:31.812   DEBUG gps_wx_logging.h:38 00 
11:20:31.814   DEBUG gps_wx_logging.h:38 50 
11:20:31.815   DEBUG gps_wx_logging.h:38 53 
11:20:31.816   DEBUG gps_wx_logging.h:38 20 

```
Note that the third byte (0x16) is missing from the OCPN trace.  And that byte happens to be the packet data length indicator...

What USB adapter are you using?

leamas, 2020-06-28:
Ashore now, will check  adapter later. But in the end, same adapter works when using gpsbabel.

Looking at the gpsbabel code I see that this also uses a bundled copy of the garmin/jeeps library. Since gpsbabel works and overall seems well  maintained I'm considering making a try to compile using the latest garmin/jeeps from gpsbabel project. Some changes are inevitable, but perhaps not that much.

Rasbats, 2020-06-30:
Alec... Please keep going! I have had to use gpsbabel for sending routes to my Garmin GPSMAP 4010 for a long time. Looked at the code and ran away. A fix would help a lot of people.

leamas, 2020-07-01:
This is a Real Mess (tm). I have  updated the jeeps drectory and friends with files from the gpsbabel project. The result is the same. This leads me to think that the low-level stuff is OK, the problem is how it is used. Here are globals to get rid of (again...). 

Will try to find a way to update protocolhandler and/or multiplexer to use the official API. Will take a lot of time I don't have...

bdbcat, 2020-07-01:
@Rasbats 
If you have the time, I would like to see the debug logfiles from a session on your machine, as you try to upload a route to Garmin. Similar to Alec's debug output.

After some tiny patch, Alec's dataflow looks almost OK, just had one dropped byte.  So I am suspicious of hardware problems.
btw, works fine for me locally.

Thanks
Dave

Rasbats, 2020-07-01:
Sorry Dave... My wife is shielding from Covid at the moment. Hope things ease from 01 Aug when I hope to get back to my boat. No gear at home for a test at the moment.

bdbcat, 2020-07-01:
Mike...
I understand.
I wish you and your family well in these tough times....
Be safe...
Dave

leamas, 2020-07-01:
@bdbcat: could it be collisions between the m_garmin_serial_thread  in ProtocolHandler.cpp and the route upload code which executes in the main thread? 

Seems that the thread accesses the port at random times, issuing a command and waits for input. Could it steal a byte or two, so to speak?

I'm only occasionally at my boat, grandchild at home. Long turn-around times.

tai-fun, 2020-07-01:
Dave
I have an old Garmin GPS 128.
If that is interesting, tell me what to do.

From a recent route upload, that did not make it to the Garmin
19:14:29.946 MESSAGE datastream.cpp:227 Opening NMEA Datastream
Serial:/dev/ttyUSB0
19:14:30.046 MESSAGE multiplexer.cpp:669 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECWPL,5606.647,N,01529.922,E,001*6D
19:14:30.246 MESSAGE multiplexer.cpp:669 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECWPL,5606.793,N,01529.553,E,002*6C
19:14:30.446 MESSAGE multiplexer.cpp:669 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECWPL,5606.728,N,01529.325,E,003*6A
19:14:30.647 MESSAGE multiplexer.cpp:669 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECWPL,5606.472,N,01529.133,E,004*64
19:14:30.847 MESSAGE multiplexer.cpp:669 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECWPL,5606.345,N,01529.105,E,NM001*64
19:14:31.047 MESSAGE multiplexer.cpp:669 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECWPL,5605.881,N,01528.976,E,005*6E
19:14:31.248 MESSAGE multiplexer.cpp:669 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECWPL,5605.260,N,01528.911,E,006*69
19:14:31.448 MESSAGE multiplexer.cpp:669 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECWPL,5604.185,N,01529.505,E,007*69
19:14:31.648 MESSAGE multiplexer.cpp:669 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECWPL,5603.911,N,01530.089,E,NM001*67
19:14:31.848 MESSAGE multiplexer.cpp:669 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECWPL,5559.688,N,01548.140,E,008*65
19:14:32.049 MESSAGE multiplexer.cpp:669 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECWPL,5600.261,N,01549.775,E,009*69
19:14:32.249 MESSAGE multiplexer.cpp:669 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECWPL,5603.316,N,01551.612,E,010*6A
19:14:32.449 MESSAGE multiplexer.cpp:669 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECWPL,5604.624,N,01552.100,E,011*6F
19:14:32.649 MESSAGE multiplexer.cpp:669 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECWPL,5605.605,N,01551.525,E,012*6E
19:14:32.850 MESSAGE multiplexer.cpp:899 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECRTE,7,1,c,1,001,002*12
19:14:33.050 MESSAGE multiplexer.cpp:899 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECRTE,7,2,c,1,003,004*15
19:14:33.250 MESSAGE multiplexer.cpp:899 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECRTE,7,3,c,1,NM001,005*14
19:14:33.450 MESSAGE multiplexer.cpp:899 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECRTE,7,4,c,1,006,007*15
19:14:33.651 MESSAGE multiplexer.cpp:899 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECRTE,7,5,c,1,NM001,008*1F
19:14:33.851 MESSAGE multiplexer.cpp:899 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECRTE,7,6,c,1,009,010*1E
19:14:34.051 MESSAGE multiplexer.cpp:899 -->GPS Port:Serial:/dev/ttyUSB0
Sentence: $ECRTE,7,7,c,1,011,012*14
19:14:34.452 MESSAGE datastream.cpp:256 Closing NMEA Datastream
Serial:/dev/ttyUSB0
19:14:34.452 MESSAGE datastream.cpp:263 Stopping Secondary Thread
19:14:35.452 MESSAGE datastream.cpp:275 Stopped in 1 sec.

Thomas

leamas, 2020-07-01:
@tai-fun : Could you please start opencpn using --loglevel debug, try to upload and submit the log?

The logg will be bigger, it might be easier if you attach the log instead of pasting it directly in the bug.

tai-fun, 2020-07-01:
Here you go...
Latest session.

 
________

19:58:05.199    INFO logger.cpp:169 ------- OpenCPN version 5.1.621 restarted at 2020-07-01 -------

19:58:05.199    INFO logger.cpp:169 Using loglevel DEBUG
19:58:05.199 MESSAGE chart1.cpp:1883 wxWidgets version: wxWidgets 3.0.4 Linux 64 bit wxGTK
19:58:05.202 MESSAGE chart1.cpp:1888 OS Version reports as:  5.4
19:58:05.203 MESSAGE chart1.cpp:1890 MemoryStatus:  mem_total: 15899 mb,  mem_initial: 377 mb
19:58:05.203 MESSAGE chart1.cpp:1899 SData_Locn is /usr/local/share/opencpn/
19:58:05.203 MESSAGE chart1.cpp:1915 PrivateDataDir is /home/thomas/.opencpn
19:58:05.203 MESSAGE chart1.cpp:1972 Using existing Config_File: /home/thomas/.opencpn/opencpn.conf
19:58:05.211 MESSAGE navutil.cpp:1198 winPluginDir, read from ini file: 
19:58:05.212 MESSAGE navutil.cpp:1291 Setting Ownship Lat/Lon 56.1053, 15.4998
19:58:05.213 MESSAGE navutil.cpp:3160 Switch InlandEcdis mode Off
19:58:05.215 MESSAGE styles.cpp:854 Styles loading from /usr/local/share/opencpn/uidata/styles.xml
19:58:05.217 MESSAGE styles.cpp:831 No styles found at: /home/thomas/
19:58:05.217 MESSAGE styles.cpp:831 No styles found at: /home/thomas/.opencpn/
19:58:05.228 MESSAGE styles.cpp:786 Using SVG Icons
19:58:05.228 MESSAGE chart1.cpp:2030 Detected display size (horizontal): 270 mm
19:58:05.228 MESSAGE OCPNPlatform.cpp:925 Getting DefaultSystemLocale...
19:58:05.228 MESSAGE chart1.cpp:2067 System default Language:  en_GB
19:58:05.228 MESSAGE chart1.cpp:2070 Config file language:  en_US
19:58:05.228 MESSAGE chart1.cpp:2075 Adjusted App language:  en_US
19:58:05.228 MESSAGE OCPNPlatform.cpp:1010 ChangeLocale: Language load for:  en_US
19:58:05.229 MESSAGE OCPNPlatform.cpp:1062 ChangeLocale: Locale Init OK for:  en_US
19:58:05.229 MESSAGE OCPNPlatform.cpp:1081 Loading catalog for opencpn core.
19:58:05.229 MESSAGE OCPNPlatform.cpp:925 Getting DefaultSystemLocale...
19:58:05.229 MESSAGE chart1.cpp:2083 Opencpn language set to:  en_US
19:58:05.229 MESSAGE ConfigMgr.cpp:686 Loading Configs catalog: /home/thomas/.opencpn/Configs/configs.xml
19:58:05.230 MESSAGE chart1.cpp:2305 Creating MyFrame...size(1024, 732)  position(0, 0)
19:58:05.323 MESSAGE chart1.cpp:2877  **** Connect stuff
19:58:05.373 MESSAGE chcanv.cpp:970 Creating glChartCanvas
19:58:05.504 MESSAGE chcanv.cpp:2349 Metrics:  m_display_size_mm: 270     g_Platform->getDisplaySize():  1024:768   
19:58:05.504 MESSAGE chcanv.cpp:2354 wxDisplaySize(): 1024 768
19:58:05.504 MESSAGE TCDataSource.cpp:52 Loading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONIC.IDX
19:58:05.535 MESSAGE TCDataSource.cpp:52 Loading Tide/Current data source: /home/thomas/Sailing/Navigation/Tides/TIDALfiles/Harmonic_France/HARMONIC.IDX
19:58:05.686 MESSAGE glChartCanvas.cpp:1211 OpenGL-> Renderer String: Mesa DRI Intel(R) HD Graphics 3000 (SNB GT2)
19:58:05.686 MESSAGE glChartCanvas.cpp:1220 OpenGL-> Version reported:  3.0 Mesa 20.0.4
19:58:05.687 MESSAGE glChartCanvas.cpp:1271 OpenGL-> Detected Intel renderer, disabling stencil buffer
19:58:05.688 MESSAGE glChartCanvas.cpp:1314 OpenGL-> Texture rectangle format: de1
19:58:05.688 MESSAGE glChartCanvas.cpp:1359 OpenGL-> Using Vertexbuffer Objects
19:58:05.688 MESSAGE glChartCanvas.cpp:1177 OpenGL Framebuffer OK, size = 1022
19:58:05.690 MESSAGE glChartCanvas.cpp:1440 OpenGL-> Using Framebuffer Objects
19:58:05.690 MESSAGE glChartCanvas.cpp:1443 OpenGL-> Using FBO Stencil buffer
19:58:05.690 MESSAGE glChartCanvas.cpp:1451 OpenGL-> Using Depth buffer clipping
19:58:05.690 MESSAGE glChartCanvas.cpp:1570 OpenGL-> Using s3tc dxt1 compression
19:58:05.690 MESSAGE glChartCanvas.cpp:1600 OpenGL-> Compressed tile size: 128kb (8:1)
19:58:05.690 MESSAGE glChartCanvas.cpp:1464 OpenGL-> Minimum cartographic line width:  1.0
19:58:05.690 MESSAGE glChartCanvas.cpp:1466 OpenGL-> Minimum symbol line width:  1.1
19:58:05.690 MESSAGE gshhs.cpp:1377 Loading World Chart Q=0 in 0 ms.
19:58:05.690 MESSAGE gshhs.cpp:176 GSHHS World chart files have wrong version. Found -1, expected 210-220.
19:58:05.690 MESSAGE gshhs.cpp:1377 Loading World Chart Q=4 in 0 ms.
19:58:05.792 MESSAGE chartdb.cpp:240 ChartDB Cache policy:  Application target is 1024 MBytes
19:58:05.792 MESSAGE chartdbs.cpp:145 Loading chart db version: V018
19:58:05.792 MESSAGE chartdbs.cpp:1227 Chartdb: Chart directory list follows
19:58:05.792 MESSAGE chartdbs.cpp:1248   Chart directory #0: /home/thomas/Sailing/Charts/CM93
19:58:05.792 MESSAGE chartdbs.cpp:1248   Chart directory #1: /home/thomas/Sailing/Charts/Raster/DMA/0
19:58:05.792 MESSAGE chartdbs.cpp:1248   Chart directory #2: /home/thomas/Sailing/Charts/Raster/EUR/Baltic
19:58:05.792 MESSAGE chartdbs.cpp:1248   Chart directory #3: /home/thomas/Sailing/Charts/Raster/EUR/Sweden
19:58:05.792 MESSAGE chartdbs.cpp:1248   Chart directory #4: /home/thomas/Sailing/Charts/MBTiles
19:58:05.792 MESSAGE chartdbs.cpp:1248   Chart directory #5: /home/thomas/Sailing/Charts/TD-ST-HGX
19:58:05.792 MESSAGE chartdbs.cpp:1248   Chart directory #6: /home/thomas/Sailing/Charts/Fugawi
19:58:05.792 MESSAGE chartdbs.cpp:1248   Chart directory #7: /home/thomas/Sailing/Charts/Vector/oeSENC
19:58:05.792 MESSAGE chartdbs.cpp:1248   Chart directory #8: /home/thomas/Sailing/Charts/Raster/UKHO_Merged_2015
19:58:05.792 MESSAGE chartdbs.cpp:1248   Chart directory #9: /home/thomas/Charts/ENC
19:58:05.792 MESSAGE chartdbs.cpp:1248   Chart directory #10: /home/thomas/Sailing/Charts/Fugawi2
[... 510 lines cut ...]
[... cut ...]
19:58:11.634 MESSAGE pluginmanager.cpp:1189 Skipping not enabled candidate.
19:58:11.634 MESSAGE pluginmanager.cpp:1141 Checking plugin candidate: /usr/local/lib/opencpn/libvdr_pi.so
19:58:11.634 MESSAGE pluginmanager.cpp:1189 Skipping not enabled candidate.
19:58:11.634 MESSAGE pluginmanager.cpp:1141 Checking plugin candidate: /usr/local/lib/opencpn/libvfkaps_pi.so
19:58:11.634 MESSAGE pluginmanager.cpp:1189 Skipping not enabled candidate.
19:58:11.634 MESSAGE pluginmanager.cpp:1141 Checking plugin candidate: /usr/local/lib/opencpn/libsquidd

leamas, 2020-07-01:
Hm, sorry, my bad, I was too short. What we need is the log after doing something like:

  - Start using --loglevel debug (as you did last time).
  - In  _Settings | Connections,_ verify  that "Use Garmin GRMN (Host) Mode" is checked
  - If you don't have any routes defined, create a new route in the Route Editor.
  - In the Route Manager, select a route.
  - Push the Send To GPS button, and wait until something happens.
  - At this very point , kill opencpn.
  - In the log, you can search for` MESSAGE garmin_wrapper_utils.h:64 GPS Serial Open at 9600, `and remove everything above it.
  - submit the log

tai-fun, 2020-07-01:
Pretty much what I did, but..
My Garmin is configured to use NMEA0183, so "Use Garmin GRMN (Host) Mode"
does not work  at all.
I get a pop up:Garmin GPS not Connected. No surprise.

rgleason, 2020-07-01:
There is a garmin program called Homeport which I used to use to connect to my very early GPS I believe that program accepted the Garmin Host mode.  It is still on my laptop. I may have some routes in it. I wonder if output from that would help?

rgleason, 2020-07-01:
Would wireshark work somehow?

transmitterdan, 2020-07-01:
No, we are speaking about serial ports.  Wireshark is for Ethernet style networks.

bdbcat, 2020-07-01:
Alec:
re: "could it be collisions between the m_garmin_serial_thread in ProtocolHandler.cpp and the route upload code which executes in the main thread?"

Well, anything is possible.  But, the code was built to close and terminate the"m_garmin_serial_thread" before starting the upload, then restarting the background thread after completion of the upload.
Easy way to test:  Disable the normal GPS position input from this port.  Then the upload process is assured exclusive access to the port.

Dave

bdbcat, 2020-07-01:
But I come back to the debug traces above.  After adding the Sleep() call between port Open() and first flush(), we seem to be getting almost sensible data receive bytes from the device.  Except, in one trace we see a "lost" byte, which breaks the protocol.
Sure smells like a hardware problem to me.
I find that this process works fine for me on Ubuntu16.04, old Garmin GPRS-38 handheld unit.  Never fails.

bdbcat, 2020-07-01:
tai-fun...
Please try again, with the port deactivated for normal NMEA messages. However, be sure to enable the tick-box for Garmin upload globally. ("Use Garmin host-mode for uploads").  Create a route, right click, "Send to New GPS".

Done this way, there is no possibility of thread collisions.

leamas, 2020-07-02:
@bdbcat yes, smells hw problem. But gpsbabel always work...will try disabling thread, later

leamas, 2020-07-02:
@bdbcat: Stopping the thread is done using GarminProtocolHandler::StopSerialThread(), right?

That code contains logging, and should leave  a message "Stopped after xx seconds" or "Not stopped after 5 sec". I can see these messages in he log when opencpn terminates, but not when doing an upload. Can you see them in your logs?

tai-fun, 2020-07-02:
@bdbcat
As said before, when "Use Garmin host-mode for uploads" is active I get the
error message that  "Garmin GPS not Connected". My Garmin 128 is set to
NMEA/NMEA.

bdbcat, 2020-07-02:
@tai-fun 
And if you set your GPS to GRMN mode?

bdbcat, 2020-07-02:
@leamas 
Please, let us try without the normal receive thread running at all.  Just to confirm there is no thread funniness interfering.
If that fixes the problem, then we investigate the thread interaction.

leamas, 2020-07-02:
@bdbcat: some progress. Using the debugger I can step, and then it works, have been able to  upload routes.

The thread is disabled while doing it; havn't checked if it's part of the issue. However, here is definitely timing problems, without breakpoints it fails.

In my garmin wrapper.cpp I  have this:

```
           if(!bfound_empty)
            {
                  int rv = OCPNMessageBox(NULL, _("Overwrite Garmin device route number 1?"),
                                          _("OpenCPN Message"), wxOK | wxCANCEL | wxICON_QUESTION);
                  wxLogDebug("OCPNMessageBox returns: %d", rv);
                  if (rv != wxOK)
                       return 0;
            }
```

Looks innocent, but the messagebox returns 5101 for cancel button and 5100 for  the Ok button. Definitely wrong. Ideas? Currently walking around by  simply disabling the rc != wxOK test.

Need some food.

leamas, 2020-07-02:
@tai-fun: It might report "Garmin GPS not connected" as a result af the bug we are chasing. If you could submit your log when getting this it might be helpful, preferably handled as described in https://github.com/OpenCPN/OpenCPN/issues/1957#issuecomment-652579160

bdbcat, 2020-07-02:
@leamas 
Should be:
                  if (rv != wxID_OK)

leamas, 2020-07-02:
Right. This bug was committed 2010 in fcf37b7bac. Seems that this code have had problems for a long time.

Back from dinner. Let's see how long time I can spend with the debugger...

leamas, 2020-07-02:
Currently looking into multiplexer.cpp, around line  469:

```
// Initialize the Garmin receiver, build required Jeeps internal data structures
        wxLogDebug("Initializing Garmin port %s", short_com);
        wxSleep(5);
        int v_init = Garmin_GPS_Init(short_com);
        if(v_init < 0)
```

With a breakpoint before the Garmin_GPS_Init() call, it works, Without breakpoint it fails, despite the wxSleep() I added just to test.

Why does the mere act of executing a breakpoint affect the result? Other threads? signals?

Yes, debugger affecting things is not the first time. But still, sort of frustrating.

leamas, 2020-07-02:
@bdbcat : ^

leamas, 2020-07-02:
@bdbcat: Next problem: in a new test session, I try to upload. The device has changed from /dev/ttyUSB0 to /dev/ttyUSB1 i . e ttyUSB1 exists but not ttyUSB0. This is normal, udev stuff.

Despite this, the device I can select in the drop-down list is ttyUSB0. Together with the fact that manually entering the device name is broken here are problems.

Walking around by entering the device name *serial:/dev/ttyUSB1*

Why on earth does it offer me a non-existing device? (yes, I have restarted program).

tai-fun, 2020-07-02:
> in #1957 (comment)

@Alec Leamas <leamas.alec@gmail.com>
 Log attached.
@dsr <bdbcat@yahoo.com>
Still using nmea/nmea, as nothing works using GRMN/GRMN.
I looked in the Garmin 38 manual and it seems identical to my Garmin 128.
Which are your exact settings for the option GRMN/GRMN?

 
20:45:25.118   DEBUG chart1.cpp:8881 SatsInView: 10
20:45:25.118   DEBUG chart1.cpp:8881 SatsInView: 12
20:45:25.118   DEBUG chart1.cpp:8881 SatsInView: 12
20:45:25.118   DEBUG chart1.cpp:8881 SatsInView: 12
20:45:25.240 MESSAGE chartdb.cpp:1194 All chart in cache locked, size: 2
20:45:26.258   DEBUG gps_wx_logging.h:38 Tx Data:
20:45:26.258   DEBUG gps_wx_logging.h:38 10 
20:45:26.258   DEBUG gps_wx_logging.h:38 fe 
20:45:26.258   DEBUG gps_wx_logging.h:38 00 
20:45:26.258   DEBUG gps_wx_logging.h:38 02 
20:45:26.258   DEBUG gps_wx_logging.h:38 10 
20:45:26.258   DEBUG gps_wx_logging.h:38 03 
20:45:26.258   DEBUG gps_wx_logging.h:38 : 
20:45:26.258   DEBUG gps_wx_logging.h:38 .
20:45:26.258   DEBUG gps_wx_logging.h:38 .
20:45:26.258   DEBUG gps_wx_logging.h:38 .
20:45:26.258   DEBUG gps_wx_logging.h:38 (UNKNOWN )

20:45:26.258   DEBUG gps_wx_logging.h:38 Rx Data:
20:45:26.259   DEBUG gps_wx_logging.h:38 80 
20:45:26.259   ERROR gps_wx_logging.h:14 GPS_Packet_Read: No DLE.  Data received, but probably not a garmin packet.

20:45:26.259   ERROR gps_wx_logging.h:14 GPS_Get_Ack error: -7
20:45:26.259   DEBUG gps_wx_logging.h:38 Tx Data:
20:45:26.259   DEBUG gps_wx_logging.h:38 10 
20:45:26.259   DEBUG gps_wx_logging.h:38 fe 
20:45:26.259   DEBUG gps_wx_logging.h:38 00 
20:45:26.259   DEBUG gps_wx_logging.h:38 02 
20:45:26.259   DEBUG gps_wx_logging.h:38 10 
20:45:26.259   DEBUG gps_wx_logging.h:38 03 
20:45:26.259   DEBUG gps_wx_logging.h:38 : 
20:45:26.259   DEBUG gps_wx_logging.h:38 .
20:45:26.259   DEBUG gps_wx_logging.h:38 .
20:45:26.259   DEBUG gps_wx_logging.h:38 .
20:45:26.259   DEBUG gps_wx_logging.h:38 (UNKNOWN )

20:45:26.259   DEBUG gps_wx_logging.h:38 Rx Data:
20:45:26.260   DEBUG gps_wx_logging.h:38 18 
20:45:26.260   ERROR gps_wx_logging.h:14 GPS_Packet_Read: No DLE.  Data received, but probably not a garmin packet.

20:45:26.260   ERROR gps_wx_logging.h:14 GPS_Get_Ack error -7: 
20:45:26.351 MESSAGE RolloverWin.cpp:161 New texture  138
20:45:26.351 MESSAGE RolloverWin.cpp:174 Render texture  138
20:45:26.423 MESSAGE RolloverWin.cpp:308 Draw texture  138
20:45:26.499 MESSAGE RolloverWin.cpp:308 Draw texture  138
20:45:26.576 MESSAGE RolloverWin.cpp:308 Draw texture  138
20:45:26.600   DEBUG chart1.cpp:8881 SatsInView: 10
20:45:26.653 MESSAGE RolloverWin.cpp:308 Draw texture  138
20:45:26.727 MESSAGE RolloverWin.cpp:308 Draw texture  138
20:45:26.843   DEBUG chart1.cpp:8881 SatsInView: 12
20:45:26.894 MESSAGE RolloverWin.cpp:308 Draw texture  138
20:45:26.968   DEBUG chart1.cpp:8881 SatsInView: 12
20:45:27.088 MESSAGE RolloverWin.cpp:308 Draw texture  138
20:45:27.114   DEBUG chart1.cpp:8881 SatsInView: 12
20:45:27.386 MESSAGE RolloverWin.cpp:308 Draw texture  138
20:45:27.460 MESSAGE RolloverWin.cpp:308 Draw texture  138
20:45:27.612 MESSAGE RolloverWin.cpp:308 Draw texture  138
20:45:27.865 MESSAGE RolloverWin.cpp:308 Draw texture  138
20:45:28.088 MESSAGE RolloverWin.cpp:308 Draw texture  138
20:45:28.365 MESSAGE RolloverWin.cpp:308 Draw texture  138
20:45:28.547   DEBUG chart1.cpp:8881 SatsInView: 12
20:45:28.842   DEBUG chart1.cpp:8881 SatsInView: 12
[... 324 lines cut ...]
20:45:50.159 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/local/lib/opencpn/libchartdldr_pi.so
20:45:50.159 MESSAGE chartdldr_pi.cpp:224 chartdldr_pi: DeInit
20:45:50.159 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/local/lib/opencpn/libwmm_pi.so
20:45:50.160 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/local/lib/opencpn/libgrib_pi.so
20:45:50.160 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libofc_pi.so
20:45:50.160 MESSAGE chartdb.cpp:326 Chart cache PlugIn purge
20:45:50.160 MESSAGE ofc_pi.cpp:443 ofc_pi: Shutdown_server
20:45:50.160 MESSAGE xtr1_inStream.cpp:817 ofc_pi: Could not open PUBLIC pipe
20:45:50.160 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libobjsearch_pi.so
20:45:50.465 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libwatchdog_pi.so
20:45:50.466 MESSAGE chart1.cpp:3878 opencpn::MyFrame exiting cleanly.
20:45:50.495 MESSAGE chartdb.cpp:326 Chart cache PlugIn purge
20:45:50.496 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /home/thomas/.local/lib/opencpn/liboesenc_pi.so
20:45:50.496 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/local/lib/opencpn/libdashboard_pi.so
20:45:50.496 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/local/lib/opencpn/libwmm_pi.so
20:45:50.496 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libofc_pi.so
20:45:50.496 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libwatchdog_pi.so
20:45:50.539 MESSAGE datastream.cpp:256 Closing NMEA Datastream Serial:/dev/ttyGPS
20:45:50.539 MESSAGE datastream.cpp:263 Stopping Secondary Thread
20:45:51.539 MESSAGE datastream.cpp:275 Stopped in 1 sec.
20:45:51.539 MESSAGE datastream.cpp:256 Closing NMEA Datastream Serial:/dev/ttyUSB0
20:45:51.539 MESSAGE GarminProtocolHandler.cpp:237 Stopping Garmin Serial thread
20:45:52.540 MESSAGE GarminProtocolHandler.cpp:251 Stopped in 1 sec.
20:45:52.573 MESSAGE chart1.cpp:2614 opencpn::MyApp starting exit.
20:45:52.574 MESSAGE chart1.cpp:2651 LOGBOOK:  2020-07-02 18:45:52 UTC OFF: Lat   56.09397 Lon   15.85747 COG ----- SOG -----

leamas, 2020-07-02:
@tai-fun: You really need to change to GRMN/GRMN. My use it the following way:
  - Normally I use NMEA/NMEA; indeed, this is the only thing which works in normal mode.
  - When I should upload routes I change my garmin to GRMN/GRMN
  - After this, I upload the route.
  - Andf them, resets the garmin to GRMN/GRMN

Yes, it ugly. But it's the only way which works since garmin does not support uploading via NMEA. Nor does it support setting the protocol over the communication protocol, it must be done manually.

So, please just put it in GRMN/GRMN mode, never mind that nothing seems to work, Then try to make an upload and submit the log.

leamas, 2020-07-02:
@bdcat

Looking at the threads in a breakpoint just before calling Garmin_GPS_Init() I see this using `thread apply all bt` in gdb:

<pre>Thread 11 (Thread 0x7fffcbfff700 (LWP 23778)):
#0  0x00007ffff582f116 in pselect () at /lib64/libc.so.6
#1  0x0000000000a2a63f in serial::Serial::SerialImpl::waitReadable(unsigned int) (this=0x3b89890, timeout=249) at /home/mk/src/OpenCPN/libs/serial/src/impl/unix.cc:502
#2  0x0000000000a2a991 in serial::Serial::SerialImpl::read(unsigned char*, unsigned long) (this=0x3b89890, buf=0x7fffcbffe44f &quot;&quot;, size=1) at /home/mk/src/OpenCPN/libs/serial/src/impl/unix.cc:565
#3  0x0000000000a2704b in serial::Serial::read(unsigned char*, unsigned long) (this=0x3b36e30, buffer=0x7fffcbffe44f &quot;&quot;, size=1) at /home/mk/src/OpenCPN/libs/serial/src/serial.cc:129
#4  0x0000000000807c36 in OCP_DataStreamInput_Thread::Entry() (this=0x3b36e10) at /home/mk/src/OpenCPN/src/OCP_DataStreamInput_Thread.cpp:424
#5  0x00007ffff708bcda in wxThread::CallEntry() () at /lib64/libwx_baseu-3.0.so.0
#6  0x00007ffff708c057 in  () at /lib64/libwx_baseu-3.0.so.0
#7  0x00007ffff5e654c0 in start_thread () at /lib64/libpthread.so.0
#8  0x00007ffff5837553 in clone () at /lib64/libc.so.6
</pre>

Does  this look like a thread ready to steal a byte from the main thread (which is thread 1)? 

Thoughts?

leamas, 2020-07-02:
(con'd...)
The patch I use to disable the serial thread is
```

--- a/src/GarminProtocolHandler.cpp
+++ b/src/GarminProtocolHandler.cpp
@@ -208,7 +208,7 @@ GarminProtocolHandler::GarminProtocolHandler(DataStream *parent, wxEvtHandler *M
         // Start handler thread
         m_garmin_serial_thread = new GARMIN_Serial_Thread(this, m_pparent, m_pMainEventHandler, m_port );
         m_Thread_run_flag = 1;
-        m_garmin_serial_thread->Run();
+      //m_garmin_serial_thread->Run();
     }
```

Isn't this enough, or are there other threads to cope with?

tai-fun, 2020-07-02:
OK, try this...

21:53:25.819   ERROR gps_wx_logging.h:14 GPS_Packet_Read: Timeout.  No data received.
21:53:25.819   ERROR gps_wx_logging.h:14 GPS_Get_Ack error: -7
21:53:25.819   DEBUG gps_wx_logging.h:38 Tx Data:
21:53:25.819   DEBUG gps_wx_logging.h:38 10 
21:53:25.819   DEBUG gps_wx_logging.h:38 fe 
21:53:25.819   DEBUG gps_wx_logging.h:38 00 
21:53:25.819   DEBUG gps_wx_logging.h:38 02 
21:53:25.819   DEBUG gps_wx_logging.h:38 10 
21:53:25.819   DEBUG gps_wx_logging.h:38 03 
21:53:25.819   DEBUG gps_wx_logging.h:38 : 
21:53:25.819   DEBUG gps_wx_logging.h:38 .
21:53:25.819   DEBUG gps_wx_logging.h:38 .
21:53:25.819   DEBUG gps_wx_logging.h:38 .
21:53:25.819   DEBUG gps_wx_logging.h:38 (UNKNOWN )

21:53:25.819   DEBUG gps_wx_logging.h:38 Rx Data:
21:53:25.922 MESSAGE chart1.cpp:6857 OnInitTimer...3
21:53:26.023 MESSAGE chart1.cpp:6857 OnInitTimer...4
21:53:26.085 MESSAGE ConfigMgr.cpp:1369 UseGarminHostUpload
21:53:26.085 MESSAGE ConfigMgr.cpp:1371 UseNMEA_GLL
21:53:26.760 MESSAGE options.cpp:5819 options: got device count: 11
21:53:26.805 MESSAGE chartdb.cpp:1346 Creating new chart
21:53:26.810 MESSAGE chartsymbols.cpp:1155 ChartSymbols loaded from /usr/local/share/opencpn/s57data/chartsymbols.xml
21:53:26.843 MESSAGE chart1.cpp:1318 Using s57data in /usr/local/share/opencpn/s57data
21:53:26.920 MESSAGE chartdb.cpp:1475 Initializing Chart /home/thomas/Sailing/Charts/Vector/oeSENC/sgl001CD480-SE-2020-25/OC-46-RX0IE5.oesenc
21:53:26.920 MESSAGE oesenc_pi.cpp:3170 Validation Path to oeserverd is: /home/thomas/.local/bin/oeserverd
21:53:26.920 MESSAGE oesenc_pi.cpp:3175 Validation Path to server test is: /home/thomas/.local/bin/oeserverd
21:53:26.920 MESSAGE oesenc_pi.cpp:3181 oesenc_pi: Checking oeserverd utility at {/home/thomas/.local/bin/oeserverd}
21:53:26.920 MESSAGE oesenc_pi.cpp:3222 oesenc_pi: starting oeserverd utility: /home/thomas/.local/bin/oeserverd
21:53:28.230 MESSAGE oesenc_pi.cpp:3294 oesenc_pi: oeserverd Check OK...LoopCount: 2
21:53:28.457 MESSAGE chartdb.cpp:1296 OpenChartUsingCache, NOT in cache:   cache size: 1

21:53:28.457 MESSAGE chartdb.cpp:1299    OpenChartUsingCache:  type 8  /home/thomas/Sailing/Charts/Vector/oeSENC/sgl001CD480-SE-2020-25/OC-46-RX0IE4.oesenc
21:53:28.457 MESSAGE chartdb.cpp:1346 Creating new chart
21:53:28.457 MESSAGE chartdb.cpp:1475 Initializing Chart /home/thomas/Sailing/Charts/Vector/oeSENC/sgl001CD480-SE-2020-25/OC-46-RX0IE4.oesenc
21:53:28.709 MESSAGE chartdb.cpp:1194 All chart in cache locked, size: 2
21:53:28.711 MESSAGE fpr.cpp:85 ls results:
21:53:28.750   ERROR gps_wx_logging.h:14 GPS_Packet_Read: Timeout.  No data received.
21:53:28.750   ERROR gps_wx_logging.h:14 GPS_Get_Ack error -7: 
21:53:29.143 MESSAGE fpr.cpp:98 oeaserverd results:
21:53:29.143 MESSAGE fpr.cpp:101 1
21:53:29.636 MESSAGE chart1.cpp:6857 OnInitTimer...5
21:53:29.737 MESSAGE chart1.cpp:6857 OnInitTimer...6
21:53:29.737 MESSAGE chart1.cpp:7152 OnInitTimer...Last Call
21:53:29.787 MESSAGE chartdb.cpp:1194 All chart in cache locked, size: 2
21:53:29.789 MESSAGE chart1.cpp:7170 OnInitTimer...Finalize Canvases
21:53:29.797   ERROR svgxml.cpp:444 File  doesn't exist.
21:53:29.797   ERROR svgxml.cpp:444 File  doesn't exist.
21:53:29.798   ERROR svgxml.cpp:444 File  doesn't exist.
21:53:29.798   ERROR svgxml.cpp:444 File  doesn't exist.
21:53:29.801   ERROR svgxml.cpp:444 File  doesn't exist.
21:53:29.801   ERROR svgxml.cpp:444 File  doesn't exist.
21:53:31.005   DEBUG gps_wx_logging.h:38 Tx Data:
21:53:31.005   DEBUG gps_wx_logging.h:38 10 
21:53:31.005   DEBUG gps_wx_logging.h:38 fe 
21:53:31.005   DEBUG gps_wx_logging.h:38 00 
21:53:31.005   DEBUG gps_wx_logging.h:38 02 
21:53:31.005   DEBUG gps_wx_logging.h:38 10 
21:53:31.005   DEBUG gps_wx_logging.h:38 03 
21:53:31.005   DEBUG gps_wx_logging.h:38 : 
21:53:31.005   DEBUG gps_wx_logging.h:38 .
21:53:31.005   DEBUG gps_wx_logging.h:38 .
21:53:31.005   DEBUG gps_wx_logging.h:38 .
21:53:31.005   DEBUG gps_wx_logging.h:38 (UNKNOWN )

21:53:31.005   DEBUG gps_wx_logging.h:38 Rx Data:
21:53:32.006   ERROR gps_wx_logging.h:14 GPS_Packet_Read: Timeout.  No data received.
21:53:32.006   ERROR gps_wx_logging.h:14 GPS_Get_Ack error: -7
[... 363 lines cut ...]
21:54:34.473 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/local/lib/opencpn/libchartdldr_pi.so
21:54:34.473 MESSAGE chartdldr_pi.cpp:224 chartdldr_pi: DeInit
21:54:34.473 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/local/lib/opencpn/libwmm_pi.so
21:54:34.473 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/local/lib/opencpn/libgrib_pi.so
21:54:34.473 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libofc_pi.so
21:54:34.473 MESSAGE chartdb.cpp:326 Chart cache PlugIn purge
21:54:34.473 MESSAGE ofc_pi.cpp:443 ofc_pi: Shutdown_server
21:54:34.473 MESSAGE xtr1_inStream.cpp:817 ofc_pi: Could not open PUBLIC pipe
21:54:34.473 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libobjsearch_pi.so
21:54:34.570 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libwatchdog_pi.so
21:54:34.572 MESSAGE chart1.cpp:3878 opencpn::MyFrame exiting cleanly.
21:54:34.605 MESSAGE chartdb.cpp:326 Chart cache PlugIn purge
21:54:34.605 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /home/thomas/.local/lib/opencpn/liboesenc_pi.so
[... cut ...]
21:54:35.688 MESSAGE chart1.cpp:2614 opencpn::MyApp starting exit.
21:54:35.689 MESSAGE chart1.cpp:2651 LOGBOOK:  2020-07-02 19:54:35 UTC OFF: Lat   56.09400 Lon   15.85750
21:54:35.704 MESSAGE TCDataSource.cpp:43 UnLoading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONIC.IDX
21:54:35.706 MESSAGE TCDataSource.cpp:43 UnLoading Tide/Current data source: /home/thomas/Sailing/Navigation/Tides/TIDALfiles/Harmonic_France/HARMONIC.IDX
21:54:35.806 MESSAGE chart1.cpp:2674 opencpn::MyApp exiting cleanly...

bdbcat, 2020-07-03:
@leamas 
This old code is riddled with bugs.  Let me start:
1.  For normal position reporting, GRMN/GRMN mode does not work.  Indeed, it cannot even be persistently selected in the connections setup.  A bug. I will investigate.
2.  This means that the GarminProtocolHandler and friends never runs.  The port is always setup as a standard serial NMEA port, as you know and use it.
3.  So, your thread trace showing and open select() call to the port is problematic.  What is supposed to happen for route upload is this:
a,  Stop the existing standard serial NMEA port.  This is done by:
```
int Multiplexer::SendRouteToGPS(...
.
.
.
    DataStream *old_stream = FindStream( com_name );
    if( old_stream ) {
        SaveStreamProperties( old_stream );
        StopAndRemoveStream( old_stream );
    }
```
This should produce log messages like 
```
           wxLogMessage(_T("Stopping Secondary Thread"));
           msg.Printf(_T("Stopped in %d sec."), 10 - tsec);
``` 
But you mentioned earlier that you expected these messages, and did not see them.
That is suspicious.

So, what needs to happen is to debug step through the FindStream( com_name ); and then the StopAndRemoveStream( old_stream );  calls.
If for some reason the stream was not found in the FindStream() call, then it would not be stopped, and so we could get a collision.

But before doing all that, please simply disable the port in the connections dialog list.  The use "Send to new GPS" option. That will initialize the port specifically for GRMN upload, and there will be no collision possible.

leamas, 2020-07-03:
After a night's  sleep: with patch below it works. Ergo: this is about thread synchronization, in particular the three threads accessing the serial device.

Current solution seems to be to try to stop the secondary threads when accessing the device in the main thread. Obviously, this is error prone. Possible solutions includes trying to fix the current logic or to use a  mutex semaphore guarding access to the device.

In a longer perspective it would probably be better to migrate from wxWidgets threads to standard C++ threads, but IMHO not for 5.2 if it can be avoided

```
diff --git a/src/GarminProtocolHandler.cpp b/src/GarminProtocolHandler.cpp
index 775844752..c809e7325 100644
--- a/src/GarminProtocolHandler.cpp
+++ b/src/GarminProtocolHandler.cpp
@@ -208,7 +208,7 @@ GarminProtocolHandler::GarminProtocolHandler(DataStream *parent, wxEvtHandler *M
         // Start handler thread
         m_garmin_serial_thread = new GARMIN_Serial_Thread(this, m_pparent, m_pMainEventHandler, m_port );
         m_Thread_run_flag = 1;
-        m_garmin_serial_thread->Run();
+//      m_garmin_serial_thread->Run();
     }
 
     TimerGarmin1.SetOwner(this, TIMER_GARMIN1);
diff --git a/src/OCP_DataStreamInput_Thread.cpp b/src/OCP_DataStreamInput_Thread.cpp
index 4d4bde7de..cba60d59b 100755
--- a/src/OCP_DataStreamInput_Thread.cpp
+++ b/src/OCP_DataStreamInput_Thread.cpp
@@ -396,6 +396,7 @@ void *OCP_DataStreamInput_Thread::Entry()
     bool not_done = true;
     bool nl_found = false;
     wxString msg;
+return 0;
     
     
     //    Request the com port from the comm manager
diff --git a/src/garmin_wrapper.cpp b/src/garmin_wrapper.cpp
index a5a1fc117..c6959e243 100644
--- a/src/garmin_wrapper.cpp
+++ b/src/garmin_wrapper.cpp
@@ -329,7 +329,7 @@ int Garmin_GPS_SendRoute( const wxString &port_name, Route *pr, wxGauge *pProgre
             {
                   int rv = OCPNMessageBox(NULL, _("Overwrite Garmin device route number 1?"),
                                           _("OpenCPN Message"), wxOK | wxCANCEL | wxICON_QUESTION);
-                  if(rv != wxOK)
+                  if(rv != wxID_OK)
                         return 0;
             }

```

leamas, 2020-07-03:
@bdbcat: sorry, I missed you reply above my last post (wrote while disconnected).

Anyway, seems that we are reaching a common conclusion that something is wrong in the thread synchronization. I'll try to make a try with a simple mutex; if it works we could just drop the complicated start/stop secondary threads logic. 

BTW:  The StopStream might not work. I havn't checked, but in order to work it must check the "exit thread" flag directly after select() and before read(); I'm not sure wxWidgets allows this type of granularity. Anyway, a mutex seems much safer, but the devil is as usual in the details.

leamas, 2020-07-03:
Following current code, with patch below I get the log `10:36:47.938   DEBUG multiplexer.cpp:397 locating old stream for Serial:/dev/ttyUSB0: Fail`

So, your suspicions are verified...

```

diff --git a/src/multiplexer.cpp b/src/multiplexer.cpp
index 8097f7c24..4b638ec3e 100644
--- a/src/multiplexer.cpp
+++ b/src/multiplexer.cpp
@@ -394,6 +394,8 @@ int Multiplexer::SendRouteToGPS(Route *pr,
 {
     int ret_val = 0;
     DataStream *old_stream = FindStream( com_name );
+    wxLogDebug("locating old stream for %s: %s",
+               com_name,  old_stream ? "OK" : "Fail");
     if( old_stream ) {
         SaveStreamProperties( old_stream );
         StopAndRemoveStream( old_stream );

```

leamas, 2020-07-03:
FindStream fails because the device I have entered as data stream device is /dev/nmea. This is a stable, udev-managed symlink which always points to the actual device (ttyUSB0 or ttyUSB1), see LINUX_DEVICES.md

If I manually enter the new device as` Serial:/dev/nmea` when uploading it works.

So, we have a working walk around. And a remaining problem to handle udev symlinks. The latter should be nothing worse than resolving any symlink entered as device.

A related problem is that the udev-generated symlink is not available as an option in the upload device download list.

leamas, 2020-07-03:
Closing this huge bug, referring to #1965 and #1964

[4 more replies]

closed 2020-07-03 by leamas