OpenCPN/OpenCPN #1960: Segmentation Fault when running OpenCPN

gromain, 2020-06-28:
Commit 116a629f8 introduced a bug where OpenCPN segfault on startup.

OpenCPN is compiled with GCC10, on an up to date Arch Linux installation.

# When compiling commit 116a629f8
The following error message on startup appears:
```
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
12:19:39: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1013,wx containers,compatible with 2.8),
and your program used 3.0 (wchar_t,compiler with C++ ABI 1014,wx containers,compatible with 2.8).
 SENC: nCPU: 12    m_max_jobs :11
[1]    72333 segmentation fault (core dumped)  ./opencpn
```

And in the log file:
```
12:19:41.236 MESSAGE chart1.cpp:1854 

________

12:19:41.237    INFO logger.cpp:169 ------- OpenCPN version 5.1.605 restarted at 2020-06-28 -------

12:19:41.237    INFO logger.cpp:169 Using loglevel INFO
12:19:41.237 MESSAGE chart1.cpp:1880 wxWidgets version: wxWidgets 3.0.5 Linux 64 bit wxGTK
12:19:41.247 MESSAGE chart1.cpp:1885 OS Version reports as:  5.7
12:19:41.247 MESSAGE chart1.cpp:1887 MemoryStatus:  mem_total: 31739 mb,  mem_initial: 785 mb
12:19:41.247 MESSAGE chart1.cpp:1896 SData_Locn is /usr/share/opencpn/
12:19:41.247 MESSAGE chart1.cpp:1912 PrivateDataDir is /home/rbazile/.opencpn
12:19:41.247 MESSAGE chart1.cpp:1969 Using existing Config_File: /home/rbazile/.opencpn/opencpn.conf
12:19:41.250 MESSAGE navutil.cpp:1195 winPluginDir, read from ini file: 
12:19:41.250 MESSAGE navutil.cpp:1248 Setting Viewpoint Lat/Lon 41.1323, -36.0232
12:19:41.250 MESSAGE navutil.cpp:1288 Setting Ownship Lat/Lon 45.5669, -1.0613
12:19:41.250 MESSAGE navutil.cpp:3151 Switch InlandEcdis mode Off
12:19:41.251 MESSAGE styles.cpp:854 Styles loading from /usr/share/opencpn/uidata/styles.xml
12:19:41.253 MESSAGE styles.cpp:831 No styles found at: /home/rbazile/
12:19:41.253 MESSAGE styles.cpp:831 No styles found at: /home/rbazile/.opencpn/
12:19:41.263 MESSAGE styles.cpp:786 Using SVG Icons
12:19:41.263 MESSAGE chart1.cpp:2027 Detected display size (horizontal): 508 mm
12:19:41.263 MESSAGE OCPNPlatform.cpp:905 Getting DefaultSystemLocale...
12:19:41.263 MESSAGE chart1.cpp:2064 System default Language:  fr_FR
12:19:41.263 MESSAGE chart1.cpp:2067 Config file language:  en_US
12:19:41.263 MESSAGE chart1.cpp:2072 Adjusted App language:  en_US
12:19:41.263 MESSAGE OCPNPlatform.cpp:990 ChangeLocale: Language load for:  en_US
12:19:41.264 MESSAGE OCPNPlatform.cpp:1042 ChangeLocale: Locale Init OK for:  en_US
12:19:41.264 MESSAGE OCPNPlatform.cpp:1061 Loading catalog for opencpn core.
12:19:41.264 MESSAGE OCPNPlatform.cpp:905 Getting DefaultSystemLocale...
12:19:41.264 MESSAGE chart1.cpp:2080 Opencpn language set to:  en_US
12:19:41.264 MESSAGE ConfigMgr.cpp:686 Loading Configs catalog: /home/rbazile/.opencpn/Configs/configs.xml
12:19:41.277 MESSAGE chart1.cpp:2302 Creating MyFrame...size(1344, 735)  position(0, 60)
12:19:41.277 MESSAGE chart1.cpp:2874  **** Connect stuff
12:19:41.390 MESSAGE chcanv.cpp:971 Creating glChartCanvas
12:19:41.409 MESSAGE chcanv.cpp:2350 Metrics:  m_display_size_mm: 508     g_Platform->getDisplaySize():  1920:1080   
12:19:41.409 MESSAGE chcanv.cpp:2355 wxDisplaySize(): 1920 1080
[... 430 lines cut ...]
12:21:14.513 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/lib64/opencpn/libchartdldr_pi.so
12:21:14.514 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/lib64/opencpn/libgrib_pi.so
12:21:14.514 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/lib64/opencpn/libclimatology_pi.so
12:21:14.514 MESSAGE pluginmanager.cpp:1630 PlugInManager: Deactivating PlugIn: /usr/lib64/opencpn/libweatherfax_pi.so
12:21:14.514 MESSAGE datastream.cpp:256 Closing NMEA Datastream Serial:/dev/ttyUSB0
12:21:14.514 MESSAGE datastream.cpp:263 Stopping Secondary Thread
12:21:15.516 MESSAGE datastream.cpp:275 Stopped in 1 sec.
12:21:15.542 MESSAGE chart1.cpp:2611 opencpn::MyApp starting exit.
12:21:15.542 MESSAGE chart1.cpp:2648 LOGBOOK:  2020-06-28 10:21:15 UTC OFF: Lat   45.56690 Lon   -1.06130
12:21:15.546 MESSAGE TCDataSource.cpp:43 UnLoading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONIC.IDX
12:21:15.547 MESSAGE TCDataSource.cpp:43 UnLoading Tide/Current data source: /home/rbazile/Documents/projets/Nautique/cartes numeriques/Tides/harmonics 10/HARMONIC.idx
12:21:15.549 MESSAGE chart1.cpp:2671 opencpn::MyApp exiting cleanly...
```

Exact steps followed to build:
```
git clone https://github.com/OpenCPN/OpenCPN.git
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr \
        -DOCPN_BUNDLE_GSHHS=CRUDE -DOCPN_BUNDLE_TCDATA=ON -DOCPN_BUNDLE_DOCS=ON \
        -DwxWidgets_CONFIG_EXECUTABLE=/usr/bin/wx-config-gtk3 \
        -DOCPN_FORCE_GTK3=ON ../
make -j12
```

bdbcat, 2020-06-28:
1.  You probably need to address this:
12:13:27: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1013,wx containers,compatible with 2.8),
and your program used 3.0 (wchar_t,compiler with C++ ABI 1014,wx containers,compatible with 2.8).

2.  Can you get us a gdb stack trace on the segfault?
Thanks

gromain, 2020-06-28:
I'll do the stack trace.

The warning comes from the fact that wxGTK has not been recompiled with gcc10 and updated yet in the arch linux repository.

gromain, 2020-06-28:
Here is the GDB backtrace on commit 116a629f8

```
rbazile@rbazile-xps:$ gdb -q ./opencpn
Reading symbols from ./opencpn...
(gdb) run
Starting program: /home/rbazile/Documents/codes/opencpn/opencpn/build/opencpn 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
[New Thread 0x7fffec608700 (LWP 117542)]
[Detaching after vfork from child process 117543]
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
[Thread 0x7fffec608700 (LWP 117542) exited]
[New Thread 0x7fffec608700 (LWP 117544)]
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
[Thread 0x7fffec608700 (LWP 117544) exited]
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
[New Thread 0x7fffec608700 (LWP 117545)]
[New Thread 0x7fffebd29700 (LWP 117546)]
[Thread 0x7fffebd29700 (LWP 117546) exited]
[Thread 0x7fffec608700 (LWP 117545) exited]
[New Thread 0x7fffec608700 (LWP 117547)]
[New Thread 0x7fffebd29700 (LWP 117548)]
[Thread 0x7fffebd29700 (LWP 117548) exited]
[Thread 0x7fffec608700 (LWP 117547) exited]
[New Thread 0x7fffec608700 (LWP 117549)]
[Thread 0x7fffec608700 (LWP 117549) exited]
[New Thread 0x7fffec608700 (LWP 117550)]
[Thread 0x7fffec608700 (LWP 117550) exited]
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
[New Thread 0x7fffec608700 (LWP 117551)]
[Thread 0x7fffec608700 (LWP 117551) exited]
[New Thread 0x7fffec608700 (LWP 117552)]
[Thread 0x7fffec608700 (LWP 117552) exited]
[New Thread 0x7fffec608700 (LWP 117553)]
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
[Thread 0x7fffec608700 (LWP 117553) exited]
18:29:35: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1013,wx containers,compatible with 2.8),
and your program used 3.0 (wchar_t,compiler with C++ ABI 1014,wx containers,compatible with 2.8).
[New Thread 0x7fffec608700 (LWP 117554)]
[New Thread 0x7fffebd29700 (LWP 117555)]
[Detaching after vfork from child process 117556]
[Detaching after vfork from child process 117558]
[Detaching after vfork from child process 117560]
[Detaching after vfork from child process 117562]
[Detaching after vfork from child process 117564]
[Detaching after vfork from child process 117568]
[Detaching after vfork from child process 117572]
[Detaching after vfork from child process 117576]
[Detaching after vfork from child process 117583]
[Detaching after vfork from child process 117585]
[New Thread 0x7fffeaa47700 (LWP 117587)]
[New Thread 0x7fffe8b0b700 (LWP 117588)]
[New Thread 0x7fffe830a700 (LWP 117589)]
[New Thread 0x7fffe7b09700 (LWP 117590)]
[New Thread 0x7fffe7308700 (LWP 117591)]
[New Thread 0x7fffe6b07700 (LWP 117592)]
[... 8 lines cut ...]
#3  0x00007ffff70f54c1 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#4  0x00007ffff70f55c4 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () at /usr/lib/libwx_baseu-3.0.so.0
#5  0x00007ffff70f590e in wxEvtHandler::TryHereOnly(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#6  0x00007ffff70f5994 in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#7  0x00007ffff70f5a32 in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#8  0x00007ffff70f57c7 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#9  0x00007ffff745aa5a in wxWindow::GTKSendPaintEvents(_cairo*) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#10 0x00007ffff7bcfd9d in wxGLCanvas::OnInternalIdle() () at /usr/lib/libwx_gtk3u_gl-3.0.so.0
#11 0x00007ffff76154d3 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#12 0x00007ffff7615508 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#13 0x00007ffff7615508 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#14 0x00007ffff74a5340 in wxFrame::SendIdleEvents(wxIdleEvent&) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#15 0x00007ffff74e5b4e in wxAppBase::ProcessIdle() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#16 0x00007ffff7413306 in wxApp::DoIdle() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#17 0x00007ffff7413424 in  () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#18 0x00007ffff5e06340 in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#19 0x00007ffff5e53fa9 in  () at /usr/lib/libglib-2.0.so.0
#20 0x00007ffff5e04c03 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#21 0x00007ffff673ac0f in gtk_main () at /usr/lib/libgtk-3.so.0
#22 0x00007ffff7430856 in wxGUIEventLoop::DoRun() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#23 0x00007ffff6fabe0e in wxEventLoopBase::Run() () at /usr/lib/libwx_baseu-3.0.so.0
#24 0x00007ffff6f70f27 in wxAppConsoleBase::MainLoop() () at /usr/lib/libwx_baseu-3.0.so.0
#25 0x00007ffff6fff7f9 in wxEntry(int&, wchar_t**) () at /usr/lib/libwx_baseu-3.0.so.0
#26 0x0000555555794f48 in main ()
```

gromain, 2020-06-28:
Backtrace on commit ffa997b3c:
```
rbazile@rbazile-xps: $ gdb -q ./opencpn                                              
Reading symbols from ./opencpn...
(gdb) run
Starting program: /home/rbazile/Documents/codes/opencpn/opencpn/build/opencpn 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
[New Thread 0x7fffec608700 (LWP 129904)]
[Detaching after vfork from child process 129905]
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
[Thread 0x7fffec608700 (LWP 129904) exited]
[New Thread 0x7fffec608700 (LWP 129906)]
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
[Thread 0x7fffec608700 (LWP 129906) exited]
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
[New Thread 0x7fffec608700 (LWP 129907)]
[New Thread 0x7fffebd29700 (LWP 129908)]
[Thread 0x7fffebd29700 (LWP 129908) exited]
[Thread 0x7fffec608700 (LWP 129907) exited]
[New Thread 0x7fffec608700 (LWP 129909)]
[New Thread 0x7fffebd29700 (LWP 129910)]
[Thread 0x7fffebd29700 (LWP 129910) exited]
[Thread 0x7fffec608700 (LWP 129909) exited]
[New Thread 0x7fffec608700 (LWP 129911)]
[Thread 0x7fffec608700 (LWP 129911) exited]
[New Thread 0x7fffec608700 (LWP 129912)]
[Thread 0x7fffec608700 (LWP 129912) exited]
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
[New Thread 0x7fffec608700 (LWP 129913)]
[Thread 0x7fffec608700 (LWP 129913) exited]
[New Thread 0x7fffec608700 (LWP 129914)]
[Thread 0x7fffec608700 (LWP 129914) exited]
[New Thread 0x7fffec608700 (LWP 129915)]
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
[Thread 0x7fffec608700 (LWP 129915) exited]
21:45:29: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1013,wx containers,compatible with 2.8),
and your program used 3.0 (wchar_t,compiler with C++ ABI 1014,wx containers,compatible with 2.8).
[New Thread 0x7fffec608700 (LWP 129916)]
[New Thread 0x7fffebd29700 (LWP 129917)]
[Detaching after vfork from child process 129918]
[Detaching after vfork from child process 129920]
[Detaching after vfork from child process 129922]
[Detaching after vfork from child process 129924]
[Detaching after vfork from child process 129926]
[Detaching after vfork from child process 129930]
[Detaching after vfork from child process 129934]
[Detaching after vfork from child process 129938]
[New Thread 0x7fffea211700 (LWP 129945)]
[New Thread 0x7fffe5a10700 (LWP 129946)]
[Detaching after vfork from child process 129959]
[Detaching after vfork from child process 129961]
[New Thread 0x7fffe4e96700 (LWP 129963)]
[New Thread 0x7fffc160a700 (LWP 129964)]
[New Thread 0x7fffc0e09700 (LWP 129965)]
[New Thread 0x7fffb3fff700 (LWP 129966)]
 SENC: nCPU: 12    m_max_jobs :11
[... 11 lines cut ...]
#5  0x00007ffff70f590e in wxEvtHandler::TryHereOnly(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#6  0x00007ffff70f5994 in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#7  0x00007ffff70f5a32 in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#8  0x00007ffff70f57c7 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#9  0x00007ffff745aa5a in wxWindow::GTKSendPaintEvents(_cairo*) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#10 0x00007ffff7bcfd9d in wxGLCanvas::OnInternalIdle() () at /usr/lib/libwx_gtk3u_gl-3.0.so.0
#11 0x00007ffff76154d3 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#12 0x00007ffff7615508 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#13 0x00007ffff7615508 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#14 0x00007ffff74a5340 in wxFrame::SendIdleEvents(wxIdleEvent&) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#15 0x00007ffff74e5b4e in wxAppBase::ProcessIdle() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#16 0x00007ffff7413306 in wxApp::DoIdle() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#17 0x00007ffff7413424 in  () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#18 0x00007ffff5e06340 in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#19 0x00007ffff5e53fa9 in  () at /usr/lib/libglib-2.0.so.0
#20 0x00007ffff5e04c03 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#21 0x00007ffff673ac0f in gtk_main () at /usr/lib/libgtk-3.so.0
#22 0x00007ffff7430856 in wxGUIEventLoop::DoRun() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#23 0x00007ffff6fabe0e in wxEventLoopBase::Run() () at /usr/lib/libwx_baseu-3.0.so.0
#24 0x00007ffff749a94f in wxDialog::ShowModal() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#25 0x0000555555871038 in ShowNavWarning() ()
#26 0x000055555587e59c in MyApp::OnInit() ()
#27 0x00007ffff6fff822 in wxEntry(int&, wchar_t**) () at /usr/lib/libwx_baseu-3.0.so.0
#28 0x00005555557937b8 in main ()
```

bdbcat, 2020-06-28:
Some help...
But could you rebuild in DEBUG mode (-o0) to get full symbol availability. We will then get line numbers....

btw, works fine on Ubuntu, various flavors....

Thanks
Dave

gromain, 2020-06-29:
Sorry, I forgot about the debug flags!
I compiled with 
```
cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX=/usr \
        -DOCPN_BUNDLE_GSHHS=CRUDE -DOCPN_BUNDLE_TCDATA=ON -DOCPN_BUNDLE_DOCS=ON \
        -DwxWidgets_CONFIG_EXECUTABLE=/usr/bin/wx-config-gtk3 \
        -DOCPN_FORCE_GTK3=ON ../
```

And the backtrace is:
```
 rbazile@rbazile-xps: $ gdb -q ./opencpn
Reading symbols from ./opencpn...
(gdb) run
Starting program: /home/rbazile/Documents/codes/opencpn/opencpn/build/opencpn 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
[New Thread 0x7fffec608700 (LWP 6829)]
[Detaching after vfork from child process 6830]
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
[Thread 0x7fffec608700 (LWP 6829) exited]
[New Thread 0x7fffec608700 (LWP 6831)]
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
[Thread 0x7fffec608700 (LWP 6831) exited]
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
[New Thread 0x7fffec608700 (LWP 6832)]
[New Thread 0x7fffebd29700 (LWP 6833)]
[Thread 0x7fffebd29700 (LWP 6833) exited]
[Thread 0x7fffec608700 (LWP 6832) exited]
[New Thread 0x7fffec608700 (LWP 6838)]
[New Thread 0x7fffebd29700 (LWP 6839)]
[Thread 0x7fffebd29700 (LWP 6839) exited]
[Thread 0x7fffec608700 (LWP 6838) exited]
[New Thread 0x7fffec608700 (LWP 6840)]
[Thread 0x7fffec608700 (LWP 6840) exited]
[New Thread 0x7fffec608700 (LWP 6841)]
[Thread 0x7fffec608700 (LWP 6841) exited]
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
[New Thread 0x7fffec608700 (LWP 6842)]
[Thread 0x7fffec608700 (LWP 6842) exited]
[New Thread 0x7fffec608700 (LWP 6843)]
[Thread 0x7fffec608700 (LWP 6843) exited]
[New Thread 0x7fffec608700 (LWP 6844)]
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
[Thread 0x7fffec608700 (LWP 6844) exited]
01:55:06: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1013,wx containers,compatible with 2.8),
and your program used 3.0 (wchar_t,compiler with C++ ABI 1014,wx containers,compatible with 2.8).
[New Thread 0x7fffec608700 (LWP 6845)]
[New Thread 0x7fffebd29700 (LWP 6846)]
[Detaching after vfork from child process 6853]
[Detaching after vfork from child process 6855]
[Detaching after vfork from child process 6857]
[Detaching after vfork from child process 6859]
[Detaching after vfork from child process 6861]
[Detaching after vfork from child process 6865]
[Detaching after vfork from child process 6869]
[Detaching after vfork from child process 6873]
[... 29 lines cut ...]
#9  0x00007ffff70f5994 in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#10 0x00007ffff70f5a32 in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#11 0x00007ffff70f57c7 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#12 0x00007ffff745aa5a in wxWindow::GTKSendPaintEvents(_cairo*) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#13 0x00007ffff7bcfd9d in wxGLCanvas::OnInternalIdle() () at /usr/lib/libwx_gtk3u_gl-3.0.so.0
#14 0x00007ffff76154d3 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#15 0x00007ffff7615508 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#16 0x00007ffff7615508 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#17 0x00007ffff74a5340 in wxFrame::SendIdleEvents(wxIdleEvent&) () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#18 0x00007ffff74e5b4e in wxAppBase::ProcessIdle() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#19 0x00007ffff7413306 in wxApp::DoIdle() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#20 0x00007ffff7413424 in  () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#21 0x00007ffff5e06340 in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#22 0x00007ffff5e53fa9 in  () at /usr/lib/libglib-2.0.so.0
#23 0x00007ffff5e04c03 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#24 0x00007ffff673ac0f in gtk_main () at /usr/lib/libgtk-3.so.0
#25 0x00007ffff7430856 in wxGUIEventLoop::DoRun() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#26 0x00007ffff6fabe0e in wxEventLoopBase::Run() () at /usr/lib/libwx_baseu-3.0.so.0
#27 0x00007ffff749a94f in wxDialog::ShowModal() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#28 0x00005555558cf13e in ShowNavWarning() () at /home/rbazile/Documents/codes/opencpn/opencpn/src/chart1.cpp:1052
#29 0x00005555558d6e6c in MyApp::OnInit() (this=0x555556166900) at /home/rbazile/Documents/codes/opencpn/opencpn/src/chart1.cpp:2593
#30 0x0000555555906c7d in wxAppConsoleBase::CallOnInit() (this=0x555556166900) at /usr/include/wx-3.0/wx/app.h:93
#31 0x00007ffff6fff822 in wxEntry(int&, wchar_t**) () at /usr/lib/libwx_baseu-3.0.so.0
#32 0x00005555558cf501 in main(int, char**) (argc=1, argv=0x7fffffffd7e8) at /home/rbazile/Documents/codes/opencpn/opencpn/src/chart1.cpp:1097
```

bdbcat, 2020-06-29:
Corrected the fault.
Root cause is failure to notice that one of the MBTiles loaded on startup is corrupt.

```
12:19:41.738 MESSAGE chartdb.cpp:1477 Initializing Chart /home/rbazile/Documents/projets/Nautique/cartes numeriques/OSM/OSM-OpenCPN2-NorthernAtlantic.mbtiles
12:19:41.739 MESSAGE mbtiles.cpp:528 mbtiles exception: unable to open database file
12:19:41.739 MESSAGE chartdb.cpp:1537 Problem initializing Chart /home/rbazile/Documents/projets/Nautique/cartes numeriques/OSM/OSM-OpenCPN2-NorthernAtlantic.mbtiles
12:19:41.739 MESSAGE chartdb.cpp:1557    OpenChartFromStack... Error opening chart /home/rbazile/Documents/projets/Nautique/cartes numeriques/OSM/OSM-OpenCPN2-NorthernAtlantic.mbtiles ... return code 2

Solution in github now.
Thanks for the catch.
```

gromain, 2020-06-29:
Thanks for the quick fix.

This is solved for me in master.

I'll report back if anything related happens, but I'm confident we're good.

closed 2020-06-29 by gromain