OpenCPN/OpenCPN #1961: m_fixtime does not get updated when GPS times out

transmitterdan, 2020-06-28:
When ```gGPS_Watchdog``` trips (becomes < 0) then ```m_fixtime``` does not update.  It stays at the previous time.  If there is no GPS connected then m_fixtime is initialized to 0 which is the 1970 epoch.  This makes plugins that rely on ```m_fixtime``` do wrong things.  Proposed solution:

In chart1.cpp at about line 7404 where the ```gGPS_Watchdog``` counter is evaluated as timed out, add this line:

```
m_fixtime = wxDateTime::Now().GetTicks();
```

transmitterdan, 2021-02-01:
No comments.  Am I the only one seeing this?

Hakansv, 2021-02-01:
TDan..
No the Dashboard satellite clock instrument is neither updated nor cleaned out but the old value remains after a time out.  Not nice!
Håkan