OpenCPN/OpenCPN #1962: Translation for managed plugins

Rasbats, 2020-06-29:
Repeat of the post here:  

[https://www.cruisersforum.com/forums/f134/opencpn-version-5-1-620-rc1-test-235885-4.html#post3173893](url)

Translations for the managed plugins are not appearing on my machine. 

[image]
[image]
[image]

Have checked with my local po files for 'sar_pi' and French translations are available, but they do not appear on the installed plugin. The '.mo' for the plugin is present in 'share/locale/fr_FR/LC_MESSAGES' folder.

bdbcat, 2020-06-30:
Mike...
Corrected in github now, CI builds available shortly as 5.1.621

Rasbats, 2020-06-30:
Thanks Dave. Your work on this is appreciated.

closed 2020-06-30 by Rasbats