OpenCPN/OpenCPN #1965: Connections: using the GRMN/GRMN protocol does not work (#1957)

leamas, 2020-07-03:
As described in https://github.com/OpenCPN/OpenCPN/issues/1957#issuecomment-653299252, using the garm,in protocol to retrieve  positions etc. does not work

bdbcat, 2020-07-04:
After refreshing my memory from working with this code years ago....

1.  To get position tracking from Garmin devices in GRMN (or Garmin) mode, the device must be capable of sending what are known as PVT packets.  This is distinct from the protocol used to upload routes and waypoints.
2.  Not all Garmin devices are PVT capable.  And there is no publicly available document which will tell us that.
3.  However, the JEEPS code contains logic to figure this out.  It passes a message when the GPS device is detected and  connected, but is found not capable of PVT transfers.
This is found in gpsapp.c, as "GPS_Warning("A001 protocol not supported");
4.  We should consider changing this message from a "Warning" to "User" message, i.e.
GPS_User("A001 protocol not supported");  Then we will at least get a log message always.
5.  We should further consider a GUI dialog notifying the user that their GPS setup will not work for position tracking as configured.

Test notes:
1.  The current code for linux actually works, with a PVT capable unit.  Occasionally the interface hangs at startup, and the GPS receiver unit must re rebooted.  However, the JEEPS drivers know how to persistently retry the interface on reboot.
2.  There is a way to use GPSBabel to verify that a device is PVT capable.
```
$ gpsbabel -T -i garmin -f /dev/ttyUSB0
Fri Jul 3 23:27:35 2020 35.091576N 76.632571W  -36.443451
Fri Jul 3 23:27:36 2020 35.091576N 76.632572W  -36.422546
Fri Jul 3 23:27:37 2020 35.091575N 76.632572W  -36.420914
Fri Jul 3 23:27:38 2020 35.091575N 76.632572W  -36.447460
F
```
This will work if:
a.  The device is PVT capable, and
b.  GPS fix is available.  **The device does not send PVT if no fix is available**.

This will produce:
```
[ERROR] GPS_Packet_Read: Timeout.  No data received.
GARMIN:Can't init /dev/ttyUSB0

```
if the unit is not PVT capable, or needs reboot.

3.  From experimentation my two available devices, Garmin GPS 38 _cannot_ do PVT.  Garmin eTrex Venture _can_ do PVT.

I'll make the log message change in source now.
@leamas 
I think the sleep() delays do no harm, and could be good for some environments.  Will you submit a PR for the sleep()'s that you have been testing?

leamas, 2020-07-04:
@bdbcat: Looking at  2bb99a7e1:  GPS_warning() is the same as wxLogWarning and GPS_User() is the same as wxLogMessage() as things are setup. Lowering the priority from warning to message means that there is less chance the message is displayed. However, since the default level displays also wxLogMessage it doesn't really make any difference unless `using opencpn --loglevel warning`.

However, warnings somewhat stands out in the log.  Perhaps you  want to revert  this commit.

I think it would be great to add the PVT keyword to the message, enhancing it to something like  _A001 protocol not supported (PVT messages not available)_ That would IMHO be a better starting point when handling problems.

leamas, 2020-07-04:
@bdbcat: Currently building a PR with the delays added. Will submit later.

I also have a more complete update of the jeeps dir pulled from gpsbabel master tip. Given how solid gpsbabel seems to be this might be a reasonable update. However, it's a bigger change probably best done in next cycle. Eventually, I will file a PR also for this.

bdbcat, 2020-07-04:
@leamas...
On the Log message:  My focus is on more casual (Windows) users, who will have no experience at starting OCPN with --loglevel Warning.  We can at least help them read their logfile and discover the problem.  wxLogMessage will always come out.  Of course, a GUI message box is optimal, just to clearly indicate the basic problem.

Will be interesting to look at the diff from recent JEEPS code.  We will learn something....

leamas, 2020-07-24:
Let's close this, basically as WONTFIX since the limitation is in the garmin devices rather than OpenCPN.

closed 2020-07-24 by leamas