OpenCPN/OpenCPN #1966: Nix environments and webview  and Plugin Manager

rgleason, 2020-07-03:
@bdbcat 
@leamas 
Petri has some observations about this. Would you comment?
There are some links here regarding this.
https://www.cruisersforum.com/forums/f134/dashboard-tactics-dash-t-226601.html#post3177907

bdbcat, 2020-07-03:
Rick..

Executive summary, please?

Thanks
Dave

rgleason, 2020-07-04:
I am probably not qualified to do this, but here goes.

When asked about incorporating the Plugin Manager into his plugin, Petri basically wrote that this was not really his interest, but  asked

`"on those "13 platforms", is the wxWebView working out of the box? That would be interesting information (read here about my findings on *nix platforms)
`
and he provided a link to one of his gitub documents  https://github.com/canne/dashboard_tactics_pi/releases/tag/v1.98.104
where there is a section title "RPI Users"  which describes his experiences and suggestions for gtk2 systems, or compile your own wxwidges, and libwx_gtk2u_webview_3.0.

Which also leads me to ask if anyone had tried  installing plugins in the three raspbian environments with any success?  

So those are the two questions.

leamas, 2020-07-04:
> "on those "13 platforms", is the wxWebView working out of the box?

No, just some. But opencpn works anyway, using it if it is available. The cmake setup probes for the wxwebview availability, and sets up the build according to the results.

closed 2020-07-04 by rgleason

rgleason, 2020-07-04:
Petri's response with additional information about Dashboard-Tactics webview dependency.

End of line version v1.5.111 has no webview.
v2.0 does have webview dependency so only gtk3.0 is supported , older gtk2 is not.

He notes "Good news for RPI is that there is a RPIOS/64 beta now out."