OpenCPN/OpenCPN #1967: Flatpak: missing dependency in plugins

Carstijn, 2020-07-04:
Hi All,

I am using the latest flatpak build on Fedora 32. I installed a few of the plugins via the pluginmanager. It seems the install was successful but my system is unable to load the plugins due to a missing  dependency.

This is what I see in the logs, _ZTV13TiXmlDocument is missing.
```
12:02:45.187 MESSAGE pluginmanager.cpp:1134 Found 2 candidates
12:02:45.188 MESSAGE pluginmanager.cpp:1141 Checking plugin candidate: /home/sander/.var/app/org.opencpn.OpenCPN/lib/opencpn/libweather_routing_pi.so
12:02:45.188 MESSAGE pluginmanager.cpp:2044 Plugin is compatible by elf library scan: true
12:02:45.188 MESSAGE pluginmanager.cpp:2152 PlugInManager: Loading PlugIn: /home/sander/.var/app/org.opencpn.OpenCPN/lib/opencpn/libweather_routing_pi.so
12:02:45.190   ERROR dlunix.cpp:162 /home/sander/.var/app/org.opencpn.OpenCPN/lib/opencpn/libweather_routing_pi.so: undefined symbol: _ZTV13TiXmlDocument
12:02:45.190 MESSAGE pluginmanager.cpp:2211    PlugInManager: Cannot load library: /home/sander/.var/app/org.opencpn.OpenCPN/lib/opencpn/libweather_routing_pi.so 
12:02:45.190 MESSAGE pluginmanager.cpp:1141 Checking plugin candidate: /home/sander/.var/app/org.opencpn.OpenCPN/lib/opencpn/libwatchdog_pi.so
12:02:45.190 MESSAGE pluginmanager.cpp:2044 Plugin is compatible by elf library scan: true
12:02:45.190 MESSAGE pluginmanager.cpp:2152 PlugInManager: Loading PlugIn: /home/sander/.var/app/org.opencpn.OpenCPN/lib/opencpn/libwatchdog_pi.so
12:02:45.192   ERROR dlunix.cpp:162 /home/sander/.var/app/org.opencpn.OpenCPN/lib/opencpn/libwatchdog_pi.so: undefined symbol: _ZTV13TiXmlDocument
12:02:45.193 MESSAGE pluginmanager.cpp:2211    PlugInManager: Cannot load library: /home/sander/.var/app/org.opencpn.OpenCPN/lib/opencpn/libwatchdog_pi.so 
```

leamas, 2020-07-04:
Looks like a plugin problem in these specific plugins

A fast look at google reveals that this seems to be about tinyxml. The basic question is if this plugin uses tinyxml, and if it in that case bundles the tinyxml sources in the yaml file.

The main opencpn code is mostly linked in private mode, so trying to import libraries from the main app usually doesn't  work.

@jongough: Thougts?

rgleason, 2020-07-04:
We made some changes in tinyxml sometime ago I believe, but they were not extensive. What version of the plugin is this?
There was a recent update to 1.13.11.0 this morrning but there were no changes too tiny xml that I am aware of.
https://github.com/rgleason/weather_routing_pi/commits/master

Note: wxrte has high cpu usage, with the correction of language files this version is better, but it is still high.

Carstijn, 2020-07-04:
I tested it with the latest version this morning, don't know if older versions worked or not.

rgleason, 2020-07-04:
What is the version number please.

Carstijn, 2020-07-04:
Sure, tested it with version 1.13.11.0 of the weatherrouting plugin

leamas, 2020-07-04:
@Carstijn: Is this only about these plugins? In particular, do you have similar problems with  oesenc (which is sort of a reference plugin)?

Carstijn, 2020-07-04:
@leamas oesenc is a whole other story, that worked till version 3.3.8 (I guess), after this version no updates where visible in the pluginmanager. Today I did a clean install, including clean configuration dir and now its not visible in the pluginmanager.

leamas, 2020-07-04:
@Carstijn: I just checked a  fresh flatpak re-install and I can see oesenc. Are you using the nightly builds or the beta?

Carstijn, 2020-07-04:
This is the build of last night 5.1.621+90f47de
[image]

leamas, 2020-07-04:
Right. I tested the beta at http://opencpn.duckdns.org/opencpn-beta/opencpn.flatpakref Does this work for you, i. e. is this a rpoblem in the nightly builds only?

leamas, 2020-07-04:
I can reproduce in the nigthly. This is a separate issue, could you please file a new bug on this?

rgleason, 2020-07-04:
Since Jon is going to the trouble of installing fedora32 and flatpak and  Since he is going to that trouble, is there definitely something amiss with watchdog and weather_routing and is it related to tinyxml?  I don't want him to spend needless time....

rgleason, 2020-07-05:
It seems that there are other issues, should Jon continue?

leamas, 2020-07-08:
@Carstijn : The invisible oesenc should be due to #1971, now fixed. 

I can reproduce your problem for the weather_routing plugin on Fedora 30. However, oesenc loads just fine so this seems to be about the plugin and not the platform.

@jongough: Thoughts?

rgleason, 2020-07-08:
Sorry I am not going to be able to fix this as I have no idea what to do.

leamas, 2020-07-08:
@rgleason : Are you the maintainer of this plugin (the version exposed in the plugin manager)?

leamas, 2020-07-08:
One more data point: I have 6 different plugins installed and enabled, but one weather_routing show this error.

It might be as simple as the plugin is using tinyxml. Flatpak runs in sandbox, so if tinyxml isn't available in this it will fail if weather_rouing is using it, possibly with the error described.

Carstijn, 2020-07-08:
As far as I can see this is related to:
- Watchdog
- WeatherRouting
- WeatherFax

As other plugins work its plugin related. For WeatherRouting_pi and watchdog_pi there is a pullrequest by @rgleason for "frontend2" which touches the tinyxml cmake stuff. Could this be a fix?

rgleason, 2020-07-08:
@Carstijn   No I don't believe it is related to this. What is the problem with tinyxml?  Often we just load it for windows.

Carstijn, 2020-07-08:
@rgleason as far as I understand: ithe plugins are looking for a shared library object "_ZTV13TiXmlDocument", as this apparently not provided by the flatpak base image is should be compiled in to the lib it self. (static linked)

leamas, 2020-07-08:
@rgleason:

>  What is the problem with tinyxml?

The logs says
`02:45.190   ERROR dlunix.cpp:162 /home/sander/.var/app/org.opencpn.OpenCPN/lib/opencpn/libweather_routing_pi.so: undefined symbol: _ZTV13TiXmlDocument
`
The missing symbol seems, according too google, to be a tinyxml reference.

Again: Are you the maintainer of this plugin? The one who is supposed to fix this (or just drop  the flatpal version)?

It might be helpful  if someone could link to the flatpak manifest  (the yaml file) for any of the failing plugins

leamas, 2020-07-08:
>  as far as I understand: ithe plugins are looking for a shared library object "_ZTV13TiXmlDocument", as this apparently not provided by the flatpak base image is should be compiled in to the lib it self. (static linked)

Almost. It is looking for a symbol with that name, living in some library. According to google it lives in the timyxml libraray.

rgleason, 2020-07-08:
So Alec,
There is no symbol by that name that I know of.
there is nothing in cmakelists.txt other than findtinyxml.cmake
https://github.com/rgleason/weather_routing_pi/blob/master/CMakeLists.txt

Here is what is in findtinyxml.cmake
https://github.com/rgleason/weather_routing_pi/blob/master/cmake/FindTinyXML.cmake

How would I include tinyxml in flatpak, build?

Is this missing object something like a very small icon.png or something? 
Weather_routing supports svg icons but some of Sean's plguins don't

I can just remove the pluggin from flatpak, right now, but you won''t learn anything more.

rgleason, 2020-07-08:
https://github.com/OpenCPN/plugins/pulls
These will remain off the flatpak list until someone comes up with a solution.

bdbcat, 2020-07-08:
Rick...
If you add me as collaborator to weather_routing, I'll see what I can do.
Dave

rgleason, 2020-07-08:
Thank you!  I have no idea what to look for nor do I even understand the problem.  
I added you to most of my forked repos of Sean's PI's, you had fixed the paths for weather_routing so you are also a collaboration on that too.

leamas, 2020-07-09:
Serems like a trivial error. In CMakeLIsts.txt we have

```
# added by rick
 SET (SRC_TINYXML
             src/tinyxml/tinystr.h
                         src/tinyxml/tinyxml.h
)
```
i. e., you have added the headers but not the implementation .c  or .cpp files  o the sources. The headers means that the compilation will run, but the missiing .c /.cpp files means that that linkage fails -- which is what we see in the error log.

OTOH, this seems messy. The header is already included in HDRS,  but there are no corresponding .c or .cpp files in SRC. So the error seems to begin already there. Adding the header once again will certainly not help, nor should it be required to make it compile.

The solution should be simple: pick the source files from main opencpn, in libs/tinyxml/src and add them to the project using git, Then add them to  the SRCS list and that should be it. Or add them to SRC_TINYXML, but you should the remove tinyxml.h from SRCS so you don't havit it in two places.

Note that you have  added the tinystr.h header as well. I suggest that you make eperiments just dropping it,  it should not be required (it is not used in main opencpn). In order to build without tinystr.h you should use the -DTIXML_USE_STL  compilation ffag.

rgleason, 2020-07-09:
Dave, Now what should I do?

leamas, 2020-07-09:
> Alec, also that would not solve weatherfax or watchdog.

Why?

rgleason, 2020-07-09:
"n libs/tinyxml/src and add them to the project using gi"
This might work, but sean has it set up differently, so I am going to try to fix that, however Dave is fixing it now.

rgleason, 2020-07-09:
Well, they don't have my half baked code,... I hope.

    Alec, also that would not solve weatherfax or watchdog.

Why?

rgleason, 2020-07-09:
`In order to build without tinystr.h you should use the -DTIXML_USE_STL compilation ffag.`
I have wondered what that was for. What does it stand for?  DTIXML?  what is STL?

bdbcat, 2020-07-09:
Rick...
Weather_routing fixed, OK now.
What's next?
Dave

rgleason, 2020-07-09:
Watchdog?
I willl push to plugins tomorrow mornging Thanks!

bdbcat, 2020-07-09:
Watchdog not correct yet.  I'll keep looking.

rgleason, 2020-07-09:
Alec and Dave, 
Last night I remembered a similar error on this plugin or another while implementing frontend2 into the plugin.
I had lots of troubles with Links and resorted to expanding the SET listings and got to one confiiguration that worked, but it has not be terribly satisfactory because it uses too much CPU and basically freezes the screen a bit whiile computing. When the language file problems got sorted it got a little better.

So this morning I have looked at this problem differently.  As Alec suggested I consolidated all tinyxml file listings into the SET(SRCS_tinyxml   including tinyxml.h .   Perhaps years ago Sean had that listing under SRCS and I did not see it. In this configuration I left David's addition to 
SET(SRC ${SRCS} ${SRC_LIBTESS2}  ${SRC_PLUGINGL} ${SRC_TINYXML})

Then I thought, why are we doing this wen we have at the bottom of the CMakeLists.txt  PluginXML.cmake which should handle tinyxml? 
INCLUDE("cmake/PluginJSON.cmake")
INCLUDE("cmake/PluginXML.cmake")    <----????
INCLUDE("cmake/PluginGL.cmake")

So I commented out the SET(SRCS_tinyxml  and removed it from the 
SET(SRC ${SRCS} ${SRC_LIBTESS2}  ${SRC_PLUGINGL})
I compiled it and it worked better in Windows.
Now I am wondering if I should remove the SET(plugingl   and SET(libtess because of the cmake/PLuginGL.cmake file?

rgleason, 2020-07-09:
I guess one of the questions is if this will work in flatpak?

We do have  [include/PluginJSON.cmake](https://github.com/rgleason/weather_routing_pi/blob/master/cmake/PluginJSON.cmake)

and I had previously commented out SET(pluginGL.
but there is this question about  SET(SRCS   and  listing             src/jsoncpp/json/json.h
Shoulld that stay right there?

rgleason, 2020-07-09:
I removed  the listing src/jsoncpp/json/json.h  in SET(SRCS  and there were no errors and it ran fine in Windows.
Perhaps Less is more.

rgleason, 2020-07-09:
Took out the SET(tess  and got these link errors, so it going back in now.  These errors are similar to a problem I am having with Cliimatology so perhaps that will help..

```
  georef.c
     Creating library C:/Users/Rick/Documents/GitHub/weather_routing_pi/build/Release/weather_routi
  ng_pi.lib and object C:/Users/Rick/Documents/GitHub/weather_routing_pi/build/Release/weather_rout
  ing_pi.exp
BoatDialog.obj : error LNK2019: unresolved external symbol _tessDeleteTess referenced in function "
private: virtual void __thiscall BoatDialog::OnPaintCrossOverChart(class wxPaintEvent &)" (?OnPaint
CrossOverChart@BoatDialog@@EAEXAAVwxPaintEvent@@@Z) [C:\Users\Rick\Documents\GitHub\weather_routing

```

rgleason, 2020-07-09:
Can you these versions in Flatpak Opencpn?

Dave's first fix of tinyxml  this is the tarball and the metadata.  Import the tarball of course
https://cloudsmith.io/~opencpn/repos/weather-routing-beta/packages/?q=1.13.11*flatpak

Subsequent changes too tinyxml and Set(files to improve performance
https://cloudsmith.io/~opencpn/repos/weather-routing-prod/packages/?q=1.13.12.0*flatpak

I've tried both in windows. They both work, but the second one is faster in performing routing and the screen pans whille routing much much more easily. It stilll uses up to 20% cpu but does not seem to hog my laptop.

rgleason, 2020-07-09:
I see what Dave has done adding [SRC_tinyxml  at line 464  ](https://github.com/rgleason/weatherfax_pi/commit/2d0ddf8430de36912e3a34cb8cbd645e66e984ce )of CMakeLists.txt 
The file to test in Cloudsmith[ is here ](https://cloudsmith.io/~opencpn/repos/weatherfax-beta/packages/?q=1.9.9.0*flatpak)

I had big linkage problems while converting over to frontend2 so I probably went a bit overboard with listing of files, expanding what Sean had quite a lot.   I am looking at weathefax now and would like to clea[n CMakeLists.txt ](https://github.com/rgleason/weatherfax_pi/blob/master/CMakeLists.txt) up and have some questions about the best way to handle the SET( files
1. SRC_audiofile(    Move all the header src/libaudiofile/ *.h  into here?
2. SRC_plugingl(    Move all the header src/plugigl//*.h into here?

Then I would like to build and test.

Because we have pluginGL.cmake at the bottom, I would then try removing SRC_plugingl and the related additon to SRC   (that should be) at line 464.   Then build and test

Because we have plugiinXML.cmake at the bottom, I would ten try removing SRC_tinyxml and related addition to SRC at Line 464.  Then build and test.

Depending on what works, I will provide another file to test?
What do you think?

rgleason, 2020-07-09:
For weatherfax_pi there are two versions to test:
[v1.9.11.0 with tnyxml + plugingl ](https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/?q=1.9.11.0*metadata) (most like bdbcat) but also included plugingl
or[
v1.9.12.0 used pluginGL + pluginXML only](https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/?q=1.9.12.0*metadata)

I've tested both on windows and they work.
I like the last one because it is using cmake/pluginXML.cmake and cmake/pluginGL.cmake and is more consistent in its structure.
Let me know which one works with flatpak please, and I will push this to plugins.

leamas, 2020-07-09:
I have submitted an altermative solution for squiddio in a PM to you. It is a cleaner solution which should work on all platforms. It should also be easier to apply to other plugins. Use it if you like.

Attaching the patch also here. Patch is against  current version of your squiddio fork (v3.7.0 / 984a5cc). It should work on all platforms and minimizes the overall fuzz in CMakeLIsts.txt by moving linkage, flags and includes to the subdirectory.

[0001-Refactor-tinyxml-to-separate-directory.patch.gz](https://github.com/OpenCPN/OpenCPN/files/4897482/0001-Refactor-tinyxml-to-separate-directory.patch.gz)

rgleason, 2020-07-09:
**Regarding watchdog**
"12:02:45.190 MESSAGE pluginmanager.cpp:1141 Checking plugin candidate: /home/sander/.var/app/org.opencpn.OpenCPN/lib/opencpn/libwatchdog_pi.so   12:02:45.190 MESSAGE pluginmanager.cpp:2044 Plugin is compatible by elf library scan: true"
12:02:45.192   **_ERROR dlunix.cpp:162 /home/sander/.var/app/org.opencpn.OpenCPN/lib/opencpn/libwatchdog_pi.so: undefined symbol: _ZTV13TiXmlDocument_**
12:02:45.193 MESSAGE pluginmanager.cpp:2211    PlugInManager: Cannot load library: /home/sander/.var/app/org.opencpn.OpenCPN/lib/opencpn/libwatchdog_pi.so 

This is a message in some file called dlunix.cpp at Line:162  saying undefined symbol  (which is xml)
I've looked at watchdog  cmakelists.txt and it looks pretty clean to me
Also bdbcat added xml and then reverted it.

Watchdog should work in flatpak, but if it doesn't perhaps we have to add the xml files by changing "cmake/plugiinXML.cmake"

Try these [files here](https://cloudsmith.io/~opencpn/repos/watchdog-prod/packages/?q=2.4.15.0*metadata) pick the flatpak tarball to import it and test.

Let me know what works please.
https://github.com/rgleason/watchdog_pi/blob/master/cmake/PluginXML.cmake
What do we do?   ADD 

IF(WIN32 + FEDORA)?

leamas, 2020-07-09:
> What do we do? ADD IF(WIN32 + FEDORA)?

Let cmake do it's job to probe for TinyXML availability. I could rebase my patch to watchdog to do this. Or give me commit rights to proper repository in which case I ccould commit it.

bdbcat, 2020-07-09:
Working three code bases in one thread makes me crazy.
Please, at lest make the first sentence of each post contain the name of the plugin you are discussing :)
Thanks
Dave

bdbcat, 2020-07-09:
Weatherfax:
1.  The two links you asked me to test show no tarballs.  Only metadata.  Impossible to import and test.
2.  I did test my latest patch last night on flatpak.  It worked.

leamas, 2020-07-09:
@bdbcat : Indeed.
My post at https://github.com/OpenCPN/OpenCPN/issues/1967#issuecomment-656181963  (above) is about  watchdog

bdbcat, 2020-07-09:
Watchdog:
AFAICT, the current master catalog entry for watchdog works on NO platforms.  Yet, it did at least work on Windows in the not too recent past.  We do not get a link error message.  Just "fails to load".
@leamas 
If you rebase your patch to current, I will apply it.  I have commit rights...

leamas, 2020-07-09:
@bdcat:

watchdog:

Attaching rebased patch against _c0e5e14 (origin/master, origin/HEAD) Revert "Add tinyxml source"_

[0001-Build-make-tinyxml-a-separate-built-library.patch.gz](https://github.com/OpenCPN/OpenCPN/files/4898049/0001-Build-make-tinyxml-a-separate-built-library.patch.gz)

Carstijn, 2020-07-09:
@rgleason the weatherfax plugin to test directly from cloudsmith works for me!

rgleason, 2020-07-09:
**WeatherFax Plugin**  two versions to test   (Sorry bdbcat wrong search):
[v1.9.11.0 with tnyxml + plugingl ](https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/?q=1.9.11.0*flatpak)(most like bdbcat) but also included plugingl
or
[v1.9.12.0 used pluginGL + pluginXML only](https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/?q=1.9.12.0*flatpak)

**Weather-routing Plugin** 
[1.13.11-Dave's first fix of tinyxml this is the tarball and the metadata.](https://cloudsmith.io/~opencpn/repos/weather-routing-prod/packages/detail/raw/weather_routing_pi-1.13.11.0-flatpak-x86_64-18.08-flatpak-tarball/v1.13.11.0/)

[1.13.12-Subsequent changes too tinyxml and Set(files to improve performance ](https://cloudsmith.io/~opencpn/repos/weather-routing-prod/packages/detail/raw/weather_routing_pi-1.13.12.0-flatpak-x86_64-18.08-flatpak-tarball/v1.13.12/)

rgleason, 2020-07-09:
OK, that's good, Which one? Can you test both please?  We'll know more as a result.
@rgleason the weatherfax plugin to test directly from cloudsmith works for me!

rgleason, 2020-07-09:
Alec, I will add you to all of Sean's PI that we are currently working on. Then if you make your Cloudsmith profile "public" so I can send you an invite to the OpenCPN Organizattion and add you to the Plugin Team.

Carstijn, 2020-07-09:
@rgleason **Status:**
WeatherFax Plugin:
v1.9.11.0 with tnyxml + plugingl **Works**
or
v1.9.12.0 used pluginGL + pluginXML only **Does Not Work**

Weather-routing Plugin
1.13.11-Dave's first fix of tinyxml this is the tarball and the metadata. **Does Not Work**

1.13.12-Subsequent changes too tinyxml and Set(files to improve performance **Does Not Work**

Output of WeatherRouting_PI, same for both versions:
20:14:39.627 MESSAGE pluginmanager.cpp:2157 PlugInManager: Loading PlugIn: /home/sander/.var/app/org.opencpn.OpenCPN/lib/opencpn/libweather_routing_pi.so
20:14:39.631   ERROR dlunix.cpp:162 /home/sander/.var/app/org.opencpn.OpenCPN/lib/opencpn/libweather_routing_pi.so: undefined symbol: _ZTV13TiXmlDocument

leamas, 2020-07-09:
watchdog:

plugin cannot be loaded due to an missing bz2 (bzip2) dependency. We need to bundle also that.

rgleason, 2020-07-09:
ok

leamas, 2020-07-10:
bzip2 is handled, and ldd on the plugin looks ok in tge sandbox.  Still, we are facing:

```
02:25:17.811 MESSAGE pluginmanager.cpp:2340     /home/mk/.var/app/org.opencpn.OpenCPN/lib/opencpn/libwatchdog_pi.so cannot be loaded
02:25:17.811 MESSAGE pluginmanager.cpp:1254     PlugInManager: Unloading invalid PlugIn, API version 116

```

rgleason, 2020-07-10:
Weather_Routing Does this [weather_routing 1.13.13  work in flatpak](https://cloudsmith.io/~opencpn/repos/weather-routing-beta/packages/?q=1.13.13.0*flatpak*tarball) perhaps?

rgleason, 2020-07-10:
There is an icon problem in watchdog too.  (It does not show)  I had started working on that one, had not solved it.
I added 

Line 109  set(OpenGL_GL_PREFERENCE "LEGACY")  <--don't know if this is still needed
Line 115 option(USE_GL "Enable OpenGL support" OFF)   (I had commented this out, don't know if it works.)

Line 204
SET(SRC_GL
            src/GL/gl.h
            src/GL/gl_private.h
            src/GL/glext.h
            src/GL/glu.h
)

Line 291  ADD_LIBRARY(${PACKAGE_NAME} SHARED ${SRCS} ${HDRS} ${SRC_JSON} ${SRC_NMEA0183} ${SRC_GL})

Line 328 INCLUDE("cmake/PluginGL.cmake")   <---Uncommented this at the bottom.

The errors are:

```
  Alarm.cpp
  signalk_client.cpp
C:\Users\Rick\Documents\GitHub\watchdog_pi\src\watchdog_pi.cpp(335): error C2065: 'GL_BLEND': undec
lared identifier [C:\Users\Rick\Documents\GitHub\watchdog_pi\build\watchdog_pi.vcxproj]
C:\Users\Rick\Documents\GitHub\watchdog_pi\src\watchdog_pi.cpp(335): error C3861: 'glEnable': ident
ifier not found [C:\Users\Rick\Documents\GitHub\watchdog_pi\build\watchdog_pi.vcxproj]
C:\Users\Rick\Documents\GitHub\watchdog_pi\src\watchdog_pi.cpp(337): error C2065: 'GL_BLEND': undec
lared identifier [C:\Users\Rick\Documents\GitHub\watchdog_pi\build\watchdog_pi.vcxproj]
C:\Users\Rick\Documents\GitHub\watchdog_pi\src\watchdog_pi.cpp(337): error C3861: 'glDisable': iden
tifier not found [C:\Users\Rick\Documents\GitHub\watchdog_pi\build\watchdog_pi.vcxproj]
  WatchdogUI.cpp
  icons.cpp
  wddc.cpp
  TexFont.cpp
C:\Users\Rick\Documents\GitHub\watchdog_pi\src\wddc.cpp(223): error C2065: 'GL_TEXTURE_2D': undecla
red identifier [C:\Users\Rick\Documents\GitHub\watchdog_pi\build\watchdog_pi.vcxproj]
C:\Users\Rick\Documents\GitHub\watchdog_pi\src\wddc.cpp(223): error C3861: 'glDisable': identifier
not found [C:\Users\Rick\Documents\GitHub\watchdog_pi\build\watchdog_pi.vcxproj]
C:\Users\Rick\Documents\GitHub\watchdog_pi\src\wddc.cpp(225): error C2065: 'GL_BLEND': undeclared i
dentifier [C:\Users\Rick\Documents\GitHub\watchdog_pi\build\watchdog_pi.vcxproj]
C:\Users\Rick\Documents\GitHub\watchdog_pi\src\wddc.cpp(225): error C3861: 'glDisable': identifier
not found [C:\Users\Rick\Documents\GitHub\watchdog_pi\build\watchdog_pi.vcxproj]
```
I do not know if GL is related to the icon not showing problem.
Also, I must say that I did not realize watchdog was a signalk client and do not know when that happened.

[35 more replies]

closed 2020-07-21 by Carstijn