OpenCPN/OpenCPN #1970: Fedora 30 - Flatpak - incorrect data directory returned from pluginmanager

jongough, 2020-07-05:
Using Fedora 30 and with the flatpak install of OCPN 5.1529-0+41d8121 installed through Plugin Manager ocpn_draw_pi 1.8.0.16. On enabling ocpn_draw it runs and populates the Toolbar with icons, but these are the default jigsaw puzzle icons not the icons for the plugin. The OCPN log has the following:
```
09:12:39.169 MESSAGE pluginmanager.cpp:1134 Checking plugin candidate: /home/jon/.var/app/org.opencpn.OpenCPN/lib/opencpn/libocpn_draw_pi.so
09:12:39.192   ERROR pluginmanager.cpp:1829 Could not get ELF structures from "opencpn".
09:12:39.192 MESSAGE pluginmanager.cpp:2037 Plugin is compatible by elf library scan: true
09:12:39.192 MESSAGE pluginmanager.cpp:2145 PlugInManager: Loading PlugIn: /home/jon/.var/app/org.opencpn.OpenCPN/lib/opencpn/libocpn_draw_pi.so
09:12:39.194 MESSAGE OCPNPlatform.cpp:1596 Using plugin data path: /home/jon/.var/app/org.opencpn.OpenCPN/data;/app/lib/opencpn
09:12:39.194 MESSAGE pluginmanager.cpp:541 PlugInManager: Using data dirs from: /home/jon/.var/app/org.opencpn.OpenCPN/data;/app/lib/opencpn
09:12:39.195 MESSAGE pluginmanager.cpp:561 Warnińg: no data directory found, using ""
09:12:39.195 MESSAGE ODicons.cpp:63 ocpn_draw_pi data location: data/
09:12:39.195   ERROR svgxml.cpp:445 File data/ODManager_MUI.svg doesn't exist.
09:12:39.195   ERROR svgxml.cpp:445 File data/ODManagergrey_MUI.svg doesn't exist.
09:12:39.195   ERROR svgxml.cpp:445 File data/boundary_MUI.svg doesn't exist.
09:12:39.195   ERROR svgxml.cpp:445 File data/boundarygrey_MUI.svg doesn't exist.
```
The plugin manager gave a warning that it could not find the data directory so returned a zero length string. All the icon files are located in 
`/home/jon/.var/app/org.opencpn.OpenCPN/data/opencpn/plugins/ocpn_draw_pi/data`
which should be the correct place and were placed there by Plugin Manager.

bdbcat, 2020-07-05:
Corrected in github.  Probably need to wait for overnight flatpak repo update to see the results, unless @leamas can bump it.

bdbcat, 2020-07-07:
Jon...
Seems fixed, in latest overnight build.  Version 5.1.704+59355cc
[image]

jongough, 2020-07-07:
I have run 'flatpak update org.opencpn.OpenCPN' but I get the answer 'Nothing to do.'. Is there some special technique not mentioned in [https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:flatpak](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:flatpak) so that I can try out the latest version?

jongough, 2020-07-07:
I completely uninstalled and deleted the reference to the repository and reinstalled OCPN completely and managed to get version 5.1.704+59355cc up and running and the icons appear OK. So it is working OK. Thanks.

closed 2020-07-07 by jongough