OpenCPN/OpenCPN #1971: flatpak: plugins are not visible (incompatible).

leamas, 2020-07-05:
Using flatpak version .1.621-0+7c8236f:
  - The oesenc plugin is not visible in the plugin manager list of plugins.
  - The ocpn_draw plugin is visible. After installation, here is a message _This plugin is inot compatible with this version of opencpn and will be uninstalled_

See also https://github.com/OpenCPN/OpenCPN/issues/1967#issuecomment-653805748

Something broken in PluginHandler.cpp: PluginHandler::isCompatible() ? Or ?

leamas, 2020-07-05:
After trying to install ocpn_draw  the log contains:

```
08:09:30.730 MESSAGE pluginmanager.cpp:441 Installing OCPN Draw
08:09:30.730 MESSAGE PluginHandler.cpp:992 Installing ocpn_draw_pi-1.8.0.13-x86_64_flatpak-18.08.tar.gz from local cache
08:09:32.768   ERROR pluginmanager.cpp:2019     Plugin "/home/mk/.var/app/org.opencpn.OpenCPN/lib/opencpn/libocpn_draw_pi.so" is of another binary flavor than the main module.
08:09:32.768 MESSAGE pluginmanager.cpp:2044 Plugin is compatible by elf library scan: false

```

oesenc has some similar lines.

leamas, 2020-07-05:
After this patch 

```
--- a/src/pluginmanager.cpp
+++ b/src/pluginmanager.cpp
@@ -2017,6 +2017,7 @@ bool PlugInManager::CheckPluginCompatibility(wxString plugin_file)
             {
                 pi_info.dependencies.clear();
                 wxLogError( wxString::Format( _T("    Plugin \"%s\" is of another binary flavor than the main module."), plugin_file ) );
+                wxLogDebug("host magic: %.8x, plugin magic: %.8x", own_info.type_magic, pi_info.type_magic);
             }
             for( ModuleInfo::DependencyMap::const_iterator own_dependency = own_info.dependencies.begin(); own_dependency != own_info.dependencies.end(); ++own_dependency )
             {
```

The log shows 

```
09:01:40.780   ERROR pluginmanager.cpp:2019     Plugin "/home/mk/.var/app/org.opencpn.OpenCPN/lib/opencpn/liboesenc_pi.so" is of another binary flavor than the main module.
09:01:40.780   DEBUG pluginmanager.cpp:2020 host magic: 00030102, plugin magic: 00000102
```
The last part of magic seems to be  EI_ABIVERSION | EI_OSABI | EI_DATA | EI_CLASS

Seems that the flatpak code cannot retrieve the ABI and ABI version info from the plugin (?).

Looking at the ELF header inside the sandbox for the opencpn binary and the plugin they look the same.

@bdbcat: Thoughts ?

leamas, 2020-07-05:
This seems to be about the plugins. Rebuilding the flatpak from the first beta Beta_5.1.428 (e63cdd20e) shows the same problems.  OTOH, we have definitely been able to install plugins on this version.e63cdd20e

It should be possible to bisect to the failing commit. However, I'm on a slow laptop, and each build is done from scratch.

leamas, 2020-07-05:
@jongough : ^

rgleason, 2020-07-05:
Thanks Alec, that's clear. 
It's the 4th of July weekend here "Independence Day" another reason to celebrate.

As we know:
oesenc uses frontend1
ocpn_draw uses frontend2

leamas, 2020-07-05:
@rgleason: This has nothing to do with the data directories lookup whatsoever. Please read the posts above.

bdbcat, 2020-07-05:
@leamas 
Magic calculation:

```
   info.type_magic =
        ( static_cast< uint64_t >( elf_file_header.e_ident[EI_CLASS] ) << 0 ) |         // ELF class (32/64).
        ( static_cast< uint64_t >( elf_file_header.e_ident[EI_DATA] ) << 8 ) |          // Endianness.
        ( static_cast< uint64_t >( elf_file_header.e_ident[EI_OSABI] ) << 16 ) |        // OS ABI (Linux, FreeBSD, etc.).
        ( static_cast< uint64_t >( elf_file_header.e_ident[EI_ABIVERSION] ) << 24 ) |   // OS ABI version.
        ( static_cast< uint64_t >( elf_file_header.e_machine) << 32 ) |                 // Instruction set.
        0;

```
We fail to match 
`host magic: 00030102, plugin magic: 00000102`

Looks like the instruction set does not match. Could this be true?

from libelf.h

```
/* Legal values for e_machine (architecture).  */
#define EM_NONE		 0		/* No machine */
#define EM_386		 3		/* Intel 80386 */

```

Please check the elf header again, in the sandbox...
Dave

jongough, 2020-07-05:
For ODraw and using the JSON schema validation under Linux the OS/ABI gets set to 'Linux' (3) whereas OCPN is normally built with OS/ABI set to 'none' (0). I have specific code in my CMakeLists.txt file to change the OS/ABI in the elf header so that it will work. I think it is to do with the gcc compiler and what it determines is being used. It seems to be a bit too agressive and this causes issues for plugins like ODraw. Not sure if that is any help, but....

leamas, 2020-07-06:
@bdbcat:

```
$ flatpak run --command=sh --devel org.opencpn.OpenCPN 
$ readelf -h /home/mk/.var/app/org.opencpn.OpenCPN/lib/opencpn/liboesenc_pi.so | head -10
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              DYN (Shared object file)
  Machine:                           Advanced Micro Devices X86-64
  Version:                           0x1

```
In the log: 
`
10:06:36.552   ERROR pluginmanager.cpp:1996     Plugin "/home/mk/.var/app/org.opencpn.OpenCPN/lib/opencpn/liboesenc_pi.so" is of another binary flavor than the main module.`

bdbcat, 2020-07-06:
Jon...
I think you are correct.  I think I mis-counted bits.  We are failing on the OS/ABI field.
I'll take a look at your code.
Thanks
Dave

bdbcat, 2020-07-07:
@leamas...
Looks mike a mismatch of OSABI.
OpenCPN checks the elf header of a wxWidgets dependent library to get host magic.  Normally, and I have never seen otherwise, this produces "0" for OSABI.  And normally the plugin builds produce the same, "0"  So all is good.
Now, on flatpak build, the host OSABI is coming out as "3" (linux specific).  This means that something has changed recently in the flatpak build recipe, or ???
I suspect (and read) that this difference does not matter in real life, but we should try to understand it.
Meanwhile, I will commit a change that ignores OSABI mismatch for flatpak builds, and see how it goes overnight. The worst that can happen is that the plugin will fail to link...
Dave

bdbcat, 2020-07-07:
@leamas 
Well, my quick patch to ignore OSABI differences for flatpak did not work as expected.
So, can you give me a quick tutorial on how to build OCPN locally for flatpak, so that I can add some messages and see what is happening?  This will be quicker than waiting for overnight build....
Thanks

bdbcat, 2020-07-07:
@leamas 
New news...
Seems all OK now.  There was a small problem with oesenc metadata.
Also, no trouble with draw_pi, either.
So, if you report OK, we can close.

leamas, 2020-07-07:
@bdbcat: Instructions for local build-install are in flatpak/README.md (Developer Usage)

I will revise the nigthly builds to run hourly. Does not mean that we actually rebuilds every hour, just polling and a rebuild if something is pushed to master.

Short of time, will test after dinner I am preparing.

leamas, 2020-07-07:
All: everything looks OK, thanks for your efforts!

I have updated the "nightly" builds to run hourly. This means that changes to master branch should be reflected in the flatpak repo within at most an hour. 

Closing.

closed 2020-07-07 by leamas