OpenCPN/OpenCPN #1973: No sound on RPi

Fabien83, 2020-07-06:
Hello,

I have a fresh install of the latest Raspberry pi OS on my Raspberry pi 2 model B.
I have installed OpenCPN via the following repository :
deb http://ppa.launchpad.net/opencpn/opencpn/ubuntu/ bionic main

The sound works great with all other programs (VLC, QTVLM, ...) but doesn't work with OpenCPN.
The sound for the anchor alarm or AIS CPA alarm doesn't work and I have the following errors when I start OpenCPN :

Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM input
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
18:36:19: Device: 0: dmix
18:36:19: Device: 0: dmix
18:36:19: Device: 0: dmix
18:36:19: Device: 0: dmix

rgleason, 2020-07-06:
I believe that portaudio was removed at a some point because we now have to add that library for weatherfax.

Fabien83, 2020-07-08:
Hi,
I have libportaudio2 already installed by default. I have also installed libportaudio0 but it doesn't change anything, still got the same error message and no sound (but VLC and others works).

Am I the only one with this error on raspberry ? Should I try to re-install completely ?

leamas, 2020-07-21:
@fabien: First question: what does the opencpn logfile at ~/.opencon/opencpn.log say? If you don't find anything related to sound, please submit the complete log as an attachment.

Fabien83, 2020-09-05:
Hello,
thank you for your answer and sorry for the delay of mine.
In the log I have only found this :
20:57:11 CEST: Device: 0: dmix
As requested ,please find the complete following this google drive link :
https://drive.google.com/file/d/1wwkL021LPV-zXnfnPwJA3efNXLTObFH2/view?usp=sharing

Thanks for your help.
Fabien

leamas, 2020-09-05:
Hm... If I understand this correct you are trying to play mp3 files. Unfortunately, the rPi opencpn seems to be compiled without libsndfile support. This means that the only accepted format  is .wav.

Fabien83, 2020-09-05:
No the mp3 file that appears in the log is a random file I've downloaded to try to play as an opencpn alarm. Because it doesn't work with the two genuine wav files, I wanted to try with an mp3. Ok opencpn can't play mp3, but on my install it doesn't play wav either.

leamas, 2020-09-06:
OK, we will then need some help.

@bdbcat: Do you have the build logs  from the rpi PPA builds available?

bdbcat, 2020-09-06:
Alec...
https://launchpad.net/~opencpn/+archive/ubuntu/opencpn/+build/19848886/+files/buildlog_ubuntu-bionic-armhf.opencpn_5.2.0.9-0~bionic1_BUILDING.txt.gz

leamas, 2020-09-06:
Seems like an upstream bug: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=944509. 

There is probably not much we can do about it, unless someone makes a PPA with an earlier portaudio version so we can downgrade from current version 19.6.0-1.

Sorry to bring bad news...

leamas, 2020-10-04:
Hm... it seems like the upstream bug is fixed in the latest Debian portaudio version 19.6.0-1.1. Problem is that this is not backported to current releases, it's only available on debian sid and ubuntu groovy/20.10. However, we could backport it ourselves to the PPA builds if there is some time; it's basically a question of priorities.

@bdbcat: Thoughts?

rgleason, 2021-04-14:
Is this fixed now?

bdbcat, 2021-04-14:
Nope.

commit a70d1c98 (2021-04-18): Correct AIS Alert sound enable logic in Settings. Closes #1973
closed 2021-04-18 by bdbcat