OpenCPN/OpenCPN #1976: watchdog_pi: nio Bundled metadata missing.

leamas, 2020-07-09:
The bundled metadata required to import plugins is missing in watchdog_pi. This needs to be fixed, and also checked in other plugins before release.

leamas, 2020-07-09:
hm... it's there, but in the wrong location.

rgleason, 2020-07-09:
leamas, I will copy the frontend2 files back over to fix it. thanks!

rgleason, 2020-07-09:
From testplugin_pi

```
if [ "$TRAVIS" ] || [ "$CIRCLECI" ]; then
    mkdir build_tar
    cp $tarball_tar build_tar/.
    cd build_tar
    tar -xf $tarball_tar
    rm *.tar
    cp ../metadata.xml .
			   
						
    tar -cf build_tarfile.tar *
    tar -tf build_tarfile.tar
    rm ../$tarball_tar
    cp build_tarfile.tar ../$tarball_tar
    cd ..
    rm -rf build_tar
else
    tar -rf $tarball_tar metadata.xml
fi
```

rgleason, 2020-07-09:
Leamas change
```

cp -f $xml_here metadata.xml
if [ "$TRAVIS" ] || [ "$CIRCLECI" ]; then
    mkdir build_tar
    cp $tarball_tar build_tar/.
    cd build_tar
    tar -xf $tarball_tar
    rm *.tar
    cp ../metadata.xml .
    rm -rf root
    cp ../metadata.xml *
    tar -cf build_tarfile.tar *
    tar -tf build_tarfile.tar
    rm ../$tarball_tar
    cp build_tarfile.tar ../$tarball_tar
    cd ..
    rm -rf build_tar
else
    tar -rf $tarball_tar metadata.xml
```

leamas, 2020-07-09:
Hm... this is IMHO better presented as a the patch:

```
--- a/cmake/in-files/cloudsmith-upload.sh.in
+++ b/cmake/in-files/cloudsmith-upload.sh.in
@@ -193,7 +193,8 @@ if [ "$TRAVIS" ] || [ "$CIRCLECI" ]; then
     cd build_tar
     tar -xf $tarball_tar
     rm *.tar
-    cp ../metadata.xml .
+    rm -rf root
+    cp ../metadata.xml *
     tar -cf build_tarfile.tar *
     tar -tf build_tarfile.tar
     rm ../$tarball_tar

```

rgleason, 2020-07-09:
I agree, I've got to learn how to create and use those.  I wrote Jon about it linked to here.

Weather_routing has this
https://github.com/rgleason/weather_routing_pi/blob/master/CMakeLists.txt
will this work?

```
IF(NOT UNIX)
    SET(SRC_BZIP	
            src/bzip2/bzlib.c
            src/bzip2/blocksort.c
            src/bzip2/compress.c
            src/bzip2/crctable.c
            src/bzip2/decompress.c
            src/bzip2/huffman.c
            src/bzip2/randtable.c
			src/bzip2/bzlib_private.h
    )
    ADD_LIBRARY(LIB_BZIP_WR STATIC ${SRC_BZIP})
    TARGET_LINK_LIBRARIES( ${PACKAGE_NAME} LIB_BZIP_WR )
    INCLUDE_DIRECTORIES(${PROJECT_SOURCE_DIR}/src/bzip2)
#	INCLUDE_DIRECTORIES(${PROJECT_SOURCE_DIR}/src/jsoncpp)
ENDIF(NOT UNIX)
```

rgleason, 2020-07-09:
Watchdog was installling and working!   So what happened?  I did update the frontend2 files and made a new version.

rgleason, 2020-07-09:
Here is another example from climatology
```

IF(NOT UNIX)
    SET(SRC_BZIP
            src/bzip2/bzlib.c 
            src/bzip2/blocksort.c 
            src/bzip2/compress.c 
            src/bzip2/crctable.c 
            src/bzip2/decompress.c 
            src/bzip2/huffman.c 
            src/bzip2/randtable.c
    )
    ADD_LIBRARY(LIB_BZIP_CL STATIC ${SRC_BZIP})
    TARGET_LINK_LIBRARIES( ${PACKAGE_NAME} LIB_BZIP_CL )

    INCLUDE_DIRECTORIES(${CMAKE_SOURCE_DIR}/buildwin/include)
    INCLUDE_DIRECTORIES(${PLUGIN_SOURCE_DIR}/src/bzip2)
ENDIF(NOT UNIX)

IF(WIN32)
  IF(MSVC)
    SET(OPENCPN_IMPORT_LIB "../../${CMAKE_CFG_INTDIR}/${PARENT}")
 	INCLUDE_DIRECTORIES(${CMAKE_SOURCE_DIR}/buildwin/include)
    TARGET_LINK_LIBRARIES(${PACKAGE_NAME} "${CMAKE_SOURCE_DIR}/buildwin/zlib1.lib")
  ELSE()
    TARGET_LINK_LIBRARIES(${PACKAGE_NAME} "-lz")
  ENDIF()
ENDIF(WIN32)
```

rgleason, 2020-07-09:
Alec Thanks. 
I've added your [tinyxml sub module to weatherrouting](https://github.com/rgleason/weather_routing_pi)

rgleason, 2020-07-11:
I am now trying to understand exactly what the problem was and whether or not changes are needed to Frontend2, 

Conclusion: I am finding that there is no change needed to Frontend2 and can only quess that Watchdog was having troubles with the flatpak build due to the many configuration and API updates that were missed, which Alec eventually found.  Additionally I am mystified why there was a problem with the weatherfax tarball and why it needed to be changed. 

Methodology: Build each plugin tarball locally and compare them. I attach the results below.
Watchdog 2.4.17.0  (which was using Frontend2 v1.0.124) + Alec's changes
Weatherfax 1.9.13.0 (which was using Frontend2 v1.0..125) + Alec's changes
Weather_routing 1.13.14.0 (which was unchanged and used Frontend2 v1.0.125
Testplugin_pi  (frontend2 v1.0.125  direct from Jon's repo)
Testplugin_rg (frontend2 v1.0.125  direct from Rick's fork)

Visual comparison of the structure of these tarballs indicates that they have the same structure!
So my conclusion is that there was something very weird going on with watchdog and weatherfax, because it was not caused by Frontend2!

All the tarball files are in this zip file.
[tarball-zip.zip](https://github.com/OpenCPN/OpenCPN/files/4907755/tarball-zip.zip)

Therefore IMHO there is no need to do anything with Frontend2 v1.0.125!

leamas, 2020-07-12:
weather_routing: 

No problems I know of. In particular, the tinyxml xml dependency should be fixed in master.

It turns out that this has probably never been a problem. It was just that I tested the wrong version (#1979), and started fixing a thing which wasn't broken. 

O dear...

leamas, 2020-07-24:
@rgleason: is this fixed  i. e., the metadata in right location, import plugin works in current version(s)

rgleason, 2020-07-24:
Yes, I believe so. Thanks very much.

leamas, 2020-07-24:
OK, let's close this then.

closed 2020-07-24 by leamas