OpenCPN/OpenCPN #1977: flatpak: plugin library is not loaded.

leamas, 2020-07-11:
In flatpak, a plugin with a helper library is not able to load this library. One example is the weather_routing plugin, currently using a dynamic libtinyxml.so library. One other linux platform libtinyxml.so is found using LD_LIBRARY_PATH which is set by main opencpn, but this does not work in the flatpak sandbox.

Problem has slipped through the net because the libraries we have had so far has been for example in oesenc_pi, but these are linked to a separate binary which seems to work fine.

Possible work-arounds includes avoiding dynamic libraries or  using a $ORIGIN rpath. Both of these applies to the plugins rather that main opencpn.

See als #1967, in particular on the weather_routing plugin

bdbcat, 2020-07-12:
@leamas...
I found you...
in this post:
https://www.linuxquestions.org/questions/programming-9/ld_library_path-and-dlopen-422816/

We get the idea that weather_routing_pi.so may be compiled with an RPATH specification of simply "dot".

`-Wl,-rpath,. `
Worth a try?

leamas, 2020-07-12:
I have applied both solutions to weather_routing:

  - Ditched the dynamic library, using a static instead. https://github.com/rgleason/weather_routing_pi/commit/62af5662b95927
  - Added a rpath: https://github.com/rgleason/weather_routing_pi/commit/ba98759bf81dc8e7b1ae3206fd30b4ab13a4d54e

If we can make sure the rpath is propagated to the other plugins we should be able to close this. As for the more pressing issues with weather_routing the should now be fixed. This has actually been not been a problem at all, I was just hit by #1979 and misinterpreted the situation

leamas, 2020-07-12:
It should be noted that this bug IMHO not is a release blocker; we can sort out things in next cycle as long as the plugins works.

rgleason, 2020-07-12:
Summary: flatpak seems to be pretty picky about running stuff. everything Alec has done with the three plugins was not the problem (but he fixed many other watchdog problems) and now you guys know what the problem is.

rgleason, 2020-07-12:
Alec thanks for revert to static.  Do we know the rpath works yet?
When is this needed?  Can it just be added universally? Is rpath going to be needed now in all plugins?
Also what needs to be done to Frontend1?

**FRONTEND2 in cmake/pluginSetup.cmake**
at about line 238
```
find_program(HAVE_LD_SO
    PATHS /lib NO_DEFAULT_PATH
    NAMES ld.so ld-linux.so.1  ld-linux.so.2
)

set(CMAKE_SKIP_BUILD_RPATH true)
if (HAVE_LD_SO)   # linux.
    message(STATUS Setting RPATH: \$ORIGIN:\$ORIGIN/..)
    set(CMAKE_BUILD_RPATH "\$ORIGIN;\$ORIGIN/..")
    set(CMAKE_INSTALL_RPATH "\$ORIGIN;\$ORIGIN/..")
endif ()

```

rgleason, 2020-07-12:
The most current flatpak tarball is here
https://cloudsmith.io/~opencpn/repos/weather-routing-beta/packages/detail/raw/weather_routing_pi-1.13.14.0-flatpak-x86_64-18.08-flatpak-tarball/1.13.14.0+1988.138cec8/

leamas, 2020-07-12:
> Do we know the rpath works yet? 

I would take it for granted, checked that it is in place.

> When is this needed? 

As soon as a plugin uses a dynamic library

> Does this work for all libraries? 

All dynamic, yes, if installed together with the plugin

> Can it just be added universally?

Yes
 
> Is rpath going to be needed now in all plugins?

Not  required, but adds some more redundancy to library linking and loading.

I'd recommend to apply this to all plugins.

leamas, 2020-07-12:
> Summary: flatpak seems to be pretty picky about running stuff. 

My changes are basically just general cleanup, nothing besides the rpath is flatpak-specific

"Stuff" here is in particular dependencies. I'd recommend moving some more of these to libs/ and create CMakeList.txt similar to tinyxml and bzip2. This would make it far easier to reuse code between plugins an minimizes the impact on the main CMakeLists.txt

leamas, 2020-07-22:
@rgleason : what is the status of this? Is the rpath added to the "frontend" plugins?

rgleason, 2020-07-23:
No, not to my knowledge. Should we do that?  If you think so, I guess we'd better do it.  Jon is the guy in charge there, I am just his helper/troublemaker. Thanks for suggesting.

rgleason, 2020-07-23:
PS: Should I be pushing this new version of weather_routing up to opencpn/plugins now?

leamas, 2020-07-24:
There are many wheels turning around right now. Let's wait a little so the dust settles before doing this change.

leamas, 2020-08-12:
@rgleason 

hm... the dust seems to have settled. Time to push the new version so we can close this.

leamas, 2020-08-29:
It turns out that the required patch is simpler:

```
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -108,6 +108,9 @@ set(SRC_ShipDriver
 add_library(${PACKAGE_NAME} SHARED ${SRC_ShipDriver})
 include_directories(${PROJECT_SOURCE_DIR}/api-16)
 include_directories(${CMAKE_BINARY_DIR}/include)
+if (UNIX AND NOT APPLE)   # linux.
+  set_target_properties(${PACKAGE_NAME} PROPERTIES INSTALL_RPATH "$ORIGIN:$ORIGIN/..")
+endif ()

```

Done this way, it does not affect other targets, notably test runners...

rgleason, 2020-08-29:
So at LIne 108 in this file [CMakeLists.txt](https://github.com/rgleason/weather_routing_pi/blob/master/CMakeLists.txt)
Just before  the line  "if("${CMAKE_BUILD_TYPE}" STREQUAL "")"

Put these lines:
```
set(SRC_weather_routing  add_library(${PACKAGE_NAME} SHARED ${SRC_weather_routing })
 include_directories(${PROJECT_SOURCE_DIR}/api-16)
 include_directories(${CMAKE_BINARY_DIR}/include)
+if (UNIX AND NOT APPLE)   # linux.
+  set_target_properties(${PACKAGE_NAME} PROPERTIES INSTALL_RPATH "$ORIGIN:$ORIGIN/..")
+endif ()

```
I don't really see how that ties in with the other "SET" being used down at Line 293

But what do I do about this ?
[Add Proper Rpath](https://github.com/rgleason/weather_routing_pi/commit/ba98759bf81dc8e7b1ae3206fd30b4ab13a4d54e)

leamas, 2020-08-29:
The first line looks just weird, typo, copy-paste error? It's not in the repo, which is fine.

If apply the patch to your file I get:
```

diff --git a/CMakeLists.txt b/CMakeLists.txt
index b6c81b6..47a93eb 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -329,6 +329,10 @@ SET(SRC ${SRCS} ${SRC_LIBTESS2} )
 
 ADD_LIBRARY(${PACKAGE_NAME} SHARED ${SRC})
 #Add_Library (${PACKAGE_NAME} is used below. Cannot use same name.
+if (UNIX AND NOT APPLE)   # linux
+  # See OpenCPN #1977
+  set_target_properties(${PACKAGE_NAME} PROPERTIES INSTALL_RPATH "$ORIGIN:$ORIGIN/..")
+endif ()
 
 # Used in textplugin. Just added.
 add_definitions(-DTIXML_USE_STL)

```

leamas, 2020-08-29:
BTW: the next lines in the file are

```
# Used in textplugin. Just added.
add_definitions(-DTIXML_USE_STL)

# Alec from watchdog
add_subdirectory(libs/tinyxml)
target_link_libraries(${PACKAGE_NAME} ocpn::tinyxml)
```

But the DTIXML_USE_STL definition is not required. It *is* needed to include the tinyxml lib. However,  target_link_libraries() provides all the libraries, include directories and definitions required (if not, it's a bug in the library). So, you could drop these lines.

leamas, 2020-08-29:
@rgleason: perhaps one thing too much, but: when I look in the code I find the jsoncpp dependency. If you are interested, I have converted it to a library available at https://github.com/Rasbats/shipdriver_pi. Using this should clean up a little.

rgleason, 2020-08-29:
Alec I have never figured out  or used patches.
I am going to try apply in this path using  OS Windows.
Do I make this into a txt file and name it  add-rpath.patch?

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index b6c81b6..47a93eb 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -329,6 +329,10 @@ SET(SRC ${SRCS} ${SRC_LIBTESS2} )
 
 ADD_LIBRARY(${PACKAGE_NAME} SHARED ${SRC})
 #Add_Library (${PACKAGE_NAME} is used below. Cannot use same name.
+if (UNIX AND NOT APPLE)   # linux
+  # See OpenCPN #1977
+  set_target_properties(${PACKAGE_NAME} PROPERTIES INSTALL_RPATH "$ORIGIN:$ORIGIN/..")
+endif ()
 
 # Used in textplugin. Just added.
 add_definitions(-DTIXML_USE_STL)

```

And  then put that file in my local github/weather_routing_pi  directory  
Then from the command prompt while in that directory, I would type 

git apply --stat add-rpath.patch   <--- to check the patch to see what it does.
git apply --check add-rpath.patch <---test the patch before actually applying it.
git am --signoff < add-rpath.patch  <-- apply the patch with signoff.

git patch command is pretty complicated I think.

What do I do about this previous commit, you did not respond to that question.
But what do I do about this previous commit  [Add Proper Rpath](https://github.com/rgleason/weather_routing_pi/commit/ba98759bf81dc8e7b1ae3206fd30b4ab13a4d54e)

rgleason, 2020-08-29:
Alec I am perfectly happy to improve weather_routing with these changes, including the Jsoncpp library change, but right now my main focus is fixing the watchdog_pi  inclusion boundary errors and making sure that works properly. It seems to have gone off track somehow and I think it has to do with using the Opencpn  nmea librarry rather than the one that came with watchdog, so I am focused on that.  [Watchdog alarms not working properly](https://github.com/OpenCPN/plugins/issues/235)  - any thoughts about this?

leamas, 2020-08-29:
Patches: in this case it's just about adding three lines, you could to this by hand.

If you want to apply patches, they come in two variants: git patches and standard patches. Git patches adds a commit, but are extremely  picky and hard to use in contexts like this. Here, I would recomment to use the program called patch (sic!). You should then

  - Install the program patch, choco is your friend
  - Save the patch to  a file, say fix.patch
  - Move to the project directory: cd somewhere, and put fix.patch here
  - Apply the patch using `patch -p1 < fix.patch`

Standard patch is more forgiving and handles offsets caused by mismatches between the file use to generate the patch and the file(s) patchednto some extent.

rgleason, 2020-08-30:
Alec,  I have two branches  master and frontend2.
Master is using your libs and I have made the change to use the jsoncpp library.
There is an rpath statement in CMakeLists now (previously), but I will next try to add the  patch that you want and after completing watchdog, when I get to weather_routing I will also remove the commit  [Add Proper Rpath](https://github.com/rgleason/weather_routing_pi/commit/ba98759bf81dc8e7b1ae3206fd30b4ab13a4d54e)

rgleason, 2020-08-30:
Is  cmake/in-files/org.opencpn.OpenCPN.Plugin.yaml.in 
https://github.com/leamas/weather_routing_pi/commit/6e6d6c06528ddab448bde2d5cb9d41dcc9a89dae#
dependent on tinyxml being a lib?

Remove this change  https://github.com/rgleason/weather_routing_pi/commit/ba98759bf81dc8e7b1ae3206fd30b4ab13a4d54e

[Revert building in manifext, use static instead ](https://github.com/leamas/weather_routing_pi/commit/8a100fc43872d2aba19cf12042284d6d155dbfa8)

Does org.opencpn.OpenCPN.Plugin.yaml  change at all?

leamas, 2020-08-30:
>Is cmake/in-files/org.opencpn.OpenCPN.Plugin.yaml.in
 eamas/weather_routing_pi@6e6d6c0 dependent on tinyxml being a lib?

No
> Remove this change rgleason/weather_routing_pi@ba98759

Yes, i. e. `git revert ba98759`

> Does org.opencpn.OpenCPN.Plugin.yaml change at all?

No

leamas, 2021-01-29:
seems to be  fixed in most plugins, Closing

closed 2021-01-29 by leamas