OpenCPN/OpenCPN #1984: Missing Flatpak link to install 5.2

tai-fun, 2020-07-17:
Anyone got the correct link to install O 5.2.
We need to publish that link.
The previous bet link does not work.

rgleason, 2020-07-17:
I would suggest this one
https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/

bdbcat, 2020-07-17:
Thomas...
That repo belongs to @leamas .  He has configured a build-bot to update the repo daily, from github master.
However, he is out cruising for a while.  So we will have to wait for him to re-surface.
The plan is eventually to move the package to the "official" flathub repo.  But this work is incomplete.

lskillen, 2020-07-18:
Aside: Flatpak is on the roadmap for support on @cloudsmith-io too. :)

leamas, 2020-07-18:
> The plan is eventually to move the package to the "official" flathub repo. But this work is incomplete.

It as complete it can be until there is a 5.2 release. We have an open offer to publish a beta, but to be honest I think it's better to wait for the release to avoid needless badwill.

tai-fun, 2020-07-18:
The 5.2 is released. Happened a couple of days ago.

leamas, 2020-07-18:
Great things  happens when I sail. Should do it more often.

Will pursue the flatpak release on flathub as soon as time permits.

leamas, 2020-07-18:
The flathub PR is update: https://github.com/flathub/flathub/pull/1533

It now depends on the flathub admins when the package is published.

Note: In the link page, we need to point out that we only support flatpak/x86_64

leamas, 2020-07-18:
From the flathub PR: until the flathub package is published, the test build of 5.2.0 can be installed using 

    flatpak install --user https://dl.flathub.org/build-repo/23066/org.opencpn.OpenCPN.flatpakref

This should be identical to the upcoming, published package

bdbcat, 2020-07-19:
Download instructions added to opencpn.org downloads page:
https://opencpn.org/OpenCPN/info/downloadopencpn.html

Using interim flathub link until package is finalized as a published package.

leamas, 2020-07-19:
@bdbcat :

flatpak moves quite fast. New download link at https://flathub.org/apps/details/org.opencpn.OpenCPN. This is a point-and-click interface, so the text could be much simplified.

With this link I think it should be possible to close this bug.

leamas, 2020-07-19:
BTW: I have been able to just build x86_64 on flathub, so we don't need to document the missing support for other platforms.

bdbcat, 2020-07-19:
opencpn.org website updated.
Ready to close.

Thanks
Dave

bdbcat, 2020-07-19:
@leamas 
"so we don't need to document the missing support for other platforms."
Sorry, not following??

leamas, 2020-07-19:
@bdbcat: never mind, I have been pushing for documenting that we don't support non-x86_64 flatpak platforms since the test build actually built arm, i386, aarch64.  But once the package was published I learned that it was simple to restrict the build to a single platform, which i did.

Executive summary: this is not a problem, and bug can be closed.

leamas, 2020-07-20:
@tai-fun: ^

closed 2020-07-20 by tai-fun