OpenCPN/OpenCPN #1985: Chrome and macOS warn for dangerous/unidentified software

jorijnsmit, 2020-07-17:
Chrome (Version 83.0.4103.116 (Official Build) (64-bit)) doesn't even let me open the file. Using Firefox I am able to download, however macOS complains about the developer being unknown.

What is needed in order to set this straight?

[image]
[image]

bdbcat, 2020-07-17:
Don't know about Chrome...
But, after downloading from Firefox, go to MacOS  System Preferences/Privacy and Security, and from there you will be allowed to execute the application installer.

jorijnsmit, 2020-07-17:
Yeah thanks, followed this guide: https://support.apple.com/guide/mac-help/open-a-mac-app-from-an-unidentified-developer-mh40616/mac (<kbd>control</kbd>+click `.pkg` file, select open).

nohal, 2021-04-23:
There is nothing we can do about this without paying to Apple the yearly fee and signing the build artifacts.

closed 2021-04-23 by nohal