OpenCPN/OpenCPN #1993: debian packaging tracking bug.

leamas, 2020-07-20:
Filing a bug to track the downstream 5.2.0 Debian packaging. 

Currently the package lingers at https://mentors.debian.net/package/opencpn. We'll see if/when someone steps up to review and eventually upload. It usually takes at least some weeks.

Besides the packaging patches in #1992 the package uses a wrapper script to set OPENCPN_COMPAT_TARGET and this way define what plugins to use. I find this cleaner than messing with the code determining if a plugin is compatible, it is already IMHO to complicated.

At a quick test, the package seems to be able to use the ubuntu plugins.

bdbcat, 2020-07-20:
How do we tell debian mentors to use the correct (5.2.x) branch for a production build?
Is this arrangement "normal"

leamas, 2020-07-20:
They will not use anything not provided by me, so this should not be a problem.

leamas, 2020-07-20:
BTW: The packaging is available at https://gitlab.com/leamas/opencpn

leamas, 2020-09-08:
This process is real slow. There is a faster track, but then I have to confront a person I don't really can deal with.  Avoiding that.

Test builds of the official plugins on focal are available at https://launchpad.net/~leamas-alec/+archive/ubuntu/opencpn/+sourcepub/11570928/+listing-archive-extra. Did some basic testing; plugins seems to install and run just fine.

leamas, 2020-09-14:
5.2.0 is uploaded to Debian. Closing

---

Accepted:

-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

Format: 1.8
Date: Sun, 13 Sep 2020 21:51:16 +0200
Source: opencpn
Architecture: source
Version: 5.2.0+dfsg1-1
Distribution: unstable
Urgency: medium
Maintainer: Alec Leamas <leamas.alec@gmail.com>
Changed-By: Alec Leamas <leamas.alec@gmail.com>
Closes: 948702 962213
Changes:
 opencpn (5.2.0+dfsg1-1) unstable; urgency=medium
 .
   * New upstream release including plugin downloader. Closes: #948702
   * Add Build-Depends libdrm-dev. Closes: #962213
   * Adjust libcurl b-d, add libcurl4-openssl-dev alternative.
   * Update Replace/Break to match actual release.
   * Update debian/copyright due to new upstream source layout.
   * Drop opencpn-plugins transition package.
   * Bump Standards-Version to 4.5.0, no changes.
   * Replace override_dh_install with declarative stuff.
   * Move debian source hosting to https://gitlab.com/leamas/opencpn.
   * Drop upstreamed patches and rebase remaining one.
   * New patches:
       - Add a warning for downloaded, binary plugin contents --
         0008-pluginmanager-conditional-warning-for-plugin-binary-.patch
       - Add udev rule installation patch  - opencpn#1988.
       - Add metadata installation patch - opencpn#1989.
       - Remove bogus svg file - opencpn#1990.
       - Fix cmake parallel execution problem - opencpn#1991.
       - Add two plugin compatibility patches - opencpn#1997.
       - Drop pdf file without source --
         0010-build-Drop-CoC-909_2013-InlandECDIS_20170308s.pdf.patch
Checksums-Sha1:
 ceaaf03779c8e9af46b5002f5021b510cd756a07 2215 opencpn_5.2.0+dfsg1-1.dsc
 bcb6c90b9d88637013b7e04cd1320ec1908cd688 18230328 opencpn_5.2.0+dfsg1.orig.tar.xz
 58642d791d6dc05b9dd4a26b6013f78fb30505bf 26536 opencpn_5.2.0+dfsg1-1.debian.tar.xz
 4c7c61adb3f923c2c6288928aabb41698a1223ea 12493 opencpn_5.2.0+dfsg1-1_source.buildinfo
Checksums-Sha256:
 5fadc84fe88dc6aac2405ea0cd145d2e85ef1e6a9702f7189cfcbfaa5978d084 2215 opencpn_5.2.0+dfsg1-1.dsc
 7d715d3bbb0f990fe42b8ee4839f8abb40d98ce1f2efa2c1ead6e4f91d5528cd 18230328 opencpn_5.2.0+dfsg1.orig.tar.xz
 5153541c86b27b70c5f67c4e9079658e7cef3ff9d1f59a0b40b41533dded569f 26536 opencpn_5.2.0+dfsg1-1.debian.tar.xz
 d56eb53c064117df8070680867d31731af2f7e2be955a24c7b0dab8ae78dbe10 12493 opencpn_5.2.0+dfsg1-1_source.buildinfo
Files:
 f407703915fb3d668298fbf950659d9d 2215 misc optional opencpn_5.2.0+dfsg1-1.dsc
 d6d22a22d5f1ea0c295208e19e5323a2 18230328 misc optional opencpn_5.2.0+dfsg1.orig.tar.xz
 e81bcce82132b41e1df7a58dda699565 26536 misc optional opencpn_5.2.0+dfsg1-1.debian.tar.xz
 0d824a86e054c8f5876767ac033b157b 12493 misc optional opencpn_5.2.0+dfsg1-1_source.buildinfo

-----BEGIN PGP SIGNATURE-----

iQIzBAEBCAAdFiEE/d0M/zhkJ3YwohhskWT6HRe9XTYFAl9fmV0ACgkQkWT6HRe9
XTZS3BAAvaL4Ak1QOLkUYGWjZsgewhTUaTtbHJ+UewTwzIvnaQyHV2XPmGKM2xUv
t0eNzxHDPABN+oiQP8nyeO0NPBkpfGUbSqY1p5Sm4j+e6d4sGa5WJlZA11WuwpUf
LSgvYotxSzbC96k40wcNh+72nq4XyX9o1tt2nKNqdTpaebtuLa6QoGGxppw86QzR
+3yD0q4wRQvVYqk8z6D+aB6h0ibZJelk9aHH3F6wtHfIYQtm6oCfsPmcfbDQ6OBv
AGvvrCuarcCgniSLId+sUDifjiB+nz+dDj4rGuQ1SGJZhYocE54zFdUHL6qb1WTy
wQ1s35ZB5be8DJko97wY8eTar7ancDwjAsVKCo99E6/qH/8bqYIGUP6Yt+qBgaen
cWsx+8bv3sfuiBZiW2NcpHCjm0H1kSHrNYVTCaOzjjDHcmL+2HpBfjHb+XlkeTUA
Elc3Jq8xcEQXDNGw6qSlxhJ6NFayDojQdWNrsKg3bedHfjwz7A9JfoYuylzoZ7hR
qqaXwraLmG7sEFMt0mFS3RoOiJkOBzx7UgxYrAWDDt1hefGVxDctwEy9xXXXpiPL
l8gA9CyhJT57DaN0LE0CY9K7Hk20GUjBHsrajjfXy8Y4aJ4kmnc0XZy3vU4fXHi6
045vsijuGVNYN4YMiMuETsQcvRcFWRDMFkzpAQFibBTWypPvi5s=
=F6IJ
-----END PGP SIGNATURE-----

Thank you for your contribution to Debian.

closed 2020-09-14 by leamas