OpenCPN/OpenCPN #1994: RFE:  Add development release artifacts.

leamas, 2020-07-20:
Plugins are compiled against a specific level of  main opencpn.  Technically, this level is defined by the ocpn_plugin.h header and the two files opencpn.lib (msvc) and opencpn.a (mingw).

The current practice is to copy these items from "somewhere" when building the plugin. This might cause problems if  the files used doesn't match main opencpn.

I suggest that we add a new item in the downloads list called "development files" aimed for plugin devs. To clarify, this should be ocpn_plugin.h, opencpn.lib, opencpn.a and possibly a simple README. The goal is to establish a well-defined version for these items, and make it readily available

leamas, 2020-07-21:
Talk is cheap. Enclosing an example for what I mean, aimed to be included in the list of downloads. The .lib and .dll.a are built directly from 5.2.0 besides a patch to appveyor-upload.sh to make it upload the lib.

[api.zip](https://github.com/OpenCPN/OpenCPN/files/4952108/api.zip)

leamas, 2020-08-31:
I have created a new repository https://github.com/leamas/opencpn-api with complete definitions for API 1.16 (opencpn 5.0.0) and 1.17 (opencpn 5.2.0). Given that developers is the audience for this and the overall heavy github usage in the project this seems like a better way than a download at opencpn.org.

Problem fixed, closing.

closed 2020-08-31 by leamas

rgleason, 2020-08-31:
You might want to refer to 
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:plugins:plugin_api:plugin_api_versions