OpenCPN/OpenCPN #1995: No Sound with Flatpak

tai-fun, 2020-07-20:
I installed the 5.2 flatpakversion, and tried the alarm test. Ubuntu Focal.
No sound at all.
In the opencpn.log....

> 18:41:32.249 WARNING SystemCmdSound.cpp:66 Exit code 127 from command aplay /app/share/opencpn/sounds/2bells.wav

Which, as far as I understand, means that aplay is not available in the opencpn flatpak sandbox.
 Sound works using the normal install.

leamas, 2020-07-20:
Hm... good catch!

The core problem is might be that portaudio isn't visible in the flatpak build, and the the sound thus uses the SystemCmdSound backend. This is probably a dead end, we need to use another sound backend.

This could be either portaudio or something else; the latter means writing a new sound backend. First attempt would be to build portaudio, if reasible.

leamas, 2020-07-20:
@tai-fun: Can you possibly test a patch? That is, do you have a linux host with flatpak-builder installed?

If so, I enclose a patch. The way to test it is:
   - Apply it (to current master)
   - `cd build`
   - `make -f ../flatpak/org.opencpn.OpenCPN.yaml  build`
   - `flatpak run org.opencpn.OpenCPN`

The patch seems to work for me. But then again, that's just me.

[0001-flatpak-Add-portaudio-sound.patch.gz](https://github.com/OpenCPN/OpenCPN/files/4949657/0001-flatpak-Add-portaudio-sound.patch.gz)

rgleason, 2020-07-20:
Alec, I thought I saw it being removed and said something about it being needed for plugins. This was months ago.

tai-fun, 2020-07-21:
@Alec Leamas <leamas.alec@gmail.com>
 Sorry for the delay......
I have problems patching. I get the ' missing separator ' syndrome....
If you still want me to test, send a complete patched
org.opencpn.OpenCPN.yaml file.

leamas, 2020-07-21:
"blushes"

I owe you a beer, at least. You get the "missing separator" because I wrote you should use the yaml file instead of the Makefile.

I have edited my reply (https://github.com/OpenCPN/OpenCPN/issues/1995#issuecomment-661286157). Bottom line: there are better instructions in flatpak/REDAME.md, in the Developer Usage section.

I'm so sorry for misguiding you.

tai-fun, 2020-07-22:
> I have edited my reply (#1995 (comment)

@Alec Leamas <leamas.alec@gmail.com>
I installed flatpak-builder, compiled and tested.
The patch works and the sound feature is back in action.

leamas, 2020-07-22:
@tai-fun :

Many thanks for reporting this and, not least, testing the patch!

This issue will be closed automatically when #1998 is merged.

PR #1998 (2020-07-22): flatpak: Add portaudio sound (#1995)
Add portaudio to the flatpak dependencies which enables the portaudio
sound backend when configuring.

Closes: #1995
commit 0b6d5152 (2020-07-22): flatpak: Add portaudio sound (#1995)

Add portaudio to the flatpak dependencies which enables the portaudio
sound backend when configuring.

Closes: #1995
commit 72ca6b03 (2020-07-28): flatpak: Add portaudio sound (#1995)

Add portaudio to the flatpak dependencies which enables the portaudio
sound backend when configuring.

Closes: #1995
closed 2020-07-22 by bdbcat