OpenCPN/OpenCPN #1996: Compiling 5.2.0 on Slackware 14.2.  Can't find wxWidgets help.

mdinslage, 2020-07-21:
I maintain the OpenCPN package for Slackware over at www.slackbuilds.org.   5.0.0 compiles fine, but I decided to try to compile 5.2.0 today and I am getting an error because it looks like opencpn cant find wxWidgets.  I have tried building with wxGTK, wxGTK3, and wxPython and it gives me the same error that it can't find wxwidgets.   Below is the code, if anyone can help me figure those I would appreciate it.

```
OpenCPN-5.2.0/src/scrollingdialog.cpp
OpenCPN-5.2.0/src/semantic_vers.cpp
OpenCPN-5.2.0/src/shaders.cpp
OpenCPN-5.2.0/src/styles.cpp
OpenCPN-5.2.0/src/tcmgr.cpp
OpenCPN-5.2.0/src/testvers.cpp
OpenCPN-5.2.0/src/thumbwin.cpp
OpenCPN-5.2.0/src/toolbar.cpp
OpenCPN-5.2.0/src/trackprintout.cpp
OpenCPN-5.2.0/src/undo.cpp
OpenCPN-5.2.0/src/update_mgr.cpp
OpenCPN-5.2.0/src/viewport.cpp
OpenCPN-5.2.0/src/wificlient.cpp
OpenCPN-5.2.0/vagrant/
OpenCPN-5.2.0/vagrant/README
OpenCPN-5.2.0/vagrant/Vagrantfile
-- cmake version: 3.5.2
-- *** Build Architecture is i386
-- Setting C++11 standard via cmake standard mechanism
-- Default compiler options:
-- CMAKE_CXX_FLAGS: -O2 -fPIC
-- CMAKE_CXX_FLAGS_DEBUG: -g
-- CMAKE_CXX_FLAGS_MINSIZEREL: -Os -DNDEBUG
-- CMAKE_CXX_FLAGS_RELEASE: -O3 -DNDEBUG
-- CMAKE_CXX_FLAGS_RELWITHDEBINFO: -O2 -g -DNDEBUG
-- Building version 5.2.0
-- *** Staging to build opencpn 5.2.0 ***
-- *** Package will NOT include documentation ***
-- Building against GTK3...
-- Found OpenGL....
--     GL Lib: /usr/lib64/libGLU.so/usr/lib64/libGL.so
--     GL Include: /usr/include
--     Revised GL Lib: /usr/lib64/libGL.so
-- Could NOT find wxWidgets (missing:  wxWidgets_LIBRARIES wxWidgets_INCLUDE_DIRS) 
-- Could not find wxWidgets webview add-on
-- Could NOT find wxWidgets (missing:  wxWidgets_LIBRARIES wxWidgets_INCLUDE_DIRS) 
CMake Error at CMakeLists.txt:739 (message):
  Cannot find wxWidgets, giving up.  On linux you might want to try using
  -DOCPN_FORCE_GTK3.

-- Configuring incomplete, errors occurred!
See also "/tmp/SBo/OpenCPN-5.2.0/build/CMakeFiles/CMakeOutput.log".
root@Slackware64:/tmp/opencpn# 
```

leamas, 2020-07-21:
Well, the message is pretty clear. I have vague ideas how slackware works, but on other distros the basic assumption would be  that a devel package is misssing since it does not find the headers. Or, perhaps slackware install the headers in some location cmake does not check?

The failing command is find_package(wxWidgets). You might want to check https://cmake.org/cmake/help/v3.5/module/FindwxWidgets.html  on how you might be able to tweak variables used by this command. 

OTOH, we are from 5.2.0 shipping a flatpak version. This should work just fine also on slackware.

mdinslage, 2020-07-21:
Yeah I get the message, it is not finding wxWidgets.  Just trying to figure out why since I have compiled everything from source so the header packages are included.

leamas, 2020-07-22:
Right. However, as I see it this is probably between Slackware and cmake (that is not to say there isn't any changes in 5.2.0 which afftects this. In fact, CMakeLists has been subject to a major overhaul.

When it comes to linux our current idea is to support Debian + flatpak. For Slackware, this basically means that we are in a "patches welcome" state without spending that much resources beyond that.

Also note that the new plugin system in 5.2.0 will, in a longer perspective, make it hard to maintain native packages in other linux distros.

mdinslage, 2020-07-22:
Thanks for you response.   The Flatpak option is probably the way to go in the future if someone wants to try use OpenCPN on Slackware.

mdinslage, 2020-07-22:
Leamas, I figured it out.  the wxGTK3 package for Slackware builds with gtk2 libraries, after tweaking the build to use gtk3 and correcting a few symlink issues OpenCPN now finds wxWidgets and builds on Slackware 14.2.  However is it possible to have OpenCPN build again a wxGTK3 using gtk2 libraries?   Some Distros like Arch compile them both, but it looks like with Slackware, as of now, it only compiles in gtk2.

Thank You for your help

closed 2020-07-22 by mdinslage