OpenCPN/OpenCPN #1997: OpenCPN 5.2.0 - Plugin Manager does not show plugins on Linux Mint 20

jongough, 2020-07-22:
The latest OpenCPN 5.2.0 for Ubuntu focal installs and shows a full list of plugins. However on Linux Mint 20, based on Ubuntu 20.4 does not with no explanation. When using the 'Import plugin...' button a message box is shown saying the plugin is incompatible. There is no information provided in the opencpn.log to help determine what is happening. I have tried this with:
[https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-prod/raw/names/ocpn_draw_pi-1.8.0.17-ubuntu-x86_64-20.04-focal-gtk3-tarball/versions/v1.8.0.17/ocpn_draw_pi-1.8.0.17-ubuntu-x86_64-20.04-focal-gtk3.tar.gz](https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-prod/raw/names/ocpn_draw_pi-1.8.0.17-ubuntu-x86_64-20.04-focal-gtk3-tarball/versions/v1.8.0.17/ocpn_draw_pi-1.8.0.17-ubuntu-x86_64-20.04-focal-gtk3.tar.gz)
I have built OCPN with wxWidgets 3.1.3 and gtk3.

leamas, 2020-07-22:
Hm... I'm facing the similar problems on Debian, creating the official 5.2 package (which BTW eventually will hit MInt as well).

Basically, I don't think we can cope with distros downstream ubuntu (like MInt) or upstream Debian by patching the opencpn source for two reasons:

  - The conditions just become too complicated (the isCompatible() function is already messy)
  - The release schedule: We cant really be forced to release a new version of OpenCPN  just because some distro makes a new release.

Ergo: we need to decouple the various distro versions from the main opencpn code (besides the core support of Debian/Ubuntu, Flatpak, MacOS and windows).

In my Debian package I have added a shell wrapper. This is not visible users.  The wrapper sets up OPENCPN_COMPAT_TARGET depending on Debian release and machine architecture and the invokes opencpn. This is simple to do when making a package, but complicatet in a downstream distro like Mint, assuming that the Ubuntu packages should be used as-is.

What we IMHO could do is to add some hook for dowsntream distros. Something which works like:

  - Main opencpn is installed in /usr/libexec/opencpn.
  - /usr/bin/opencpn is a shellscript which invokes /usr/libexec/opencpn after possibly adjusting the environment.
  - The shellscript checks for a distro shellscript like /etc/opencpn/opencpn.sh and uses that instead if it exists.

This way, a /etc/opencpn.sh script could be maintained by Mint packagers, and likewise for other distros. Handling a single distro is way easier than trying to handle all existing distros and -versions in main opencpn.

bdbcat, 2020-07-23:
Alec...
Leaving aside the downstream problems with Mint, etc...
What problems are you having upstream configuring 5.2.0 for vanilla debian?
Not knowing anything about the problem, my first inclination is to fix this in the 5.2.x branch, and use that for the official debian build.  That will be picked up by Ubuntu eventually, and should be good also.
I have not cherry-picked anything into 5.2.x yet, pending your debian tests.

jongough, 2020-07-23:
I have a similar problem with a Raspberry PI 3B+ running openplotter and OCPN 5.2 . I can install a plugin via a deb, but Plugin Manager will not display the plugin. It would appear that the processing for non mainstream linux releases does not always work.

bdbcat, 2020-07-23:
Jon, many other users are using rPI, openPlotter, and OCPN5.2, without difficulty. All the master catalog plugins load and run, according to many reports.
What OS are you testing? What plugin fails for you?  Can you post that plugin's metadata file that you think SHOULD work?

jongough, 2020-07-23:
cat /etc/os-release:
```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
cat /etc/debian_version
```
9.11
```
Trying ODraw, relevant part of ocpn-plugins.xml
```
  <plugin version="1">
    <name> OCPN Draw </name>
    <version> 1.8.0.13 </version>
    <release> 13 </release>
    <summary> Drawing Plugin to allow extra objects to be drawn and used within OCPN </summary>
    <api-version> 1.16 </api-version>
    <open-source> yes </open-source>
    <author> Jon Gough </author>
    <source> https://github.com/jongough/ocpn_draw_pi </source>
    <description>
OCPN_DRAW (OD) is a plugin that extends the capability of OCPN to include non-standard object: Boundaries, Boundary Points, Text Points, Dead Reckoning paths, Electronic Bearing Lines (EBL) and Guard Zones. The plugin also has an API to allow other plugins to directly interact with it.
</description>
    <target>raspbian-armhf</target>
    <target-version>9.4</target-version>
    <target-arch>armhf</target-arch>
    <tarball-url>
https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-prod/raw/names/ocpn_draw_pi-1.8.0.13-raspbian-armhf-9.4-stretch-armhf-tarball/versions/v1.8.0.13/ocpn_draw_pi-1.8.0.13-raspbian-armhf-9.4-stretch-armhf.tar.gz
</tarball-url>
    <info-url> https://opencpn.org/OpenCPN/plugins/draw.html </info-url>
  </plugin>
```
Installed vi deb: `dpkg -s ocpn_draw_pi-1.8.0.17-raspbian-9.4-stretch-armhf` shows:
```
Package: ocpn_draw_pi-1.8.0.17-raspbian-9.4-stretch-armhf
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 2883
Maintainer: Jon Gough
Architecture: armhf
Version: 1.8.0.17-ov50
Depends: opencpn, bzip2, gzip
Recommends: xcalib, xdg-utils
Description: ocpn_draw_pi PlugIn for OpenCPN
```

bdbcat, 2020-07-23:
Jon...
That all looks good.  Same as I see on rPI3/Stretch.
And I see all plugins, including Draw 1.8.0.17, which I think is in the master catalog now.
Do you see any other plugins after updating the catalog?  oeSENC, for example?

jongough, 2020-07-23:
This is a screen shot of the plugins page:
[image]
All the plugins available are from deb files, none of them are available from Plugin Manager
Here is part of the log:
```
15:44:01.436 MESSAGE pluginmanager.cpp:1075 Loading plugins from dir: /usr/lib/opencpn
15:44:01.437 MESSAGE pluginmanager.cpp:1101 PlugInManager searching for PlugIns in location /usr/lib/opencpn
15:44:01.440 MESSAGE pluginmanager.cpp:1138 Found 30 candidates
15:44:01.440 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/librotationctrl_pi.so
15:44:01.445 MESSAGE pluginmanager.cpp:2098 Plugin is compatible: true
15:44:01.445 MESSAGE pluginmanager.cpp:2161 PlugInManager: Loading PlugIn: /usr/lib/opencpn/librotationctrl_pi.so
15:44:01.449   ERROR dlunix.cpp:162 /usr/lib/opencpn/librotationctrl_pi.so: undefined symbol: _ZN11wxJSONValueixERK8wxString
15:44:01.449 MESSAGE pluginmanager.cpp:2220    PlugInManager: Cannot load library: /usr/lib/opencpn/librotationctrl_pi.so 
15:44:01.450 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libgradar_pi.so
15:44:01.450 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libvdr_pi.so
15:44:01.450 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/librtlsdr_pi.so
15:44:01.450 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libsar_pi.so
15:44:01.450 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libsweepplot_pi.so
15:44:01.450 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libweatherfax_pi.so
15:44:01.451 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libwatchdog_pi.so
15:44:01.451 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libstatusbar_pi.so
15:44:01.451 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libaisradar_pi.so
15:44:01.451 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libprojections_pi.so
15:44:01.451 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libs63_pi.so
15:44:01.451 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libotcurrent_pi.so
15:44:01.451 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libDR_pi.so
15:44:01.452 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/liblauncher_pi.so
15:44:01.452 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libofc_pi.so
15:44:01.452 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libobjsearch_pi.so
15:44:01.452 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libweather_routing_pi.so
15:44:01.452 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libpypilot_pi.so
15:44:01.467 MESSAGE pluginmanager.cpp:2098 Plugin is compatible: true
15:44:01.468 MESSAGE pluginmanager.cpp:2161 PlugInManager: Loading PlugIn: /usr/lib/opencpn/libpypilot_pi.so
15:44:01.473   ERROR dlunix.cpp:162 /usr/lib/opencpn/libpypilot_pi.so: undefined symbol: _ZNK11wxJSONValue9HasMemberERK8wxString
15:44:01.473 MESSAGE pluginmanager.cpp:2220    PlugInManager: Cannot load library: /usr/lib/opencpn/libpypilot_pi.so 
15:44:01.473 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libiacfleet_pi.so
15:44:01.473 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libpolar_pi.so
15:44:01.474 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libcelestial_navigation_pi.so
15:44:01.474 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libclimatology_pi.so
15:44:01.480 MESSAGE pluginmanager.cpp:2098 Plugin is compatible: true
15:44:01.480 MESSAGE pluginmanager.cpp:2161 PlugInManager: Loading PlugIn: /usr/lib/opencpn/libclimatology_pi.so
15:44:01.485   ERROR dlunix.cpp:162 /usr/lib/opencpn/libclimatology_pi.so: undefined symbol: _ZN11wxJSONValueaSEi
15:44:01.485 MESSAGE pluginmanager.cpp:2220    PlugInManager: Cannot load library: /usr/lib/opencpn/libclimatology_pi.so 
15:44:01.485 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libsquiddio_pi.so
15:44:01.485 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/liboesenc_pi.so
15:44:01.486 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libbr24radar_pi.so
15:44:01.486 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libocpndebugger_pi.so
15:44:01.486 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/liblogbookkonni_pi.so
15:44:01.486 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libroute_pi.so
15:44:01.486 MESSAGE pluginmanager.cpp:1145 Checking plugin candidate: /usr/lib/opencpn/libwindvane_pi.so
15:44:01.495 MESSAGE oesenc_pi.cpp:917 oesenc_pi:  Calculated pix/mm = 3.78698
15:44:01.781 MESSAGE chartsymbols.cpp:950 oeSENC_PI RasterSymbols texture: 101
15:44:01.898 MESSAGE OCPNPlatform.cpp:1010 ChangeLocale: Language load for:  en_US
```

leamas, 2020-07-23:
Debian:

In  ocpn-plugins.xml there is  are compiled versions for debian-10/x86_64. Neither oldstable stretch/9 or sid/11 is supported. Nor is other architectures.

However, it *looks* like plugins for xenial are usable on stretch. Likewise, plugins for focal seems usable on sid. IMHO, both of these are important. Oldstable because, well, it's oldstable. And sid is the upcoming bullseye, and we need to support this to have things in place when bullseye is released.

The outdated documentation of the Debian download (/#1987) makes it even more interesting.

I'm handling this using a shell wrapper like:
```
#!/bin/sh
  
machine=$(uname -m)
if [ "$machine" != "x86_64" ]; then
    exec /usr/libexec/opencpn
fi

release=$(lsb_release -sc)
case $release in

    sid) TARGET='ubuntu-gtk3:20.04'
        ;;
    buster) 
        exec /usr/libexec/opencpn
        ;;
    stretch) TARGET='ubuntu:16.04'
        ;;
    *)  echo "Version $release is not supported" >&2
        exit 2
        ;;
esac
OPENCPN_COMPAT_TARGET=$TARGET exec /usr/libexec/opencpn                                                    
```

Note that from a downstream perspective I actually prefer having this in a shellscript. The situation will change when bullseye is branched, and possibly also after plugin updates. In such situations is easier to adapt by modifying the wrapper than to meddle with the mess in isPluginCompatible().

jongough, 2020-07-23:
[https://github.com/OpenCPN/OpenCPN/issues/1997#issuecomment-662530683](https://github.com/OpenCPN/OpenCPN/issues/1997#issuecomment-662530683) The platform distribution packages already handle issues like this, i.e. ubuntu packages quite happily install on mint. So there should not be an issue installing packages, via Plugin Manager, into 'downstream' releases. Could we do the same checking as the package managers do to work out if it is installable? We cannot generate packages for every possible platform and most of them will have common bases, so lets try and use that.

We could include in the XML the required packages that must be installed for the plugin to work and use that as a sanity check.

leamas, 2020-07-23:
Debian: 

Adding to above: Using ubuntu plugins on Debian is kind of upside-down; after all, Ubuntu is a Debian derivative, not the other way around. It might have been better to use Debian plugins on Ubuntu, But then again, this is in hindsight.

leamas, 2020-07-23:
Downstream:
> We cannot generate packages for every possible platform and most of them will have common bases, so lets try and use that.

Indeed.

>Could we do the same checking as the package managers do to work out if it is installable?

A similar test is done when we load the plugin. The "failed to load" condition is basically similar to "package cannot be installed" using apt (talking Debian).

The real problem lies IMHO in the process. We can create a version and test it against a reference platform, currently Ubuntu. This means that we need to track the Ubuntu releases, and build new plugins when a new (LTS) release is upon us.

However, the downstream Mint situation is different. Since we cannot keep track of what or when Mint releases anything, the Mint packaging model should be used. Which, as I understand it, basically is to either use the Ubuntu packages as-is or to modify them as required. In this case, this would mean modifying the opencpn  Mint .deb package so it loads the correct plugins.  

This is actually trivial using the official package, which has a source package. But when it comes to the PPA packages there is no source package, and modifying becomes  harder.

One thing we could do is to modify the PPA packages to use the same model as the official. Besides supporting derivates like Mint with a source package it would also save some work.Still, it's a learning curve. 

I'm definitely not sure about the proper route. But basically, I think Mint and others needs to be decoupled from the main opencpn code, making their own adaptations of opencpn as required. Main opencpn should support this by providing hooks etc. but without adding knowledge of actual downstream distros and their needs.

transmitterdan, 2020-07-23:
Threads like this are useful references when people ask why most of the world has not abandoned Windows for Linux...

It seems as if the promise of Linux, for the general public, is being lost forever amid a morass of incompatible distributions and uncoordinated release cycles.  As Yoda might say, "entropy is high with this one".  As a "free" OS, Linux appears to require a lot of human capital.

Please don't be offended.  My thoughts just flow through my keyboard because it seems I can't control my fingers some days...

leamas, 2020-07-23:
@transmitterdan:

Indeed. Still, we are basically just supporting Debian/Ubuntu + Flatpak, so we will probably eventually  clean up this mess. But it will take some time. 

And, remember, the diversity is also a strength. It has not stopped Linux in the server space, for sure.  But in the end, this discussion belongs somewhere else...

leamas, 2020-07-23:
Downstream:

@jongough @bdbcat 

Actually, there *is* source packages available for the PPA .debs (sorry, my bad). This means that it should be straight-forward to handle the Mint situation be creating specific Mint packages  based on the Ubuntu ones.. This would then be a task for whoever wants to support Mint. 

For others running Ubuntu derivatives we could just make the CompatOS configuration variable visible in the GUI + add documentation. 

Boiling down to two ways to support derivates:  specific opencpn packages or user configuration. None requires any explicit knowledge from the opencpn sources, which IMHO is the way to go

bdbcat, 2020-07-23:
@jongough 
Your log shows OCPN considering all of the legacy unmanaged plugins installed from .deb's  Fair enough.
But roll that screenshot down.  What managed plugins are available, and not yet installed?

bdbcat, 2020-07-23:
@leamas 
I am pretty sure we can support debian 9/10/11 using a small tweak to the source code, thus avoiding a shell script.  I'd like to do that in 5.2.x branch.
Then, we simply declare that anything downstream of  debian (except Ubuntu) shall employ a script to set the environment, if necessary.
Yes, there will be some not-so-pretty code in isCompatible().  But maybe not truly awful....

leamas, 2020-07-23:
> I am pretty sure we can support debian 9/10/11 using a small tweak to the source code, thus avoiding a shell script. I'd like to do that in 5.2.x branch.

Fair enough. This means that we will be able to drop the shell wrapper in official Debian 5.2.1, but still needs it in 5.2.0.

> Then, we simply declare that anything downstream of debian (except Ubuntu) shall employ a script to set the environment, if necessary.

Basically agreed.  But how should users get the proper script? The reasonable way would be that there was a Mint-specific opencpn package setting up things correctly (using Mint as an example).

For users on derivates without fixed packages it might make sense to make the  CompatOS configuration variable visible in the GUI...

> Yes, there will be some not-so-pretty code in isCompatible(). But maybe not truly awful....

As long as it's only about Debian: agreed.

leamas, 2020-07-23:
>  Hindsight is great....

The best of all sights...

bdbcat, 2020-07-23:
@leamas 
I committed code to master in order to try to use Ubuntu plugins with debian 9 and 11, without need of a shell script.  Tested here on Stretch, works OK.  I think debian 10 should be OK as released.
Please have a look on sid, and other interesting variants.  Enable loglevel-debug for lots of noisy but useful diagnostics from isCompatible().

leamas, 2020-07-23:
@bdbcat: sid looks OK. Don't have any other VM where I stay right now. I agree that D10 should be ok as released.-

bdbcat, 2020-07-23:
@leamas 
OK, I will cherry-pick some stuff from master into 5.2.x, tag the results as 5.2.1, and then we can move ahead with debian "official" push and acceptance?

jongough, 2020-07-24:
[https://github.com/OpenCPN/OpenCPN/issues/1997#issuecomment-663025964](https://github.com/OpenCPN/OpenCPN/issues/1997#issuecomment-663025964) I have just upgraded to OpenPlotter 2.3.0 and had to do an upgrade from Stretch to Buster as the underlaying pi install. Have re-downloaded and installed OCPN and refreshed the list of plugins. Here is the screenshot after that:
[image]
I has to install oeSENC via the deb file provided from the OpenPlotter installation software.
The log of this run is provided below
[opencpn.log.pi.zip](https://github.com/OpenCPN/OpenCPN/files/4969753/opencpn.log.pi.zip)

jongough, 2020-07-24:
The build process used by frontend2 generates the code which is used in both Plugin Manager and, in this case, deb installation files. It does this in the same build at the packaging stage and three files are uploaded to CloudSmith, the xml, the gz and the deb file. The deb file installs without a problem the xml/gz file does not.
This is a screenshot after installing this file from CloudSmith `https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-prod/raw/names/opencpn-package-ocpn_draw-1.8.0.17-raspbian-armhf-10-buster-armhf.deb/versions/v1.8.0.17/ocpn_draw_pi-1.8.0.17-raspbian-10-buster-armhf.deb'
[image]
This is the extract from the ocpn-plugins.xml file:
```
  <plugin version="1">
    <name> OCPN Draw </name>
    <version> 1.8.0.17 </version>
    <release> 17 </release>
    <summary> Drawing Plugin to allow extra objects to be drawn and used within OCPN </summary>
    <api-version> 1.16 </api-version>
    <open-source> yes </open-source>
    <author> Jon Gough </author>
    <source> https://github.com/jongough/ocpn_draw_pi </source>
    <description>
OCPN_DRAW (OD) is a plugin that extends the capability of OCPN to include non-standard object: Boundaries, Boundary Points, Text Points, Dead Reckoning paths, Electronic Bearing Lines (EBL) and Guard Zones. The plugin also has an API to allow other plugins to directly interact with it.
</description>
    <target>debian-x86_64</target>
    <target-version>10</target-version>
    <target-arch>x86_64</target-arch>
    <tarball-url>
https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-prod/raw/names/ocpn_draw_pi-1.8.0.17-debian-x86_64-10-buster-tarball/versions/v1.8.0.17/ocpn_draw_pi-1.8.0.17-debian-x86_64-10-buster.tar.gz
</tarball-url>
```
The whole file is attached
[ocpn-plugins.xml.pi.zip](https://github.com/OpenCPN/OpenCPN/files/4969781/ocpn-plugins.xml.pi.zip)

bdbcat, 2020-07-24:
Jon...
Please try running opencpn with "--loglevel=debug".
Go to plugins page, refresh the catalog, and then close OpenCPN.
Post the logfile.  It will contain debug messages related to plugin catalog scan.

Thanks
Dave

bdbcat, 2020-07-24:
Jon...
I am assuming you are running Raspbian, official rPi download?
Is this correct?

The plugin XML file for draw_pi on Raspian Buster (armhf) looks like this:
```

 <plugin version="1">
    <name> OCPN Draw </name>
    <version> 1.8.0.17 </version>
    <release> 17 </release>
    <summary> Drawing Plugin to allow extra objects to be drawn and used within OCPN </summary>
    <api-version> 1.16 </api-version>
    <open-source> yes </open-source>
    <author> Jon Gough </author>
    <source> https://github.com/jongough/ocpn_draw_pi </source>
    <description>
OCPN_DRAW (OD) is a plugin that extends the capability of OCPN to include non-standard object: Boundaries, Boundary Points, Text Points, Dead Reckoning paths, Electronic Bearing Lines (EBL) and Guard Zones. The plugin also has an API to allow other plugins to directly interact with it.
</description>
    <target>raspbian-armhf</target>
    <target-version>10</target-version>
    <target-arch>armhf</target-arch>
    <tarball-url>
https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-prod/raw/names/ocpn_draw_pi-1.8.0.17-raspbian-armhf-10-buster-armhf-tarball/versions/v1.8.0.17/ocpn_draw_pi-1.8.0.17-raspbian-armhf-10-buster-armhf.tar.gz
</tarball-url>
    <info-url> https://opencpn.org/OpenCPN/plugins/draw.html </info-url>
  </plugin>

```
Note target, target-version, and target-arch.

jongough, 2020-07-24:
I did include the whole ocpn-plugins.xml file so you could see all the entries, this is the one for raspbian-armhf-10-buster:
```
  <plugin version="1">
    <name> OCPN Draw </name>
    <version> 1.8.0.17 </version>
    <release> 17 </release>
    <summary> Drawing Plugin to allow extra objects to be drawn and used within OCPN </summary>
    <api-version> 1.16 </api-version>
    <open-source> yes </open-source>
    <author> Jon Gough </author>
    <source> https://github.com/jongough/ocpn_draw_pi </source>
    <description>
OCPN_DRAW (OD) is a plugin that extends the capability of OCPN to include non-standard object: Boundaries, Boundary Points, Text Points, Dead Reckoning paths, Electronic Bearing Lines (EBL) and Guard Zones. The plugin also has an API to allow other plugins to directly interact with it.
</description>
    <target>raspbian-armhf</target>
    <target-version>10</target-version>
    <target-arch>armhf</target-arch>
    <tarball-url>
https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-prod/raw/names/ocpn_draw_pi-1.8.0.17-raspbian-armhf-10-buster-armhf-tarball/versions/v1.8.0.17/ocpn_draw_pi-1.8.0.17-raspbian-armhf-10-buster-armhf.tar.gz
</tarball-url>
```

jongough, 2020-07-24:
Here is the log of OCPN with --loglevel=debug
[opencpn.log.pi.debug.zip](https://github.com/OpenCPN/OpenCPN/files/4970123/opencpn.log.pi.debug.zip)

bdbcat, 2020-07-24:
Jon...
From the logfile:
`13:45:53.774   DEBUG PluginHandler.cpp:224 Plugin compatibility check: OCPN Draw  OS:ubuntu-armhf  Plugin:raspbian-armhf`

This suggests that you system is detected as "ubuntu-armhf". There are no plugins that match that target.
1.  Did you set some environment, override for the os, such as "OPENCPN_COMPAT_TARGET", or "CompatOS=.." in config file?
2.  Let us see /etc/os-release, for this system.  I know you posted before, for stretch.  Now, for buster?

Here is mine, for buster/armhf:
```

pi@raspberrypi:~/Projects/opencpn/build $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

leamas, 2020-07-24:
> and then we can move ahead with debian "official" push and acceptance?

Hm... it might actually complicate things. The package is now waiting review, and each new upload to mentors basically is a delay. And, isn't 5.2.1 is a bit premature?

If you cherry-pick the patch into 5.2.x I could certainly add that patch to the 5.2.0 package if you feel strong about it. It would be better than a new release, but is still a delay. OTOH, delaying a day or two might be worth it, we'll probably have to wait weeks before it's reviewed.

jongough, 2020-07-24:
My current /etc/os-release:
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

bdbcat, 2020-07-24:
@leamas 
Just so you know, I am preparing 5.2.1 for Android release.  I am working diligently to avoid a separate Android branch, as the eventual merge with ongoing master branch development is always a true pain.  Also moving toward a CI/CD process for Android.
So, the 5.2.x branch will get some safe Android specific stuff.
Seems that we definitely need the debian specific plugin patch in the debian release, no?  So I thought I would add that to 5.2.x as well.
Finally, I do not think that we want a debian 5.2.0 that is not functionally the same as current 5.2.0 release.
That's my thinking...

bdbcat, 2020-07-24:
Jon...
I confirm the bug.  Relates to Ubuntu PPA build-bot definitions.
New PPA version coming....

leamas, 2020-07-24:
> Finally, I do not think that we want a debian 5.2.0 that is not functionally the same as current 5.2.0 release.

Definitely so. But the shell wrapper *is* functionally the same as patching the code w r t Debian 9 and 11.

So, again: if you push your patch for the Debian compatibility to 5.2.x I can cherry-pick it into the 5.2.0 release. It will delay the review process, but there are no other drawbacks.  Just let me know if you want me to do it. 

As for the 5.2.1 release, let's prepare in good order, taking what time it takes. As long as the review hasn't started, I can add patches to the 5.2.0 release (causing new delays) if we deem it necessary.  

When  the review has started I can probably agree with the reviewer to use 5.2.1 instead if it's available.

bdbcat, 2020-07-24:
@leamas 
There will be a 5.2.1 Release for Android, probably before we want to re-release for mainstream.
So, for mainstream, the next release point will be 5.2.2, or even 5.4.0 (new branch from master) if we have no critical hot-fixes to make.

On official debian, I agree with you.  Let us (you) cherry pick from 5.2.x, just the very nicest cherries.
So far, only the debian plugin match patch seems critical.  Do you see anything else?

leamas, 2020-07-24:
> So far, only the debian plugin match patch seems critical. Do you see anything else?

For the moment,  I don't see anything at all in v5.2.x, the tip is  the release tag.  So, hard to tell ;)

jongough, 2020-07-24:
It looks like the issue may be the isCompatible() function as for focal 20.04 it does not appear to handle the 'gtk3' text in the PluginOS name. The only plugin that shows up is NmeaConverter_pi and it has PluginOS:ubuntu-x86_64 whereas every other plugin has PluginOS:ubuntu-gtk3-x86_64 . The question is should focal have gtk3 in the name or not? If not then it breaks the naming standard, if yes then the PluginHandler needs to cater for this.

rgleason, 2020-07-24:
I thought you guys had decided this, and that when we use gtk3 we should have gtk3 in the name for consistency. That was Bdbcat's decision and that is what Jon programmed.  The fact that nmeaconverter works is nice, but it is wrong in my opinion.

bdbcat, 2020-07-24:
Jon...
I thought we were talking about bionic/buster on ARM.....
Is there some trouble with focal?  Where?  More detail needed, please.

@rgleason
Please answer email. What is your (blocked) ip?

jongough, 2020-07-24:
The original issue was with linux mint 20 and not installing/showing plugins. I thought the issue with ARM was related as it happened a the same time so I added it to this issue. I have not found why plugins are not showing up in mint20, which is to do with the name. You have found the issue with ARM which is to do with the build of OCPN. Both are in the checking of compatibility for plugins. So, in hindsight, this should have been two different issues, but....

rgleason, 2020-07-24:
Dave, responded via email several times.... it is 
in Mystic CT   66.232.246.38

bdbcat, 2020-07-24:
Rick...
Did not get any emails
Your IP was block, now cleared.
Dave

bdbcat, 2020-07-25:
Jon...
On Mint 20 (focal base)...
If you don't want to use flatpak, I understand, since your normal development environment is Mint, and you have much plugin work going on.
So,then we will need to consider this an Advanced Use Case.
You can either use a config file entry to directly set the "CompatOS" values, or an equivalent environment variable.

For your testing, please try this in OCPN config file:
[Settings]
CompatOS=ubuntu-gtk3
CompatOsVersion=20.04

jongough, 2020-07-26:
It took a while as I needed to get wxWidgets 3.0.5 built and installed before I could run the debugging environment. The two settings you suggested did work. I am currently creating a clean Mint 20 and Ubuntu 20.04 environment to test running as a user. Will see tomorrow if the current builds work like that.

bdbcat, 2020-07-27:
@jongough 
FYI, for raspberryPi, Buster:  The current PPA build (5.2.0.8) has been corrected for the Ubuntu build bot problem in cross-build environments.  Please test, if possible,

bdbcat, 2020-07-27:
@leamas 
Branch 5.2.x has been updated with my idea of the minimum changes needed for the "official" debian build.  I would be happy if this branch propagated to debian, and eventually to ubuntu as the canonical OCPN 5.2.0 release.
Unless, of course, you have any changes from recent master commits you judge really necessary, and absolutely safe, without further testing.
Comments?

jongough, 2020-07-27:
[https://github.com/OpenCPN/OpenCPN/issues/1997#issuecomment-664088085](https://github.com/OpenCPN/OpenCPN/issues/1997#issuecomment-664088085) Just installed this on my Pi 3B and it now sees the catalog and installs ODraw from cloudsmith. Thanks

leamas, 2020-07-27:
I see two patches,  6c2c4d34e and 0b83f1f71. No problem adding them, will do. The list of patches for 5.2.0 below, the first handling the missing docs is not in upstream, .

```
0001-build-Handle-missing-doc-files.patch
0002-debian-patches-Update-appstream-installation-path.patch
0003-CMakeLists-Adjust-udev-rule-install-path-1988.patch
0004-Remove-bogus-svg-file-1990.patch
0005-Build-Fix-cmake-missing-NDEBUG-flag-1991.patch
0006-Correct-managed-plugin-compatibility-test-in-cross-b.patch
0007-Allow-vanilla-debian-host-to-use-Ubuntu-managed-plug.patch
```

I see nothing in current master to be added to the Debian package.  OTOH. here is some stuff in master I think should be cherry-picked into v5.2.x ASAP for other reasons:

  - 982891027, a hard to reproduce FTBS error.
  - 3ab9c93da, a plain installation bug fixed.
  - 0b6d5152b, the flatpak sound issue . 

Will update the flatpak package shortly with the sound patch, would prefer to pick it from v5.2.x.

jongough, 2020-07-27:
[https://github.com/OpenCPN/OpenCPN/issues/1997#issuecomment-663964126](https://github.com/OpenCPN/OpenCPN/issues/1997#issuecomment-663964126) I have checked in a clean Mint 20 and it works OK. I will see if I can check out a clean Ubuntu 20.04.

bdbcat, 2020-07-27:
@leamas 
Do I understand that you agree that 5.2.x HEAD (with  6c2c4d3 and 0b83f1f now) is acceptable for debian submission? If so, I will tag it, and then we can move onward in cherry-picking more good stuff from master.

bdbcat, 2020-07-27:
@leamas 
Cannot find "0001-build-Handle-missing-doc-files.patch".  Was there a PR? Nothing pending now....

leamas, 2020-07-27:
> Do I understand that you agree that 5.2.x HEAD is acceptable for debian submission? 

No. What we have today is 5.2.0 + the patches listed in https://github.com/OpenCPN/OpenCPN/issues/1997#issuecomment-664141298. 

This means that the current package is identical to the 5.2.x branch, besides some patches 0002-..0005- which are required to handle the packaging toolchain. The package has also a non-upstreamable patch 0001- which handles the documentation which is stripped from the package. 

In short: If you want to synchronize the 5.2.x branch as much as possible with the Debian package we will need to import all of the patches in the package in the branch. I don't think this is a good strategy, some of them are minor things which does not belong to v5.2.x. And in any case, there is one patch which we cannot merge (0001-...)

So, we will need to handle the fact that downstream packaging applies patches to the package. Some of these can be upstreamed, and as of now all of them are at least in master (again, besides 0001-...).

Also, be aware of the process: there will be bugs reported in Debian. The natural way to handle such bugs is to add a patch to the package, and then upstream it. This is how packaging works.

So, in the end: there are patches in the package, this is normal and something we need to live with.

The fact that we cannot handle patches in the PPA builds is of course a problem. Is this what worries you?

leamas, 2020-07-27:
Again: we cannot merge 0001-... It handles the documentation which we strip from the Debian package, but otherwise in place.

bdbcat, 2020-07-27:
@leamas 
Here is my concern:
I will soon merge a bunch of Android-specific stuff (now in master) into v5.2.x, thus creating (tagged) 5.2.1 for Android. in branch 5.2.x  All these patches are "safe", mostly by conditional compilation.  All will be part of 5.4.x, whenever, because they are now in master.
But I did not want to lose the ability to identify the 5.2.0 debian build checkpoint, for support reasons.

If you can handle the debian submission process without a specific tag or branch, then I am happy.
From a higher level, I think we should stop fussing, if possible, and get the debian 5.2.0 release stable and pushed forward ASAP.

leamas, 2020-07-27:
I don't see any problems. The "upstream release tarball + patches" deal is just how packaging works. The thing I need is the v5.2.0 tag. For practical purposes we both need the 5.2.x branch which records bug-fixing, not destabilizing patches.

Please don't forget the patches I think should go into 5.2.x: https://github.com/OpenCPN/OpenCPN/issues/1997#issuecomment-664141298

The package is now in mentors at https://mentors.debian.net/package/opencpn. I agree that we should leave it there, so the process can proceed whenever someone finds time to review it.

bdbcat, 2020-07-28:
@leamas 
OK, I have cherry-picked the three patches from #1997
Will it disturb you now if I begin moving the discovered clean Android patches to v5.2.x at this point?  Maybe you want/need a tag right now?

jongough, 2020-07-28:
[https://github.com/OpenCPN/OpenCPN/issues/1997#issuecomment-664215130](https://github.com/OpenCPN/OpenCPN/issues/1997#issuecomment-664215130) Just checked on native Ubuntu 20.04 fresh install and it is working OK.

leamas, 2020-07-28:
@bdbcat :
> Will it disturb you now if I begin moving the discovered clean Android patches to v5.2.x at this point

No, not at all. The only thing  I use is the v5.2.0 tag. There is no direct impact of any branch. 

---

Just to explain the process: when  5.2.0 was tagged, I created a source tarball from it which then was imported to the master branch in the packaging. In the packaging, there are no direct changes to the sources from that point. The work done is just changes in the `debian/ ` directory.

Part of` debian/`  is `debian/patches`. Here lives the patches I have listed earlier. When there is a need to modify the sources, patches are added to this directory. They are then applied when the package is built. So, what I do from now w r t 5.2.0 is basically to maintain the patches in debian/, together with other stuff here. 

In other words, the packaging does not "see" the upstream git repo, it is only affected when I as the downstream packager do something like importing a patch from upstream to package.

Bottom line: don't worry ;)

bdbcat, 2020-07-28:
@leamas...
Thanks for the explanation.  
Now I understand the idea of the list of patches given above.
I will not worry....

Let's close this monster thread.....

leamas, 2020-07-28:
Hi,

> Let's close this monster thread.....

Agreed

closed 2020-07-28 by bdbcat