OpenCPN/OpenCPN #1999: Route, track & waypoint sync discussion

RooieDirk, 2020-07-24:
From what I hear of users, the syncing of between to instances op OCPN is cumbersome. Transfer using usb-sticks or even sending e-mails wit route attachments are common solutions.

My idea is to add a button to the routemanager ("Synchronize"). What is should do is make an gpx just like export, but instead of write to a file, send it as nmea sentences. The nmea sentences should be encoded with a 8 to 6bits ascii  the same way as AIS (binary) messages do. This will need a custom nmea sentence header ("$OCPN"?)

I'm aware that this is far from an efficient way of transporting, but the advantage is that is always works. Never mind  if there is a nmea multiplexer in the route between the two (or more) OCPN instances. If you add a connection over the internet it should also work.
Please shoot or add idea's
Dirk

rgleason, 2020-07-24:
See the manual here:

https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:advanced_features:send-wpt-rte-over-ip
(Note there is something wrong with the images that I am trying to fix. The opencpn.org server has blacklisted my ip now for trying to figure out what is wrong.)  - so be careful, you might be blacklisted too.  Justt wait 2 days and it will get better.

Also see the Forum Thread which might be safer
https://www.cruisersforum.com/forums/f134/sending-waypoints-routes-over-ip-177080-2.html

But the wiki was a distillation.

bcn2, 2020-07-25:
Dirk,

you might consider RTZ as a standard interchange format. Already supported by a plug-in:
https://www.cruisersforum.com/forums/f134/rtzassist-opencpn-is-not-an-ecdis-233949.html?highlight=RTZ

The ECDIS standard IEC61174 defines how to encapsulate RTZ for ECDIS route interchange:
https://webstore.iec.ch/preview/info_iec61174%7Bed4.0%7Den.pdf  (Annex "S")

Hubert

Rasbats, 2020-07-25:
Sorry... RTZassist is not a plugin for OpenCPN. It is a self contained wxWidgets app. But the code could be incorporated in a plugin or the main OpenCPN code base. But we still have the problem of the transport route. And the non-recognition of 'WPL' and 'RTE' by OpenCPN.

Mike

rgleason, 2021-04-22:
This should be tagged as Discussion  (and moved in the future to Github Discussion)

closed 2021-04-23 by nohal