OpenCPN/OpenCPN #2002: Plugin catalog version is not displayed.

leamas, 2020-07-25:
As of screenshot below, after updating the catalog. No version info.

[image]

leamas, 2020-07-25:
Probably related: message below each time I start opencpn. Updating the catalogue makes no change.

[image]

leamas, 2020-07-26:
hm... The data displayed is catalogData in PluginHandler, as returned by GetCatalogData(). However, nothing ever sets this variable.

Nasty, indeed.

bdbcat, 2020-07-26:
Hmm indeed.
We never implemented any kind of catalog versioning logic. Assumed the user always wanted the HEAD of the github branch.
Lets talk about use-cases.
1.  Why would one want a less-than-current catalog?  Any reason to keep online history?
2.  We do not want to automatically check the available catalog version vs the currently installed version.  This implies a net access, and we have an unwritten policy that we do not go out to the net without explicit user action, like a specific button push.

More?

leamas, 2020-07-26:
It's IMHO more of a communication and feedback thing. Talking to someone else (problems...), what is the version? And the simple date info shows if the catalog has changed. 

OTOH: we have a catalog version in the xml file (not that used) and a "last changed" date, automatically filled in. The latter makes a lot of sense.

leamas, 2020-07-26:
The patch looks simple:

```
 src/PluginHandler.cpp | 3 +++
 src/pluginmanager.cpp | 7 ++-----
 2 files changed, 5 insertions(+), 5 deletions(-)

diff --git a/src/PluginHandler.cpp b/src/PluginHandler.cpp
index d27a7b44d..8a18463fa 100644
--- a/src/PluginHandler.cpp
+++ b/src/PluginHandler.cpp
@@ -864,6 +864,9 @@ const std::vector<PluginMetadata> PluginHandler::getAvailable()
         file.close();
         auto status = catalogHandler->DoParseCatalog(xml, &ctx);
         if (status == CatalogHandler::ServerStatus::OK) {
+            catalogData.undef = false;
+            catalogData.version = ctx.version;
+            catalogData.date = ctx.date;
         }
     }
 
diff --git a/src/pluginmanager.cpp b/src/pluginmanager.cpp
index 75a46b8a5..4059b94d0 100644
--- a/src/pluginmanager.cpp
+++ b/src/pluginmanager.cpp
@@ -5586,12 +5586,9 @@ wxString CatalogMgrPanel::GetCatalogText(bool updated)
     //  Get the version from the currently active catalog, by which we mean
     //  the latest catalog parsed.
     auto pluginHandler = PluginHandler::getInstance();
-    std::string version = pluginHandler->GetCatalogData()->version;
+    std::string date = pluginHandler->GetCatalogData()->date;
     
-    catalog += _T("  ");
-    catalog += _("Version");
-    catalog += _T(" ");
-    catalog += wxString(version.c_str());
+    catalog += wxString("  ") + _("Last change: ") + " " + date;
     if (!updated) catalog += _T("  : ") + _("Please Update Plugin Catalog.");
     
     return catalog;
-- 
2.24.1

```

Giving 
[image]

BTW:  The wxWidgets documentation not only deprecates but actually advice against using the _T() macro:
https://docs.wxwidgets.org/3.0/group__group__funcmacro__string.html#ga7dfc2888539861afe6c4337ef315472b

We should probably at least avoid it in new code.

bdbcat, 2020-07-26:
Yes, using the date makes sense.  Needs improved message in the dialog.
I see you have already jumped ahead.

leamas, 2020-07-26:
I'm actually making an overhaul on this, a PR is under way. Give me some time...

leamas, 2020-07-26:
> . Needs improved message in the dialog.

At least some borders (fixed by me). I don't plan to meddle with it beyond that.

PR #2004 (2020-10-03): Catalog settings
Overhaul of the plugin catalog GUI handling:
  - Fixes for #2002 and #2003 
  - Simplified the main window to look like:
[image]
  - Added a new settings window like:
[image]

The  settings window requires that the CatalogExpert=1 setting is present. However, if not it displays some feedback on the situation in a messagebox.

The catalog part is more or less identical to the stuff in current plugins window.
The platform part digests available target:version from the plugins and let's user pick any of them, and see the result.. The dropdown also presents the number of plugins a [... cut]
closed 2020-10-03 by bdbcat