OpenCPN/OpenCPN #2003: Plugin compatibility to picky.

leamas, 2020-07-26:
In PluginHandler.ccc::isCompatible() we have

```
if (compatOS_ARCH  == plugin_os) {
        //  OS matches so far, so must compare versions
```
This test fails if the target in the XML file is for example ubuntu-gk3-x86_64. This is because compatOS_ARCH  is ubuntu-gk3-x86_64-x86_64 after tacking on the architecture field.

The basic problem is that we have not been crystal-clear about the architecture suffix in the target. In this situation, I think it would be best to be permissive and allow both forms, something like 
```

index ffee864f8..5b408d558 100644
--- a/src/PluginHandler.cpp
+++ b/src/PluginHandler.cpp
@@ -228,7 +228,7 @@ bool PluginHandler::isCompatible(const PluginMetadata& metadata,
     
     auto meta_vers = ocpn::split(plugin_os_version.c_str(), ".")[0];
 
-    if (compatOS_ARCH  == plugin_os) {
+    if (compatOS == plugin_os || compatOS_ARCH  == plugin_os) {
         //  OS matches so far, so must compare versions
 
         if (ocpn::startswith(plugin_os, "ubuntu")){

```

+ a similar fix some line below.

leamas, 2020-07-26:
While on it: Should we not drop the `// running OS may be "like" some known OS` part as a consequence of the statement about dowsntream distros? IMHO better to have something reasonably well defined which downstreams can adapt to then something which occasionally "works".

leamas, 2020-07-27:
@bdbcat: thoughts?

bdbcat, 2020-07-27:
@leamas 
1.  On relaxing the test condition:  I do not see any plugin in master catalog that has " target in the XML file is for example ubuntu-gk3-x86_64".  So, fixing a non-problem?
2.  On the "drop the // running OS may be "like" some known OS":  Agreed.  not needed if we push toward flatpak, or config file overrides for special cases.   The "like" stuff was added for Mint, but broke on Mint-focal.  Also could clean up population of os_detail structure in OCPNPlatform.cpp.

leamas, 2020-07-27:
@bdbcat: on the test condition:

```
$ grep 'target>' ocpn-plugins.xml  | sort | uniq
    <target>darwin</target>
    <target>debian-x86_64</target>
    <target>flatpak</target>
    <target>flatpak-x86_64</target>
    <target>mingw</target>
    <target>msvc</target>
    <target>raspbian-armhf</target>
    <target>ubuntu-arm64</target>
    <target>ubuntu-gtk3-x86_64</target>
    <target>ubuntu-x86_64</target>
```

So, they are there...

leamas, 2020-07-27:
@bdbcat: On 2:

Right. Also note that the #2004 PR provides tools for trying different settings  in the GUI (the second panel in the new Settings windows).

bdbcat, 2020-07-27:
@leamas 
On 1:  Sorry, my bad. Cannot type today. I need to walk the code and see how it is working today, since it fails the "too picky" test.  Must be then taking the "ubuntu" branch later...  If so, your patch clarifies things somewhat.  I'll come back.

leamas, 2020-07-27:
@bdbcat: The absolute superior tool to check things is actually the #2004 GUI.  It gives a helicopter perspective on all available target:version combinations, and the number of plugins for each of them. Try it, you'll like it ;)

leamas, 2020-08-25:
flatpak plugins fails unless they have` <target os>flatpak-x86_64</target-os>.`  Plain flatpak is not accepted.

leamas, 2020-08-26:
I have been working with the shipdriver plugin for a day or two. I have been forced to add the -x86_64 suffix to the` <target>` field, otherwise plugin is deemed incompatible on linux and flatpak.

As the code works today both the -arch suffix and the target-arch field are mandatory. 

OTOH, they are identical. We should IMHO sort out which one to keep (i. e., to be mandatory) and which one to be optional. The patch is about the suffix being optional, but it can be resolved the other way around as well.

leamas, 2020-08-27:
The underlying issue here is the lack of documentation. I have tried to document the current practise in the wiki at https://github.com/leamas/OpenCPN/wiki/Catalog#plugin. Quoting myself:

---
**target-arch** is the hardware platform, for example "x86_64" or "armhf". On msvc/windows, flatpak and darwin/macos it should always be *x86_64*. On linux, the same as the -arch suffix in target (see above).

---

Which means that the current test isn't too picky, it's just that the specification requires the architecture to be present in two places.  This is for now just a weird fact, so I rest my case.

We *do* need to address the specification and the related isCompatible(), but this was the wrong starting point. 

Closing.

closed 2020-08-27 by leamas

jongough, 2020-08-27:
Is this change going to force Flatpack on downstream distros like mint. If so this a retrograde step as the distribution is larger and slower than the native distribution and impacts performance on lower end systems and requires larger and faster internet connections for download. There are other ways to solve the issue if you look at the information that can be available from the downstream distros.

leamas, 2020-08-27:
So far we here are no changes, the mint situation is not in focus here. It's more about documenting the interface between the plugins and main OpenCPN, the arch and target fields in the catalag and the isCompatible() function.

That said, I doubt we can support mint with ubuntu packages. They are different distros with different release cycles, mixing packages carries all sort of risks.  It might work, but there are no guarantees. The only way to have supported debian packages on mint would be to set up a PPA with packages and plugins rebuilt for Mint, as discussed in #1997.   And until someone steps up and does this, flatpak is the supported alternative.  

I would love to be proven wrong, for sure. Until this happens we have defined ubuntu packages on mint as an "Advanced Usecase" requiring manual configuration. I have indeed submitted #2004 which should make it easier to deal with ubuntu packages on mint, but that is not to say it will be simple.

Perhaps things would be better if we had our initial releases on Debian, which both Mint and Ubuntu are based upon, and also moves more slowly. Dunno

jongough, 2020-08-28:
Frontend2 generates platform Install packages and places them in cloudsmith. It would be simple to enable plugin manager to install these for the platforms which are 'problematic'. 

Linux Mint installs Ubuntu Deb files with no issues through the apt and dpkg install programs.  As plugin manager is supplanting these install process it should be able to do the same job.

I have stayed away from doing anything with the plugin manager process to avoid conflicting changes. I did a local change to install Deb files by a minor addition to the XML and a few lines of code to handle Deb files. It does require admin access, but this can be elicited. When I get back to my development environment I can have a look at implementing both Deb and downstream processing.