OpenCPN/OpenCPN #2011: unreadable distance and bearing in measure

free-x, 2020-08-02:
Hi,

I'm using LinuxMint 20. If I try "measure" tool from right click, distance and bearing are unreadable (s. screenshot)
[image]

Regards
free-x

leamas, 2020-08-05:
hm... just out of curiosity: any difference using the flatpak version?

free-x, 2020-08-07:
what a flatpak? I did not install a flatpak

leamas, 2020-08-07:
no, but what if you do?

free-x, 2020-08-09:
Flatpak (after downloading 650MB using 2.3GB(!) on disk) version hasn't this behavior
[image]

leamas, 2020-08-10:
About the download size: yes, the initial download is huge. This is a one-time setup, subsequent package updates weighs in at about 50 MB. There is no free lunch, this is the price for the compatibility.

One basic question here is if Opencpn supports Linux  Mint 20.04 using the Ubuntu packages. We touched this issue also in #1997. As I understand it, this is far from clear. OTOH, we *do* support it using Flatpak (which, BTW, is sort of endorsed by the Mint community).

I can see two practical routes for the Mint support, which as can be seen in #1997 is not only about this issue:

  - Using Flatpak.
  - Someone steps up and creates proper Mint .deb packages distributed using a separate PPA.

As of now, there are no Mint packages available and flatpak is thus the way to go.

Thoughts?

free-x, 2020-08-10:
i think this is more because of the settings of gtk3 theme. I just don't know at which position to turn the screw
or can the behaviour be influenced somehow in the source code of OpenCPN (parameter of wxWidget?)

leamas, 2020-08-10:
Well, I think the basic message for Mint-specific problems must be the same as for other Linux distros besides the supported Ubuntu/Debian/Raspbian: these are supported using Flatpak. 

If you need a change such as modified dependencies of the .deb package (missing font?)  and/or a patch of the source code, the proper way would be a modified .deb package, distributed using a separate PPA. Note that even if you fix the issue described here, the Ubuntu .deb is hard to use due to #1997 (and possible other issues)

The basic problem is the process: Ubuntu first releases a LTS version such as 20.04, then we release the opencpn package for  this version, and at later point Mint releases a downstream release such as Mint 20 . We (i. e., opencpn) just cannot keep up with whatever changes Mint has done, it has to be handled in a Mint .deb package maintained together with Mint.

In the end, this is the very reason we need Flatpak or a similar packaging. It's the only way we can actually support anything on the vast universe of Linux platforms.

Just my 5 öre...

free-x, 2020-08-13:
i have now learned that the same package (built in a "focal" chroot) on a Mint 20 and Ubuntu 20.04+cinnamon desktop have different behavior. In the latter, bearing and distance are very nice to read

leamas, 2020-08-13:
Good to hear.

Now, you or someone else could make such packages available in a PPA and thus solve this particular problem. To be meaningful, such a package should also handle #1997 -- I did it using a shell wrapper which isn't that complicated.

Until such a PPA is available the solution for Mint users who don't want to compile their own package is to use flatpak.

From a more formal point of view I think this bug should be closed since the Ubuntu .deb package isn't expected to  work on Mint.

leamas, 2020-08-13:
BTW: Later on,  the official Debian opencpn 5.2.0 package will trickle down to the official Mint builds, and things will be resolved automagically. However, this is still in the  Debian review queue, don't hold your breath.

free-x, 2020-08-18:
I'm idiot. After change color of RouteLegInfoRollover to black works as expected.
flatpak is not required

closed 2020-08-18 by free-x