OpenCPN/OpenCPN #2012: cmake ignores build parameter and also falsly detects availability of wxWebView

ArjenR, 2020-08-03:
L.S.

My distribution of choice (Solus) does not have the libraries for wxwebview. But cmake build script still detects it as available. When I run the test manually it is clearly not available. 

I also tried to build using `cmake -DOCPN_USE_WEBVIEW=OFF`. But this is ignored by the build script.

As a work around I currently patch the build script to force the compile, but it should not be necessary.

leamas, 2020-08-03:
Hi,

When it comes to linux, our current approach is to support Debian/Ubuntu and flatpak. As for other distros like Solus we are thus in a "patches welcome" state, without substantial efforts beyond that. 

Have you tried the flatpak version which also offers some plugins?

As for your native Solus build:

  - If cmake indeed detects  wxWebview even  though the libraries does not exist this is most likely a Solus packaging problem where the headers are installed but not the libs. It could of course also be a bug the FindwxWidgets[1] cmake module which then should be filed against cmake

 - Older Debian version suffers similar problems, that's why the` OCPN_USE_WEBVIEW ` option exists. However, as could possibly be deduced from the name, it's a boolean flag and the correct usage is` -DOCPN_USE_WEBVIEW:BOOL=OFF`. This is part of the Debian CI builds, so it kind of works.

[1] https://cmake.org/cmake/help/v3.8/module/FindwxWidgets.html

ArjenR, 2020-08-05:
Hi, 

> Have you tried the flatpak version which also offers some plugins?
Yes I did. I recall I had some issues with it and moved to create a native package.

> the correct usage is -DOCPN_USE_WEBVIEW:BOOL=OFF
Thanks. This works.

> a "patches welcome" state
I think I read somewhere there is some extra to do for getting plugins to compile. Will look into this next.

closed 2020-08-05 by ArjenR

leamas, 2020-08-05:
if you have any issues with the flatpak version  please file a new issue. flatpak is the only practical way to support the new plugin system on distros like solus.