OpenCPN/OpenCPN #2015: Black background on OSX

arn0, 2020-08-05:
Hi,
After building 5.2.0 from master, I got a black background on some windows while running on OSX 10.13.3. This has fixed it:

diff --git a/src/navutil.cpp b/src/navutil.cpp
index eed3bd663..dc932c575 100644
--- a/src/navutil.cpp
+++ b/src/navutil.cpp
@@ -4832,7 +4832,7 @@ void DimeControl( wxWindow* ctrl, wxColour col, wxColour window_back_color, wxCo
     if ( depth == 0 ) {   // only for the window root, not for every child
         // If the color scheme is DAY or RGB, use the default platform native colour for backgrounds
         if ( !darkMode ) {
-            window_back_color = wxNullColour;
+            window_back_color = wxSystemSettings::GetColour(wxSYS_COLOUR_DESKTOP);
             col = wxSystemSettings::GetColour(wxSYS_COLOUR_LISTBOX);
             uitext = wxSystemSettings::GetColour(wxSYS_COLOUR_WINDOWTEXT);
         }

Regards,
Arno

nohal, 2021-04-23:
FWIW I was never able to reproduce this issue, @arn0 are you able to build the current master and check if you still see it?

bdbcat, 2021-05-03:
Closing, in-active

closed 2021-05-03 by bdbcat