OpenCPN/OpenCPN #2018: No default plugins in Xenial after upgrade 4.8.8->5.2.0

tai-fun, 2020-08-16:
None of the four default plugins appeared after upgrade.
Letting O create a new opencpn.conf file made no difference.

[opencpn.conf.upgrade52.txt](https://github.com/OpenCPN/OpenCPN/files/5080305/opencpn.conf.upgrade52.txt)
[opencpn.log.upgrade52.txt](https://github.com/OpenCPN/OpenCPN/files/5080306/opencpn.log.upgrade52.txt)

These files are copied just after the first run of the upgraded O.

tai-fun, 2020-08-16:
A similar issue here:https://www.cruisersforum.com/forums/f134/opencpn-version-5-2-0-released-237051-9.html#post3209203

leamas, 2020-08-16:
Seems like a problem building the debian package. On xenial, the system plugins are installed in ` /usr/lib/x86_64-linux-gnu/opencpn`, but are expected to live in` /usr/lib/opencpn` (since opencpn isn't a multiarch package).

You should be able to walk around the issue using `sudo ln -s /usr/lib/x86_64-linux-gnu/opencpn /usr/lib/opencpn`

The root cause is that the Debian builds in the PPA are broken. The correct value /usr/lib/opencpn is the default on Debian (see CMakeLists around line 566). However, the build is using the cdbs build system (generated from cmake), and it is possible that this sets up the multi-arch paths as default.  I have no idea how cdbs works, it basically legacy  -- nowadays dh is used.

One probably needs to tweak the PPA build somehow, I don't understand how.  Also, the first step would be to inspect the build logs on launchpad.

@bdbcat: thhoughts?

leamas, 2020-08-16:
BTW: trusty suffers from the same problem. focal and bionic looks OK.

bdbcat, 2020-08-16:
I'll take this.  Please stand by.
Dave

bdbcat, 2020-08-17:
Alec...
To make it work on Xenial, I must change...

From:
```

if (EXISTS /etc/debian_version AND NOT DEFINED CMAKE_INSTALL_LIBDIR)
  set(CMAKE_INSTALL_LIBDIR "lib")
endif ()
```
To:

```
if (EXISTS /etc/debian_version )
  set(CMAKE_INSTALL_LIBDIR "lib")
endif ()
```
Evidently on the build machine, or cdbs startup scrips, the CMAKE_INSTALL_LIB is set to "lib/x86_64-linux-gnu"

Comments?
Dave

leamas, 2020-08-17:
> [...] cdbs startup scrips, the CMAKE_INSTALL_LIB is set to "lib/x86_64-linux-gnu"

Indeed, I suspected something like that.

As for the patch it should IMHO not go into master or release branch. CMAKE_INSTALL_LIBDIR is an offical cmake variable, and those who builds opencpn rightfully expects this to be respected.

The best would probably be the regular packaging solution: add a patch to the trusty and xenial builds. But I guess this is not an option. Other alternatives:

  - Dive into the CPack and Cdbs docs to figure out how to reconfigure the build. It's basically about patching debian/rules and is probably possible (?)
  - Create separate git branch(-es)  for trusty and xenial PPA source,  apply the patch there and build from them.
  - Use a separate option like OLD_DEBIAN_LIBDIR_HACK to override the  CMAKE_INSTALL_LIBDIR value. If you do, I suggest that you put a comment with a reference to this bug in the code...

bdbcat, 2020-08-17:
Leamas..
A simple addition to debian/rules file for Ubuntu PPA build does the trick.
`DEB_CMAKE_EXTRA_FLAGS += -DCMAKE_INSTALL_LIBDIR=lib
`
That's one more arcane interface that I don't need to know any more about...
I'll publish to production PPA shortly.
Thanks
Dave

leamas, 2020-08-17:
Dave,

Looks like the best possible solution. You deserve a big, really big, golden medal for dealing with the cdbs documentation....

--a

tai-fun, 2020-08-19:
Just updated from the PPA, and all is well.
Closing this issue.

closed 2020-08-19 by tai-fun