OpenCPN/OpenCPN #2019: Flatpak path issue

hreuver0183, 2020-08-16:
While it might not be an OpenCPN/Flatpak problem but a general Flatpak problem:
I tried installing a chart and due to tight space on the partition I tried a symlink to another partition with more space.

Findings up till now:
- flatpak does not work with symlinks
- flatpak is very limited in using  a path outside of the user directory

Practical, this works:
/home/opencpn/Charts/
/home/2nd_drive/opencpn/

But this does not work:
/home/2nd_drive/Charts/
/home/2nd_driver/opencpn/Charts/ 

Is this a know problem or a general flatpak problem?
Basically I want to be free to place my charts and tidal data on a public partition available to all users on the system.
Even better would be read-only access, but with oesENC that might not be possible.

leamas, 2020-08-17:
TL;DR: the limited access is a feature, not a bug.

Basically, flatpak runs in a sandbox which by default has very limited access to the host and it's file system. It's a question about security and not least portability -- any dependency outside the sandbox is a  possible break on another system.

However, given how opencpn is used it is configured with full access to the home directory. This is basically the end of it, besides some read-only system directories like /usr/lib. In practice, this also includes symlinks which points outside of $HOME since the target cannot be accessed.

You can give the sandbox extended permissions on the ocmmand line. See for example https://askubuntu.com/questions/1086529/how-to-give-a-flatpak-app-access-to-a-directory

hreuver0183, 2020-08-20:
Thanks. Need to test this, might be useful.
If useful I will add a line to the OpenCPN Flatpak Documentation.

But I guess this thread has served it's purpose and will soon be closed: for OpenCPN in principle extended path's are not reachable from Flatpak.

leamas, 2020-08-20:
> If useful I will add a line to the OpenCPN Flatpak Documentation.

Great, thanks.!

>  for OpenCPN in principle extended path's are not reachable from Flatpak.

Note that it would be simple to add more directories to the list of directories by default accessible from the sandbox. The question is just if it could be justified. Thoughts?

BTW: "extended paths" in this context should be "paths outside the home directory", right?

In a longer perspective (wxWidgets 3.2) we will need to update the  flatpak platform i. e., an ABI break. At this point we will also migrate to gtk3 which will sort of solve this, since the flatpak version of gtk3 incorporates a flatpak portal. This basically means that when  user selects a  directory in a file selection dialog flatpak will open up access to this dir. 

But then again, this is in the long run.

hreuver0183, 2020-08-20:
The issue is more "can we put the charts at a place which is convenient for both native OpenCPN and Flatpak".
I don't think a lot of users will be asking for it. But it should be documented. (No change to OpenPCN).

With the GTK3 version in mind I will only look into basic capabilities, thanks for the hint.
I assume you aim to switch from GTK2 to GTK3.2 (bypassingt GTK3.0) to limit the needed programming effort. That would be indeed longterm (looks like a good starting point to me).

leamas, 2020-08-20:
Thanks for looking into the documentation, it is indeed required.,

If modifying the sandbox permissions doesn't work (I have not tested, for sure...) you should be able to work around the issue using a bind mount instead, making whatever directory you use fully accessible from $HOME. Linux has, after all, some cards up it's  sleeves  ;)

hreuver0183, 2020-08-30:
Took a little longer than planned since it was not top priority and I made some mistake on my system.

Documented and tested.
Issue closed.

closed 2020-08-30 by hreuver0183