OpenCPN/OpenCPN #2027: Plugins: Wrong architecture for win32 plugins.

leamas, 2020-08-25:
As of today, all win32 plugins have TARGET_ARCH  set to x86_64. This is plain wrong, and will bite us when we want to start with 64-bit windows builds.

However, 5.2.0 is out, and we need find a new arch for 64-bit windows plugins (win64?). Or find some other solution. It's a mess.

rgleason, 2020-08-25:
Alec, we are sheep, please lead us where we need to go, but consider that we are not going to recompile all plugins overnight and the previous ones should probably continue to work..

leamas, 2020-08-25:
> Alec, we are sheep

Let me, with all due respect, seriously doubt this statement. Not to mention what it would imply for my role (sheep dog?)

That said, we must of course be careful. 0.5.2 is a fact, and we cannot change anything fast. It's just a question of directions.

The situation is that x86_64 is used for i386 builds on windows. We cannot escape this.

One direction could be that we set up a goal to use win32 (and implied win64,  win32-arm). A first step could be to let main opencpn accept win32 as an alias for x86_64. After a cycle or two we could start moving plugins to win32. We are talking about years. OTOH I don't see the rush, we can use win64 or win32-arm when the need is there.

But, it's kind of sad.

rgleason, 2020-08-25:
Alec, we can't know everything at the beginning, or even at the end. It is an exploration.
Also I seem to recall  that you were considering "win32" at some point.
Your plan seems perfectly reasonable to me..but who am I. 
I am going to have to do all these plugins again once I figure out how svg icons work, so I could make the change then. Let us know.. OK?

leamas, 2020-08-25:
We must first change the main opencpn code to accept win32 (if that's the way to go). Then we must wait to at least 5.4, probably 5.6 before moving the plugins. So, hold your horses...

bdbcat, 2020-08-26:
Alec...
Why sad?
Observations:
1.  We have no idea what are the implications for OCPN on the general adoption of Win64, if that happens.  We can be pretty sure that Win32 apps will continue to function on WindowsX for the reasonable future.  MS will not abandon the legacy app base quickly.  Further, we have no idea whether wxWidgets will port to W64, and when. So, no rush to port OCPN to Win64.  
2.  For argument's sake, what could possible be the benefit to the user of porting OCPN to Win64?  We are almost always performance-limited by GPU/Display capabilities.  We are still intentionally throttled in GPU performance due to persistent bugs in Intel GPU models, especially the most popular legacy chips.
3.  Similar observation regarding win on arm.  More:  In order to run native arm code in Windows, the app must be build to a model called UWP.  This is a way of bundling an app with multiple runtime components, for each desired base architecture.  It is a requirement for some Windows storefront apps.  Otherwise, plain old win32 apps will run on Windows/Arm using an emulation layer.  Performance is unknown.  Win64 apps are not supported on Windows/Arm, yet.  And again, no idea on wxWidgets for native Windows/arm execution.
4.  Finally, as developers, we have many obligations to the user.  But we are not required to accurately predict the future.

So, again, no rush on any of this.  But good to think about now...

More here:
https://docs.microsoft.com/en-us/windows/uwp/porting/apps-on-arm

Dave

leamas, 2020-08-26:
I'm sad because we are stuck with something so embarrassing  inconsistent. 

I have also seen the very line of code involved, together with the comment (line above) why it is/was i386. Whoever did the change didn't even bother to change or drop the comment. Which makes me sad about the process, and that there wasn't any more eyes on this in the beta stage.

OTOH, I agree with you about the near-term practical consequences which are limited. In Sweden, the saying is  "there is no cow on the ice". So let's keep this issue open.

For the sake of discussion I propose that we patch main opencpn to accept 'win32' as an alias for x86_64 (obviously on Windows).

I will document the fact that all windows plugins for now should have arch set to x86_64, despite that they are 32-bit :(

leamas, 2020-08-26:
Updated docs at https://github.com/leamas/OpenCPN/wiki/Catalog#plugin-tags

leamas, 2020-08-28:
So we have concluded that in our specifications we label 32-bit builds as x86_64. Since it's  documented, it's not a bug but a feature.  Bug should be closed.

Here are several issues which basically points to IsPluginCompatible(). But any attempt to handle this need a better starting point.

leamas, 2024-01-07:
> The windows package uses architecture x86_64 taken from the builder machine and everything derived from it like filename of the tarball and target-arch metadata is wrong - the build product is 32bit x86 at the moment.

This has been discussed in #2027. The basic message is that labeling 32-bit windows plugins as x86_64 is a feature, not a bug.

nohal, 2024-01-07:
> This has been discussed in #2027. The basic message is that labeling 32-bit windows plugins as x86_64 is a feature, not a bug.

It is a bug, not a feature, and has to be fixed. Otherwise we will never get to win64 or arm64 build on Windows. We may get to the fix through leaving this mess working for now, but we must fix it.

leamas, 2024-01-07:
As you can see in #2027 I basically agree. Perhaps that bug should be reopened.

bdbcat, 2024-01-07:
Reopened.
Proposal:  
Why not make O5.9+ accept "win32" as well as "x86_64" for plugins targeting msvc, at least on x86 systems.
This would be a semi-permanent change, or at least until O584 is much older and fallen from general use (i.e. years).

We could also do this:
At some point after O5.10 releases, begin migrating msvc plugins to "win32" target.  When users complain (what happened to my plugins?), we gently steer them to O5.10.

Thoughts?

nohal, 2024-01-07:
Yes, this is the way to do it.
We will not need to hard distinguish between the architectures until we actually do win64 build, which by itself is going to be quite a bit of fun and involve changes in the plugin build system anyway.
At the same time we will probably want to also bump wxWidgets on Windows to 3.3+, because we will sure want some new features like dark mode support, so another big-ish change to the plugin build will be happening anyway.

All of this after 5.10, there we simply want to start accepting the correctly identified `x86` architecture besides the current `x86_64` error.

leamas, 2024-01-07:
> Why not make O5.9+ accept "win32" as well as "x86_64" for plugins targeting msvc, at least on x86 systems.

I agree, it's in line with my initial proposals here.

rgleason, 2024-01-07:
So, when we start compiling for 5.9, plugins should change to use "win32" instead of x86_64?
Please advise when this is anticipated. Thank you.

bdbcat, 2024-01-07:
There are no specific plugin builds required for O59.
All current plugins are (or should be) compatible with O584 and O59.  AS far as I know, there are no problems now.
If there are plugin problems discovered in Alpha testing of O59 (going on now), then those problems are bugs.

So, this means that we must wait for some time before building plugins for "win32".  Otherwise, they will not be compatible with O584.  Maybe 4Q/24?  Depends on O59 schedule.

If we have learned anything, we know that many users will not immediately adopt a new OCPN release.   There is a group that generally waits for "someone else" to prove the new release.  So we can expect O584 to be around for some time...

nohal, 2024-01-22:
I would probably keep this around to remind us that this is broken and both the plugin templates need fixing. It generally is more annoying for the testplugin template, where also the tarball and metadata file naming is wrong. For shipdriver, it really is just the target arch tag. Of course it works for now, but...

leamas, 2024-01-22:
testplugin has indeed other issues. On the list is a number of unused attributes (`<build-target>`, `<build-gtk>`).

Perhaps the correct solution is to file downstream bugs against testplugin and shipdriver?

leamas, 2024-01-22:
That it works now is no coincidence. We have been moving in the direction to have a single field identifying the ABI. I see no reason to leave this strategy, it keeps things simple. 

The MacOS situation was really that the universal plugins needed some field to for the actual architecture, a very special case which happened to map into the otherwise unused `<target-arch>` .  Nothing to make further conclusions from.

nohal, 2024-01-22:
The template plugins of course need separate issues filled and fixed. Still I would keep this one open as it is even explicitly being referenced in https://github.com/Rasbats/shipdriver_pi/blob/master/cmake/GetArch.cmake#L51-L53

leamas, 2024-01-22:
https://github.com/Rasbats/shipdriver_pi/issues/573

leamas, 2024-01-22:
https://github.com/OpenCPN/plugins/pull/1069

leamas, 2024-02-10:
@nohal. The reference to GetArch in shipdriver is fixed, so is the new ABI type in https://github.com/OpenCPN/plugins/pull/1069.  Time to close this?

nohal, 2024-02-15:
Pull request to the TP template was submitted and the logic somehow implemented otherwise, we are done here,

closed 2024-02-15 by nohal