OpenCPN/OpenCPN #2029: Unstripped flatpak plugins way too big

leamas, 2020-08-29:
The flatpak plugins are typically not stripped, which makes them excessively large. Using shipdriver as an example:

```
[ shipdriver-flatpak-2.4.7] $ ls -lh lib/opencpn/
-rwxr-xr-x 1 mk mk 5.3M Jan  1  1970 libShipDriver_pi.so
[ shipdriver-flatpak-2.4.7] $ strip lib/opencpn/libShipDriver_pi.so 
[ shipdriver-flatpak-2.4.7] $ ls -lh lib/opencpn/
-rwxr-xr-x 1 mk mk 268K Aug 29 13:45 libShipDriver_pi.so
```

I. e., the unstripped plugin is 5.3M, the stripped 268K. Even if the debugging symbols are occasionally useful, this price is just too high.

This seems ĺike  a flatpak-only problem. Should be easy to fix, but must be handled in each and every plugin by stripping the library before shipping. A natural point could be the repacking of the tarball which is done anyway.

leamas, 2020-08-29:
The following patch fixes it in shipdriver. It should work on most other plugins, the flatpak code is mostly the same.

This is now becoming big -- a separate file to include for the flatpak build would make sense

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index cdfd09c..a690029 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -135,6 +135,13 @@ if (OCPN_FLATPAK)
     COMMAND flatpak-builder --force-clean ${CMAKE_CURRENT_BINARY_DIR}/app
             ${FLATPAK_DIR}/org.opencpn.OpenCPN.Plugin.shipdriver.yaml
   )
+  add_custom_target(
+    flatpak-strip
+    WORKING_DIRECTORY ${CMAKE_BINARY_DIR}
+    COMMAND strip ./app/files/lib/opencpn/libShipDriver_pi.so
+    COMMENT "Stripping debug symbols."
+  )
+  add_dependencies(flatpak-strip flatpak-build)
   add_custom_target("flatpak-pkg")
   add_custom_command(
     TARGET flatpak-pkg
@@ -143,6 +150,7 @@ if (OCPN_FLATPAK)
       's|.*/files/|shipdriver-flatpak-${PACKAGE_VERSION}/|'
       ${CMAKE_CURRENT_BINARY_DIR}/app/files
   )
+  add_dependencies(flatpak-pkg flatpak-strip)
   return()
 endif ()
 
diff --git a/ci/control b/ci/control
index 9a214fe..7a567de 100644
--- a/ci/control
+++ b/ci/control
@@ -2,6 +2,7 @@ Source: opencpn
 Build-Depends: debhelper (>= 9),
  base-files (>>8) | mesa-common-dev,
  base-files (>>8) | libgl1-mesa-dev,
+ binutils,
  cmake3 | cmake (>> 3.5.0),
  curl,
  libarchive-dev,

```

rgleason, 2020-08-29:
Alec, I think these changes need to be made against Jon's testplugin_pi and become a new version number for the Frontend2 

Jon's repository is  https://github.com/jongough/testplugin_pi
Could you make a PR to his repository or to mine?  https://github.com/rgleason/testplugin_pi

Then Jon will commit and change the Frontend2 version and from there I will update all of Sean's plugins and others with the new frontend2 version.

Thanks.

leamas, 2020-08-30:
Downstream bugs:
  
  - radar_pi: https://github.com/opencpn-radar-pi/radar_pi/issues/148
  - test_plugin: https://github.com/jongough/testplugin_pi/issues/106
  - oesenc: https://github.com/bdbcat/oesenc_pi/issues/98
  - shipdriver: https://github.com/Rasbats/shipdriver_pi/issues/35

rgleason, 2020-08-30:
In shipdriver, the mingw build failed here  
https://app.circleci.com/pipelines/github/Rasbats/shipdriver_pi/32/workflows/f2fb418e-ae6a-457f-8e2f-478a77ef6023/jobs/176

Is this related to the changes proposed?

rgleason, 2020-08-30:
The shipdriver CMakeLists.txt file is very different from Frontend2
https://github.com/Rasbats/shipdriver_pi/blob/master/CMakeLists.txt

You want this code added starting at line 145

```
 find_program(TAR NAMES gtar tar)
  if (NOT TAR)
    message(FATAL_ERROR "tar not found, required for OCPN_FLATPAK")
  endif ()
  set(FLATPAK_DIR ${PROJECT_SOURCE_DIR}/flatpak)
  add_custom_target(
    flatpak-build ALL
    WORKING_DIRECTORY ${CMAKE_BINARY_DIR}
    COMMAND flatpak-builder --force-clean ${CMAKE_CURRENT_BINARY_DIR}/app  <----LINE 145
            ${FLATPAK_DIR}/org.opencpn.OpenCPN.Plugin.shipdriver.yaml
  )
  add_custom_target("flatpak-pkg")
  add_custom_command(
    TARGET flatpak-pkg
    COMMAND
      ${TAR} -czf ${pkg_tarname}.tar.gz --transform
      's|.*/files/|shipdriver-flatpak-${PACKAGE_VERSION}/|'
      ${CMAKE_CURRENT_BINARY_DIR}/app/files
  )
  return()
endif ()    <---DOWN TO HERE
```

Here is testplugin_pi  CMakeLists.txt
https://github.com/jongough/testplugin_pi/blob/master/CMakeLists.txt

There is no conditional for flatpak.   
Shouldn't this really be in the CI script for flatpak? 
So there would be some command to strip the tarball down.
I think that is where it would best be placed.

Also if RPath has been added in cmake/PluginSetup  you want the previous commit reverted 
https://github.com/rgleason/weather_routing_pi/commit/ba98759bf81dc8e7b1ae3206fd30b4ab13a4d54e#commitcomment-41879685

leamas, 2020-08-30:
> In shipdriver, the mingw build failed here

No,  not related. The  build might very well work if you rebuild. Seems like a transient error when accessing  the google repos.

leamas, 2020-08-30:
> The shipdriver CMakeLists.txt file is very different from Frontend2

Yes, on purpose. I can not really undertake to dive into frontend2. There is a bug filed against testplugin, let's see how Jon fixes this (I'm sure he can do this in "my" way or some other).

Personally, I think that the shipdriver setup has some advantages compared to frontend2. But then again, it is by no means a general solution for all plugins.

rgleason, 2020-08-30:
Alec,
Well, I don't care and I am not changing from frontend2, and I am not prepared to experiment with frontend2 at this point. Jon is away and not at his computer so it will be some time before this can be addressed.

I still think it is most appropriate to "strip" the tarball using the flatpak ci script so that other plugins are not affected.

Also I note that  Jon has added to PluginSetup the lines you wanted that now must be removed. Thus it has propagated to all of Seans plugins....

This has quickly turned into a mess and I don't really care about flatpak.

```
find_program(HAVE_LD_SO
    PATHS /lib NO_DEFAULT_PATH
    NAMES ld.so ld-linux.so.1  ld-linux.so.2
)

set(CMAKE_SKIP_BUILD_RPATH true)
if (HAVE_LD_SO)   # linux.
    message(STATUS "${CMLOC}Setting RPATH: \$ORIGIN:\$ORIGIN/..")
    set(CMAKE_BUILD_RPATH "\$ORIGIN;\$ORIGIN/..")
    set(CMAKE_INSTALL_RPATH "\$ORIGIN;\$ORIGIN/..")
endif ()
```

rgleason, 2020-08-30:
I will bite, what exactly are the advantages of the shipdriver frontend?

leamas, 2020-08-30:
I don't think it's a good idea that you switch, you have invested a lot in frontend2 which is supported by Jon. A support I cannot provide over time.

That said, if you just look at the code I think you will it easier to read (especially after next PR I'm preparing). It's also more modular when it comes to dependencies, note that they are just included in two lines. Makes  it easier to maintain and share. 

Again, this is not an attempt to replace frontend2. See it as a test bed for new/other ideas.

rgleason, 2020-08-30:
Thank you. There are always several ways to do something which are completely valid, and exploration leads to refinements.  I will be watching and reading. Can you label the Shipdriver example as Frontend3 and give it a version, to help others?

rgleason, 2020-08-30:
Alec since I am working on watchdog now. I find the rpath statement is not in pluginsetup, so that does not need reverting.  To get flatpak builds working should I just [add all of this](https://github.com/OpenCPN/OpenCPN/issues/2029#issuecomment-683422771)  into cmakelists.txt and see if it works?
 From   " find_program(TAR NAMES gtar tar)"  down to the end?

leamas, 2020-08-30:
No. If you don't understand the code it's better to let Jon for someone else who has read up on frontend2 fix it. My patch was a hint how to do which will more or less apply to most downstreams, but not frontend2.

leamas, 2020-08-31:
"BLushes" It turns out that there is an option in the manifest which disables the debugging symbols. The patch:

```
diff --git a/flatpak/org.opencpn.OpenCPN.Plugin.shipdriver.yaml b/flatpak/org.opencpn.OpenCPN.Plugin.shipdriver.yaml
index ff1a116..c97d954 100644
--- a/flatpak/org.opencpn.OpenCPN.Plugin.shipdriver.yaml
+++ b/flatpak/org.opencpn.OpenCPN.Plugin.shipdriver.yaml
@@ -22,6 +22,7 @@ modules:
       build-options:
           cxxflags: -DFLATPAK
           cflags: -DFLATPAK
+          strip: true
       sources:
           - type: git
             url: ..

```

Which replaces all the complicated stuff earlier. Never too late to learn...

leamas, 2020-09-01:
>  will bite, what exactly are the advantages of the shipdriver frontend?

Well, summarizing for whoever interested:
 - The cmake files have been uniformly formatted with cmake-format which makes them so much easier to read 
 - The ci files are overall smaller and IMHO also easier to read. One example is .circleci/config.
 - The MacOS build is cleaned up and runs faster
 - The Raspberry builds are more stable after changing docker image.
 - The builds runs faster when there are system libraries available for jsoncpp and tinyxml, affects slow rpi builder.
 - The cmake/ directory has been refactored, each file has a clear purpose and junk including useless deps is dropped,
 - Package deps are autogenerated to avoid hard to maintain static deps.
 - The main CMakeLists has a clear structure and a somewhat simplified setup.
 - Fixes for the RPATH and unstripped flatpak builds are in place.
 - The API is just a library, which makes it a clean solution for updates. I have published api-16 and api-17 at https://github.com/leamas/opencpn-api

There is probably more, and certainly bugs. But, it could be worse I guess.

I have no plans whatsoever to publish this as a general frontend. It's just an example how a plugin could be written. I do have a plan to rewrite the history to a set of usable patches. The *patches* branch on shipdriver contains cleaned up patches up to the point it is building. However, there is quite a large delta  after that point.

rgleason, 2020-09-01:
Thank you Alec, I am certain your improvements will be used and considered.

leamas, 2020-12-28:
This is now fixed, or at least nedled to the plugin downstreams. Closing

closed 2020-12-28 by leamas