OpenCPN/OpenCPN #2031: Q: Why are we distributing metadata files?

leamas, 2020-08-31:
Looking back at the plugin system, one question in my head: Why do we upload and distribute plugin metadata files?

Users consumes metadata either in the catalog or, when doing an import, as bundled in the tarball. So, in hindsight I see no user value in the uploaded metadata files. 

I think I know how we ended up this way, it's a about history but probably not that interesting.

Should we just drop this?

rgleason, 2020-08-31:
The reason we are distributing metadata files is because they are used to add to the opencpn/plugins metadata directory in order to make the master/beta catalog.  

Also Jon Gough has made a script named download_xml_bash.sh which makes this a very easy to do.

Since the metadata is also contained in each tarball, it might be possible to eliminate the metadata files but then it would be much harder to get the metadata files into opencpn/plugins catalogs.

Certainly we would want to modify the script to also extract the metadata file from the tarball, which would also take longer.

leamas, 2020-08-31:
Well, it's certainly not the reason they are there. OTOH, it's one use case today.

From this perspective, it could be seen as optional. If Jon want's to upload his XML files to someplace to be available for download I see no problem with that. But the point might be that this is not really mandatory, and plugins which are not part of this scheme like radar_pi doesn't need to upload the metadata. They just make PRs against OpenCPN/plugins.

Or?

rgleason, 2020-08-31:
Ok so you want to abandon the master catalog and just have internal plugins?

leamas, 2020-08-31:
But Jon cannot pull for example radar plugin, it's not maintained by him. Same for oesenc and perhaps shipdriver. The basic philosophy here is that plugin developers are independent.

So, again, if Jon uploads "his" plugins to couldsmith or somewhere else this is nothing to discuss.   Sticking to the radar_pi example: In what way would it be a problem if radar_pi didn't upload their metadata?

rgleason, 2020-08-31:
You are the one who initiated the tarball and metadata system. Why screw that it up unless you have some positive development that we can all use?   Later at the bitter end, bdbcat incorporated the metadata back into the tarball in order to make "Import Plugin" work. That has been a very good thing... Import plugin uses the tarballs with the metadata inside.  Opencpn/plugins/metadata uses just the metadata.

What exactly are you talking about doing?  

Alec, are you planning on rewriting everything now that you know where we need to go? IE this whole process has been a test to see what is needed, and now you're going to redo it?

leamas, 2020-08-31:
I have no plans whatsoever. I'm raising a simple question... and it seems that I get no answer. So: what would be the problem if radar_pi (the example) stopped uploading the metadata?

leamas, 2020-08-31:
OK, I have read up on Jon's notes in  https://github.com/opencpn/plugins, and I understand that the process is quite different than I anticipated. It has drawbacks (how handle beta channel?), but has the major advantage that is is simple. 

Answering my own question the problem would then be that if radar_pi stopped uploading metadata they would need to make their own PRs. This would cause them more work, but also more control.  It is not clear cut.

bdbcat, 2020-08-31:
Alec...
I am lost.
What in the world are you discussing?
Of course metadata files are required, in order to compose the catalog.
What is this "radar_pi uploading metadata" all about?

leamas, 2020-09-01:
Of course they are required. The issue is about the developer  -> metadata -> catalog process

What I had in mind when designing this was that plugin developers should make a PR when they wanted the plugin to be in included in master/beta/etc. In this scenario the uploaded metadata fills no purpose, it flows from the developer to the catalog directly through git.

Now, the metadata flows from the  developer to cloudsmith to the the catalog. This is sort of elegant, but not that well defined. Jon made some good notes in the OpenCPN/plugins README.md, we might need to document  the details in the process. Doing so we should IMHO take height for at least the beta catalog which will be needed in next cycle.

The basic question to document: How do I as a developer get my plugin metadata into the master/beta catalog?

The radar_pi issue is the example of a plugin which is not handled by the frontend2 community. I sidetracked here until I found Jon's notes. We can leave this aside for now.

rgleason, 2020-09-01:
"Of course they are required. "   That is exactly what I told you.

leamas, 2020-09-01:
> "Of course they are required. " That is exactly what I told you. 

Question is if they are required on cloudsmith. 

Now, if I understand the process for creating a PR it is basically

  - Fork OpenCPN/plugins
  - Download correct  metadata XML  files from cloudsmith.
  - Update the metadata/ dir  with said files.
  - Commit, push and make a PR.

Now, this is a working process anyone can use, more or less automated. There is just one point (and now I'm picky): from the overall opencpn project the basic message is that developers can make their plugins available in the catalog by issuing a PR to the plugins project.  

And that's all. How the PR is created is not defined.

The frontend2 approach is sound. There could be other ways, the basic problem is gathering the output from the builders into a PR. For a dev which makes a windows-only plugin (horrors!) she could just copy the xml file from her own build straight to the metadata dir, thus short-cutting the whole cycle.  

So, if we could just agree on the difference between the spec ("plugins are updated in the catalog using PR againt the plugins project") and the implementation (for example the frontend2 process) I'm happy.  This would basically mean that the uploaded XML files are optional as long as the PRs are sound, devs are free to invent something better.

And no, I have no secret idea of what a better way would be. I just want don't want the implementation to be the spec.

leamas, 2020-09-01:
When I think about it (honestly, havn't until now): Another way to "upload" would be to automatically push the metadata to author's clone of the github project.  Perhaps even easier for devs, especially if you are used to git and masters things like squashing the history etc.

No, I don't plan to spend time on anything like that. It's just an example on the importance of keeping one's mind open, and realize that such a different approach also would work; formally because this would also follow the spec.

rgleason, 2020-09-01:
Alec, I think you've got a good point here, perhaps it could be automated or done with some kind of script trigger:
1.  Remove your old working branch from local and your remote (assuming your previous PR has been accepted)
2.  Recreate the old working branch locally using upstream/master.
3. Add the appropriate metadata files to that branch directly without using Cloudsmith and make a Commit.
4. Push the branch up to your remote repository.
5. Initiate a Pull request to (upstream) opencpn/master 

I am not quite sure how this would really work in practice, but I can see it would be more efficient. Maybe the metadata files would first land in the actual plugin's  "Release" folder, then be copied to the developer's forked plugin branch with a script..

leamas, 2020-09-01:
My point is not that. My point is that there is a difference between specs and implementation. I think this discussion had made it clear that uploading the metadata isn't required by any spec, the only requirement is the PR against OpenCPN/plugins.

The frontend2 implementation uses the uploaded metadata and thus requires it. But if any plugin dev comes up with some scheme which creates a PR without any metadata uploaded, that's just fine. Which, in the end, boils down to that the cloudsmith metadata is optional.

bdbcat, 2020-09-01:
All...
I am philosophically in tune with leamas' comments.  This harks back to my original comments about CI as we first started that process.  I prefer NOT to specify tools and workflow, as far as is possible.  I am more focused on results, which in this case is a PR against plugins repo, containing one (or more)  plugin metadata file additions/updates.  Said additions/updates must always contain  validated URL for downloading the tarball.  Nowhere do (or should) we specify Cloudsmith.

rgleason, 2020-09-01:
"... metadata isn't required by any spec, the only requirement is the PR against OpenCPN/plugins"

The metadata is required by the way Opencpn Plugin Manager,  master catalog, github.com/opencpn/plugins/metadata directory and associated scripts work.  Any PR is very likely to have changes in the metadata, and not use tarballs.

leamas, 2020-09-02:
@bdbcat 

hm... what comes to my mind is that we should arrange OpenCPN/plugins so that any PR is validated (url, xsd) when submitted, sort of similar to how regular PRs against opencpn/opencpn are rebuilt using the CI builders.  Probably lot's of reading but perhaps not much code.

@rgleason 
What I said was that uploadiing the metadata to cloudsmith isn't required. However, making a PR is all about metadata, so we need it somehow. 

But not necessarily on cloudsmith.

rgleason, 2020-09-02:
"...so that any PR is validated (url, xsd) when submitted "
What I said, Alex was that Jon has already written a script in Github Actions that validates the PR before acceptance.
Perhaps you should really make a couple of PR to plugins so you see how it actually works, then look at what Jon has done.  Due to some of your recent changes, adding "arch" etc the script might need to be updated.

leamas, 2020-09-02:
My name is Alec.

And those changes was not originated by me, it was Dave's doing. Talk to him.

Good to know that the PR are checked, couldn't find that in the docs. Will add (preparing a PR).

rgleason, 2020-09-02:
Sorry, Alec,  a temporary lapse

rgleason, 2020-09-02:
Some links to the Internet Cafe Idea
There are some examples that have turned into 401

- https://github.com/OpenCPN/OpenCPN/issues/1839#issuecomment-613666607
- [Functionally complete version (no longer works)](https://github.com/OpenCPN/OpenCPN/issues/1839#issuecomment-618011405)
- [Mauro's example OpenCPN Plugin Downloader](https://squidd.io/api_docs)
- [Leamas ocpn-download](https://github.com/leamas/ocpn-download)

leamas, 2020-09-03:
I'm closing this. Tried to wrap-up the discussion in the revised docs in https://github.com/OpenCPN/plugins/pull/237

closed 2020-09-03 by leamas

rgleason, 2020-09-03:
Is the interned cafe download idea now defunct?
It seems to me that that what you prototyped would be a useful page for the the Opencpn.org Downloads page and another way to access and download the plugins.

leamas, 2020-09-03:
The idea is not defunct, but is stuck on #1865 i. e., the question how to deploy the service. It is all about Dave, and his is sort of busy...

rgleason, 2020-09-03:
Yes, indeed. You both are.

  Thanks.