OpenCPN/OpenCPN #2033: Revisited: Mint, debian plugin compatibility (sorry, long post)

leamas, 2020-09-03:
In 5.2 we have tried to let main opencpn figure out if some os'es like Mint and Debian are compatible with our mainline Ubuntu plugins. The Mint case broke as soon as 20.04 was out, and Debian carries a risk since sid (which we have said is compatible with focal) by definition is unstable and can break any time as it moves forward.

Even if we are able to release point releases of main opencpn to handle such breakage, this is a slow process both on our side and also when it comes to how user updates their installations. Basically, I think both some thought and our experiences shows that trying to let opencpn figure out if for example Mint is compatible is a bad idea.

Another sign of this is that the actual code in isCompatible() have become a mess, which most often is a sure sign of something wrong.

---

However, what if we move this compatibility stuff to the plugins?  That is, adds some optional part of the metadata which for example states that this very plugin is built for Ubuntu 18.04 but is compatible also with Debian 10 and Mint 19?

The main advantage would be the much quicker release cycle when it comes to plugins. It would make it possible to actually react within reasonable time when for example  Mint makes a new release. Also, the plugin code would be plain and simple, basically just a table to include in the metadata. Well, actually a table of tables, since each version needs their own compatibility table.

Still, it would just be static, well-defined data, perhaps a little large but nothing complicated, ending up in something like:

```
...
<compatible>
    <target> <os> debian </os> <version> 10 </version> </target>
    <target> <os> mint </os>  <version> 19 </version> </target>
</compatible>
...
```

Of course, the downside would be that this must be handled in each and every plugin. But, is there reasonable alternative? Yes, I can think of one or two. But I still consider letting the plugins declare their own compatibility to be the most consistent way.

I still think flatpak is the future on Linux. But for now we need to take care of our Debian legacy.

@jongough;  @bdbcat: thoughts?

rgleason, 2020-09-03:
"..much quicker release cycle when it comes to plugins." 

To be sarcastic about this, ...let the lowly plugin developers spin their wheels with this, changing their plugins all the time."

leamas, 2020-09-03:
No dev is lowly here, no way. It's just about that plugin devs can push a new version quickly, and have a chance that users actually update their plugins (at least those interested in things like mint).

> Why not get this straightened out with separate values for arch etc. It will be a mess for awhile with all the plugins but eventually it would get sorted out.

In what way would this affect anything on  w r t the subject in this issue?

rgleason, 2020-09-03:
"plugin devs can push a new version quickly, and have a chance that users actually update their plugins (at least those interested in things like mint)."

Yes, I can understand this point, but it does take significant time and effort to cycle through all of Sean's plugins, even assuming the changes are very routine.  Now if you were to narrow plugins down to the primary ones, weatherfax, weather_routing, climatology, polar, celestial_nav, watchdog, plots, statusbar  and other's plugins tactics, dashboardtactics,  logbookkonni, findit, aisradar and sometimes vdr, (not including all of them).   That is a significant amount of time.  Simply quick changing them somehow is NOT an option, it will take time.

leamas, 2020-09-03:
Sure, I have full respect for what it would take to go through all plugins. Nothing here is about running ahead quickly.

It's about a strategy. We need to figure out how to handle the compatibility issue within the Debian plugin universe. These are long-time issues, definitely nothing to do in haste.

What I refer to when talking about quick releases is the situation when everything is set up, and it would just be to add a line in the compatibility table. Keeping focus on the important plugins, this might perhaps be done with a reasonable effort (?)

Currently, if we don't can think of something else, I guess the alternative would be to leave it as it is, with some hardcoded support for upstream Debian releases, and nothing really for Mint et. al.  besides the *Advanced usecase*  manually editing the  configuration file.

The #2004 PR is related to this. making the current MInt usecase somewhat less advanced, so to speak. But it is not a final solution, more like a test/investigation tool.

rgleason, 2020-09-03:
Could this be in a single separate, standard file that is stored in  a virtual github directory somewhere (similar to what climatology does with its data)  that you and Dave change and have access too?

Then there would be some standard code in a separate standard file that would be put in every plugin, that would reach out and update from that single file at the appropriate time.  No plugin dev intervention needed, just a rebuild of the plugin.

leamas, 2020-09-03:
These are details I have not really looked into yet.  Let's see if @jongough  and/or @bdbcat chimes in before we jump into this pool.

leamas, 2020-09-03:
>The metadata file could contain a list of all required libraries and the minimum versions needed. This could then be used at runtime by plugin manager to see if the plugin is compatible. This would be similar, I think, to the way that deb and rpm files do it. This approach seems to work ok so why don't we try it.

It would indeed be similar, and take the pluginmanager to a point where it is a  complete package manager. It's nothing I think will happen. Havn't  we discussed this before?

If we should find a way to handle the compatibility it must IMHO be simpler, on the level this plugin can run on [Ubuntu-18.04, Debian-10, Mint-19]. Anything beyond that is  probably not feasible, so we need a solution along these line (if any).

This list will typically change independently of the opencpn release cycle, we need to handle that.

We also need to figure out if this is a common list for all plugins, or specific per plugin. I have so far leaned to the last alternative, but I'm certainly not sure.

jongough, 2020-09-03:
There will be common set, i.e. OCPN set, then there 'may' be a few additional items needed by specific plugins.

I have mentioned the list before as now that OCPN has a full package manager it needs to be a complete implementation, not a partial one. If it is too difficult to handle more than a couple of platforms then we need to change plugin manager to handle platform packages for the 'too difficult' platforms. This can be done via single change to the metadata and another module to handle the interaction with the platform packages manager. Yes, the user may be asked for authentication to do the install/remove, but is that really an issue? The plugins will still appear in the plugin manager, they will be installed/removed through the plugin manager, or they could be installed/removed as they used to be. By doing this all, or most, of the edge platforms can be handled.

bdbcat, 2020-09-04:
Following/thinking...
I'll be back.

Hakansv, 2020-09-04:
@leamas @bdbcat 
I've a user here running a AMD with Linux Mint 19.2 reporting a "not compatible" message and a O-crash when trying to install the managed oeSENC 4.2.15 from "scratch", i.e the "Parcel" so not installed before.
Could that be connected to this issue? If so I'll give him an answer to be patient and enjoy the rest of OpenCPN.
Håkan

leamas, 2020-09-04:
Basically, yes. But you could also tell him to try the flatpak version. There is a initial download which is huge, but after that it should be smooth sailing (well, the plugin downloads are also huge, #2029, but it should work anyway unless she uses too many plugins)

Hakansv, 2020-09-04:
OK, thanks.
He was not that eager for the moment so based on the user's and my own Linux experiences I think we'll wait until a normal plugin manager method is ready also for that distribution.
Håkan

leamas, 2020-09-04:
Well, of course. That said, I wouldn't say flatpak isn't "normal". It's actually endorsed by the Mint community, and the basic flatpak  nuts and bolts are available from Mint 18.3 by default. So it is just to visit https://flathub.org/apps/details/org.opencpn.OpenCPN, push the install button and, depending on bandwidth, perhaps take a cup of coffee.

Actually, it should be possible to install the opencpn flatpak package using the regular package manager GUI on Mint as well. Nothing I have tried, for sure. 

Bottom line: On a Mint box, flatpak is certainly normal.
 
BTW: the flatpak opencpn installation is completely different from the .deb-packaged one. This means that whatever it does, it will not affect the existing  one. It's a parallel universe.

bcn2, 2020-09-04:
To add an issue with the flatpak version right now: oeXXX is broken as some of the interactions require to leave the sandbox.

leamas, 2020-09-04:
@bcn2: hm... I sent a patch to Dave yesterday which I think is related to this. Thanks for input!

bdbcat, 2020-09-04:
bcn...
I have a tentative solution to oeXXX on flatpak, not well tested yet.  Coming soon.

jongough, 2020-09-04:
OCPN already knows what platform it is 'running' on because of the package used to install it and hence the build environment needed to create the installation package. This information could be used by making this information available at runtime time. Plugins make use of this process for version numbers etc..

Can we create an 'environment.h' file which contains information such as platform (Ubuntu, windows, Mac, android, etc), Bitness (64 or 32 possibly 128), version of platform, wxWidgers version, etc. needed to check plugins against. This means there would be no need to do convoluted checks, just check the 'internal' OCPN information against that supplied by the plugin.

Mint is based on Ubuntu and the plugin is built for Ubuntu and OCPN is built for Ubuntu and installed successfully on Mint, therefore the plugin is compatible.

Assuming the plugin doesn't have special package requirements all should be well. If it does and these cannot be incorporated into the plugin they could be specified in the metadata. As OCPN knows what platform is is running on it can use platform specific checking mechanisms to see if the plugin is compatible.

This may simplify the process.

leamas, 2020-09-05:
Well, of course possible. But I wonder if it's needed. Downstream on MInt one could just run `ldd pluginfile.so` which As for the environment..h: isn't  this is really the config.h file which already exists? Adding  items to this is simple, can you make a patch?

Or, do I get it wrong?

jongough, 2020-09-05:
Any header file could be used as long as it is build  time modifiable. There appears to be difficulty handling downstream releases as plugin manager rejects the plugins as being incompatible even though they are compatible as the deb file generated from the same platform build installs and works. as plugin manager seems to be having issues working out what is and isn't compatible I thought an alternative way of working it out would assist. It would appear that the compatibility is is becoming 'complex' and 'difficult to maintain', hence the push to use flatpak for everything that is not debian or native Ubuntu.

I cannot make a patch as I am not near my development machine and won't be for quite a while. I am currently using only a phone which is not suitable. I was hoping someone working on this issue could do this.

leamas, 2020-09-05:
> There appears to be difficulty handling downstream releases as plugin manager rejects the plugins as being incompatible even though they are compatible as the deb file generated from the same platform build installs and works. as plugin manager seems to be having issues working out what is and isn't compatible I thought an alternative way of working it out would assist. It would appear that the compatibility is is becoming 'complex' and 'difficult to maintina 

Indeed. However, the flatpak approach has more objectives, including supporting all linuxes not using .deb packages. We will need that anyway, and since it exists and works just fine on for example Mint it sets a limit for the amount of resources to spend in this issue.

And yes,  one limitation is that expanding the metadata to describe all dependencies like dpkg or rpm just won't fly. It's too much work.  

What we possibly *could* do is to maintain a table of which versions of a plugin that works on a given version of Ubuntu.

As I see it, here are a number  of practically usable alternatives:

  - Downstreams like Mint could do what downstreams do: rebuild the upstream package and apply a patch or two. Writing a patch for Mint should be straight-forward. In this context the compatibility question is so much simpler.  I still consider this the most cost-effective solution.

- Downstreams could check if plugins are loadable using ldd. This could be automated, downloading all plugins from a list of locations, apply the ldd test and produce a either a common table or snippets for each plugin metadata to be fed to plugin devs.
- Opencpn could try to load any plugin thrown upon it, handling errors by a user message and blacklisting the failing plugin. This blacklist needs to be available in the GUI somehow.

Given that this is about Mint users one could expect that the solution was provided by the Mint community, be it as downstream work or patches.  Personally, I'm  not motivated to do this work beyond some generic hooks in main opencpn which for example makes it possible to handle expanded compatiblity metadata.

The more I think about this one possible approach could be to leave things as they are, basically saying that Mint 20+ is officially supported using flatpak. The situation would change if/when there is a Mint rebuild of main opencpn

jongough, 2020-09-05:
This is NOT a Mint issue. This IS a plugin manager issue that has been highlighted by Mint. The plugin works fine if the executable is put in the correct place. Plugin manager just decided it is not compatible. It is the checking of compatability that is where the problem is. I have provided multiple possible solutions which are being ignored for whatever reason.

Flatpak, appveyor or other sandbox environments are not the answer, they are just head in the sand thinking. OCPN works fine on many linuxes, the plugins also work fine when installed via platform distribution installation packages. Using platform installation packages there is no need to do the work you are suggesting. 

If a plugin has been built for Ubuntu and packaged in two ways during the build and one installs and works and the other is rejected as incompatible then there is an issue with the compatability check. This needs to be fixed. This is about helping the user.

leamas, 2020-09-05:
Indeed. But the way to do to is in Mint. When we release Opencpn we have no way to know what the upcoming Mint version will do. This is the basic reason why Mint 20 broke. 

Also: Mint is definitely not the only Debain/ubuntu derivative...

Since there seems to be no common understanding how to resolve this in a reasonable way I personally lean to spend my work on other things. 

I'm mystified why a downstream Mint OpenCPN release isn't  seen as a viable alternative.

jongough, 2020-09-05:
Mint is based on the LTS desktop Ubuntu. There is nothing special about it. All Ubuntu LTS packages install on it, many legacy packages also install on it. This is purely a plugin manager issue. Plugin manager has a very simplistic approach to compatability. Plugin manager needs fixing. You are the main protagonist of plugin manager, so you need to fix it. Or is it now in the 'too hard' basket?

Stop blaming platforms. Which bit of 'the deb file installs OK, but the plugin manager tarball does not due to poor compatability checking' don't you get? The plugin works fine, is fully compatible and does not have any extras to install or customisations to make. So what possible 'customisation' could be made to make a compatible plugin more compatible?

You personally are washing you hands of the issues in the system you created. So you want others to fix up your issues because you cannot be bothered. Great attitude!

I have spent many days chasing your undocumented changes to the packaging process trying to make it usable. I have far better things to do than try to keep OCPN and its plugins usable as you change things on the fly. But I consider the users and developers and continue to put it large amounts of effort and testing.

Your decision to walk away is probably a good thing as others can fix up the issues you have created without you invalidating the fixes they provide.

bdbcat, 2020-09-06:
Jon...
There is much I could say here, but will withhold most.
I will say that:
a.   You are the only experienced OPCN developer with direct Mint experience, and a Mint dev environment running.
b.   OCPN plugin compatibility testing is not that complex, fully understandable by reading the code.
c.   OCPN is between release cycles now, and will not be updated to the general field for some months.
Which leads me to:
d.  Patches against current master that fix the compatibility test for Mint 20, or whatever, working within the current managed tarball installation scheme, will be gladly reviewed for incorporation into the next OCPN release.

If this cannot be done, for whatever reason, our advice to Mint users stands:
1. Install unmanaged legacy plugins from debs, if available, or
2. Switch to Flatpak.

Finally, if you want to criticize any OCPN developer over final design or implementation issues, I am your target.  That's what being DevLead means, and I gladly take responsibility...

Good day
Dave

jongough, 2020-09-06:
Dave, so we take something that works, improve so that it does not. I have been trying to sort out the 'issue' with mint and have provided multiple, simple solutions. I did try to implement some through the normal pull requests but gave up after the pulls failed due to the base being changed numerous times with flattened change histories. So I focused on building the plugins to the changing metadata and validation of XML.

I cannot get to my development environment to create fixes, which is why I haven't done this. I am unlikely to be able to do it for weeks. 

I have bitten my tongue on numerous occasions and just 'sucked it up' and done reworks. What I don't understand is that when providing possible solutions to problems they are ignored and complex partial solutions implemented. There is no acknowledgement of the root cause, that there are multiple possible solutions, and that we should not be passing the buck, i.e. mint development should be changed to fix OCPN internal issues.

You are the owner/lead developer, not a problem, but some others quote standards, tell others to use them otherwise they are stupid, and write complicate and obscure code (well to me at least) spread around like s**t on a field. I wish sometimes that they would consider others.

Just to let you know, mint is one of the most popular Ubuntu derivatives and is based on Ubuntu LTS releases. The changes made are to provide multiple, standard, desktops (cinnamon, mate, etc.) to make the user experience more 'mainstream' (more like win xp than android and win 10) and bundle/improve some of the standard apps. This makes it Ubuntu with a different desktop not some convoluted marginal system for tech heads.

Ubuntu 20.04 is different from 18.04 and mint changed from 18.04 to 20.04 after the new LTS had settled down.

I am now backing off to use the boat and my openplotter installation of OCPN on a raspberry pi 3b. As you intimate there will be no real changes to OCPN for quite some while.

G'Day
Jon

bdbcat, 2020-09-06:
Jon...
Thanks for the feedback.  Couple of points:
1.  Iterative, dynamic development models are a natural consequence of collaborative projects.  All of us spend time rebasing git commits, fixing other people's code, and sometimes making plain old guesses as to the plan forward.  There is wide diversity among the committed OCPN developers, both in experience and outlook. I see that as a good thing. To me, this is the normal model when the focus is on creating code, and following up with specs later, if at all.  So we do the best we can, almost never tossing out a working solution, even if flawed.

2.  I read the Mint managed plugin problem as an OCPN bug, same as you.  We have many bugs. However, we must prioritize our efforts, always.  We establish priority intuitively, but may be expressed something like:

Priority = (Observed user impact) divided by (cost).   Higher priority means more/faster effort to fix.

So, a problem affecting all platforms, with a "one-liner" fix, gets a very high priority.  These types of bugs are not uncommon. Conversely, a bug in one linux distro (however popular), with unknown but possibly high cost in implementation/testing, gets a relatively low priority.  That is the world we live in....

Have fun sailing this spring.
Dave

leamas, 2020-09-06:
hm... let's take a step back and consider why ubuntu plugins are not accepted in MInt 20:

1. Ubuntu focal was released April, 2020,
2. During the 5.2.0 beta cycle we discover that ubuntu plugins are not accepted in MInt 19.
3. A somewhat dirty but working patch was implemented in 5.2.0
4. 5.2.0 is released, supporting ubuntu focal.

5. Mint 20 is released, breaking the fix for Mint 19.

And here we are. Whatever fix we do for opencpn there is an obvious risk that it will break in  Mint 21 in the same way since there is no documented, stable way to find out the ubuntu version a given Mint release i based on.

As Dave pointed out, we cannot spend vast resources on this. Part of the context is of course also that flatpak, a solution we need to maintain for other reasons also works on Mint. 

I still think that rebuilding the opencpn package for Mint is the proper solution. After a nights sleep I also realize that there no need to actually patch the opencpn code in this context, using a shell wrapper which sets up the environment should be enough.

Just to as an illustration I enclose a patch against the official packaging. It's really 6 lines of code, but  3 of these lines are wrapped in a patch making it 20 lines. If there is any real interest in the Mint community to fix this, I think this is the obvious way to go when it comes to the current situation.

As for next opencpn release I concur with Dave: patches should be welcome, really. Actually, this is what downstreams do: fix packages, and send patches to upstream so they don't need to bother next release. This is a basic workflow in the open-source world.

[0001-Mint-20-Add-shell-wrapper.patch.gz](https://github.com/OpenCPN/OpenCPN/files/5179207/0001-Mint-20-Add-shell-wrapper.patch.gz)

leamas, 2020-09-06:
>  I did try to implement some through the normal pull requests but gave up after the pulls failed due to the base being changed numerous times with flattened change histories I

Yes, isn't it a pain... I have been having PR:s waiting for many months and yes, the floor is moving, What one need to do is using `git rebase` to rebase the PR to the new HEAD; this is not affected by the history in any way. Of course there are conflicts, but if done regularly they are usually not that big and can be resolved.

But it's certainly work to do.

jongough, 2020-09-06:
Mint version with Ubuntu codename is in 
/etc/os-release
The fields you want are:
ID=linuxmint
ID_LIKE=ubuntu
VERSION_ID="19"
VERSION_CODENAME=tara
UBUNTU_CODENAME=bionic

Also in:
/etc/upstream-release/lsb-release
has the following fields:
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic

I would have thought that would have been sufficient.

leamas, 2020-09-06:
As I said: patches welcome. 

That said, IIRC it was this kind of info which was the underpinnings of the MInt-19 patch. Have you any docs available which states that this interface is available also in 21? 

BTW: I was not directly involved in this work, it was late stage quick fixes, probably by Dave.

bdbcat, 2020-09-06:
I'll spin up a Mint-20 virtual, and take a look.
Dave

bdbcat, 2020-09-10:
OK. I did take a look.
The problem can, of course, be fixed by a specific Mint patch.  But this would be really just too ugly to commit, considering that the workarounds (environment, or config file override) are available.
Essentially, the problem is this:
a.  Mint 20 for X86-64  claims that it is "like" Ubuntu 20.  Fair enough, we detect and try to handle that case.  We would agree to load plugins that target "ubuntu", with ARCH=x86-64, and VERSION=20.
b. However, the target for OCPN ubuntu-20 _plugins_ is "ubuntu-gtk3".  There are NO OCPN managed plugins that target "ubuntu" for version 20.  None.

That "gtk3" thing is catching us....

So, we need to think more broadly here, without resorting to building a "package manager".
No panic, though. Let's get it right for the next release, after due deliberation.
Comments welcome

leamas, 2020-09-10:
Hm... many loose ends here:

Since we have no short-term solution in sight for Mint 21: are the oe\*\*\* flatpak plugins fixed? So that we can recommend flatpak instead?

Problem is still easily solved in a downstream Mint package rebuild, creating Mint-specific opencpn packages. Re-iterating my point that this is the correct solution, corresponding to a general, well-known open source workflow.
 
Any other solution must be able to cope with the fact that new Mint releases will happen after opencpn is released Same applies for other Debian derivatives out there. Which means that  we probably cannot hard-code the compatibility.

One possible route could be to use a compatibility table which was loaded using a plugin. We then need to introduce arch-independent data-only plugins, but that would be easy and a generally useful feature. Using this scheme we could update the table without a complete opencpn update. Many details here, none of which I have really considered...

Another could be to publish an alternative catalog for Mint, with the target os/version tweaked

Had a similar issue with Debian sid  (#2043). However, Debian is simpler, doesn't move that fast and is upstream for Ubuntu.

The question if a given plugin actually loads successfully can be tested using ldd. We could basically construct a test suite that reads a catalog, downloads all plugins and tests them for "loadability" including gtk2/gtk3 aspects without too much effort. However, such a test must be done on the target platform like Mint.
 
We should not have named the  focal ABI ubuntu-gtk3. The -gtk3 suffix was only needed on bionic where we had both ubuntu and ubuntu-gtk3. We need to get rid of that suffix, and next chance will perhaps be when wxWidgets 3.2 is out and we should have ubuntu-wx32, not ubuntu-gtk3-wx32.

bcn2, 2020-09-10:
oeXXX for flatpak is resolved (and tested ok by the user)

leamas, 2020-09-10:
Great. we actually have something working for Mint users.  Thanks for input!

leamas, 2020-09-10:
BTW: the debian packaging is a good illustration of the workflow: The Debian packager discovers a problem, adds a patch is done with it. The patch is targeted, when packaging for Debian both os and version is known, no need to be general.

In this process, the patch is sent to upstream (this project) so that it comes for free next time there should be a Debian release. That is the #2043 PR. Something similar should work just fine when rebuilding Mint packages.

rgleason, 2020-09-10:
"We should not have named the focal ABI ubuntu-gtk3."  I remember this one and Alec you wanted to name it ubuntu. thought it should be ubuntu-gtk3.   Is there a way to rename a runtime if it is mint?

bdbcat, 2020-09-10:
Rick...
We can fix Mint-20 with a very specific patch, no problem.  The risk is that Mint-21 will fail, in ways we cannot predict.
We are looking at longer term architectural changes here, for OCPN 5.4++
Meanwhile, there is a valid workaround for mint-20, if one insists on running mint native, and not flatpak.

rgleason, 2020-09-11:
I see. Thanks.

leamas, 2020-09-11:
I think I have a better idea, my best so far (besides a downstream Mint package; while unpopular, I still think it's preferable)

Backround:  We cannot implement a complete package manager, and we cannot see into the future. Using Mint as an example, this means we must actually do something when next Mint version is upon us. Using Mint 21 and Jon as examples we could work like this:

1. When 21 is out, Jon starts openpcn and uses the GUI in #2004 to figure out the correct plugin os/version to use on MInt 21. Let's assume it's ubuntu-wx32[ 21.04]. This is basically a click-try experience
2. Still on Mint 21, Jon runs a script which downloads and checks all ubuntu-wx32[21.04] plugins and verifies that they can be loaded. The output is a (short) list of plugins which does not work.
3. Jon now constructs an XML snippet like

        <compat>
            <host-os> Mint </host-os> <host-version> 21 <host-version>
            <plugin-os> ubuntu-wx32 </plugin-os> <plugin-version> 21.04 </plugin-version>
            <blacklist> radar </blacklist>
            <blacklist> vdr </blacklist>
        </compat>

     The entry above basically says that on Mint 21 ubuntu-wx32 [21.04]  plugins should be used. However, radar and vdr does not work.
  5.  Jon then forks the regular plugins project and pastes the snippet above in to ocpn-plugins.xml. The schema here is revised and looks like
        
          <plugins>
          ...
          </plugins>
          <compat-list>
              [place for snippet above]
          </compat-list> 
   6. Jon makes a regular PR against the plugins project which Dave merges.

This approach minimizes the work to be done for Mint-21. No changes needs to be applied to the plugins. The final parts, to check the blacklist and construct the XML snippet (perhaps even merging it into the catalog) can be scripted. One big advantage is that we can reuse the catalog maintenance workflow.

Of course, following this route the catalog would be updated with known info when we release 5.4. This would include Debian and older Mint releases. The code in isCompatible would be completely generic, no exceptions for any os.

One possible problem is how opencpn 5.2.0- would react when the catalog schema  is revised like this.

rgleason, 2020-09-11:
Very interesting idea. It could allow adaptation at runtime as Jon had suggested earlier, plus it captures the necessary changes.

leamas, 2020-09-15:
We now have exchanged more than 100 posts on this (I'm not innocent, I know). Maybe time to try to reach some kind of conclusion? My proposal:

  - Either go for a downstream package which basically sets up OPENCPN_COMPAT_TARGET before starting opencpn.
  - Or proposal above: https://github.com/OpenCPN/OpenCPN/issues/2033#issuecomment-690864241
 
Of course, there are other possibilities. That said, I think we should go for some of these two options. I prefer the downstream package, but can live with any alternative.

I see that Rick seems to be fine with the last option.

@bdbcat @jongough: thoughts?

leamas, 2020-09-22:
Well... no conclusion.  Issue isn't solved, but seems to need a fresh start. Perhaps a new issue, or just some code.

Closing.

closed 2020-09-22 by leamas

rgleason, 2020-09-22:
Alec, Some are on vacation and others are working hard on Android, so there is not a focus on this right now I think.

leamas, 2020-09-22:
I have  no hard feelings , none. It's just that this discussion is dead, and is better restarted somewhere else. 

I care for my issues, don't  want more open than  I can handle. I know that  there  are other priorities out there.  The conclusion is obvious: close the bug.

OTOH, I envy those sailing these days. 

If anyone is interested (and has some  input) , this bug can be repoened.

jongough, 2020-09-24:
Convenient. Brush issues under the carpet. Put into 'too hard' basket. Disappointed!

leamas, 2020-09-24:
Whatever is done here, it's certainly fully open.  Nothing is brushed under the carpet.

I have outlined the status in https://github.com/OpenCPN/OpenCPN/issues/2033#issuecomment-692524671. Without meaningful input on this, I don't see how this issue can be resolved and thus closed it. My list of open issues is already too long to handle.

If you want this issue resolved I suggest that you either give some feedback on the status above (in which case I can re-open this bug) or  file a new bug on your own.