OpenCPN/OpenCPN #2040: New GFS weather data from Saildocs does not work in CPN

bc38, 2020-09-08:
Somehow the addition of new GFS weather data from SailDocs passed us by!

http://www.saildocs.com/gribmodels

Currently available parameters are:
PRMSL (PRessure / Mean Sea Level, smoothed - SEE NOTE BELOW)
MSLP (unfiltered Mean Sea Level pressure)
WIND (10 meters above surface)
GUST (at 10 meters)
AIRTMP (temperature 2 meters above surface)
SFCTMP (temp at surface)
RH (Relative Humidity 2m above surface)
LFTX (LiFTed indeX)
CAPE (Clear Air Potential Energy)
RAIN (Precip rate, mm / hr)
APCP (Accumulated precip)
HGT500 (500mb height)
TMP500 (temperature at 500mb level)
WIND500 (Wind velocity at 500mb level)
ABSV (Absolute vorticity at 500mb)
CLOUDS (Total cloud cover)

Almost all new data is not displayed in CPN!

rgleason, 2020-09-08:
Didier had been keeping grib_pi updated but I have not seen any PR's from him lately. This will likely affect weather_routing to if the grib_pi version number changes, and weather_routing may have to be updated too.

rgleason, 2021-04-28:
BTW, Grib downloads from Saildoc work because these new data fields are not requested.
The failure is when an up to date Grib viewer is used to download a grib file with the newer data types.
This should be updated to prevent repeat user issues when downloads are from different viewers.

bc38, 2021-04-28:
Who would add new types of weather data to CPN !?

rgleason, 2021-04-28:
Not sure yet. but the question needs to be asked, how would this additional data really change what weather_routing does?
The only thing that I see that might be used for something is Gust and Cape (maybe for sail changes and a warning about bad changeable weather).

did-g, 2021-05-06:
Hi,

Which ones are missing ?

rgleason, 2021-05-07:
Hi did-g.  I could dig into it if you wanted, but bc38 Sept 8 2020 provided a list and a link that might be what you need? 

https://github.com/OpenCPN/OpenCPN/issues/2040#issue-695840663

bc38, 2021-05-09:
> Hi,
> 
> Which ones are missing ?

Now for the GFS request from SailDocs there is only this
WIND
PRESS
APCP
TCDC
AIRTMP
HTSGW
WVPER
WVDIR
SEATMP
GUST
CAPE
WIND500
HGT500

Missing request and read in CPN
PRMSL (PRessure / Mean Sea Level)
SFCTMP (temp at surface)
RH (Relative Humidity 2m above surface)
LFTX (LiFTed indeX)
TMP500 (temperature at 500mb level)
ABSV (Absolute vorticity at 500mb)
CLOUDS (Total cloud cover)

did-g, 2021-05-09:
Hi,
Old parameters are still working.
PRMSL is PRESS
SFCTMP is SEATMP
CLOUD is TCDC (IIRC)

we can't request RH , LFTX, TMP500 and ABSV, we do use RH if available for dew point, I'm not sure we are doing anything with it though. Assuming we do  we could request RH if  AIRTMP is set.

Do other matter for a layman ?

I'm adding ICON source.

Regards
Didier

rgleason, 2021-05-09:
Didier,  I've added a small review of Parameters at the end of [Grib_pi documentation.](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:plugins:included_plugins:grib_weather#saildocs_gfs_grib_data)

Could you review it? Also where do these parameters that are supported by OpenCPN, belong in the Saildocs GFS list?

<code>
IND
HTSGW
WVPER
WVDIR
</code>

did-g, 2021-05-10:
IND no idea  we haven't an IND parameter, typo ?

HTSGW, WVPER, WVDIR: about wave , we are displaying them.

cf http://www.saildocs.com/gribmodels
ICON is a global model run by Deutscher Wetterdienst (DWD)
also available with Xygrib

rgleason, 2021-05-10:
Thank you Didier.
I guess that IND is a typo from bc38 and it means nothing.
Thank you for the links.
Currently Grib_pi  can select   
"Saildocs"  GFS, COAMPS, RTOFS, HRRR  are you going to add ICON as a selection?
"Zygrib"   GFS

---- from saildocs  WW3 / GFSwave:
HTSGW Significant Height of Combined Wind Waves and Swell [100]  *  (not supported?)
WVDIR Wind Waves Direction (towards) [109]
WVPER Wind Waves Mean Period [110]
* default parameter is "HTSGW" (height of significant waves, i.e. the average of the largest 1/3 of combined swell and wind-waves), "WAVES" means the same.

Didier do I understand correctly that HTSGW is not supported?
When I select  GFS  "waves"  the message sent is
```
send GFS:46N,34N,58W,52W|,|0,12,24..72|=
WIND,PRESS,HTSGW,WVPER,WVDIR

```
So it appears that grib_pi also supports HTSCW?

rgleason, 2021-05-10:
I've revised  https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:plugins:included_plugins:grib_weather#saildocs_gfs_grib_data

Accordingly.

rgleason, 2021-05-10:
Didier is updating grib_pi, and adding ICON but I believe he has confirmed that the grib parameters are as current as is possible addressing the title and this could be closed.

rgleason, 2021-05-17:
Bc38. This has been fully addressed. You should close it now.

closed 2021-05-18 by rgleason