OpenCPN/OpenCPN #2041: RFE: checkum/sign plugins

leamas, 2020-09-08:
We have discussed checksumming/signing the plugins for obvious reasons. The metadata already has provisions in place.  Since the issue came up in downstream packaging, I'm filing a bug to track this feature.

rgleason, 2020-09-08:
I am not ready to deal with this at all, another change with all kinds of problems.

leamas, 2020-09-08:
Nothing will happen in this area for a quite  some time unless something unexpected (what?) forces us in this direction. 

That said, creating a simple checksum would be simple, just a few lines of code in the plugins I'd guess. Signing is trickier, since we then need to handle keys, most likely gpg keys.. 

But then again, this is not for now. Just want to track the feature.

rgleason, 2020-09-08:
I am very much in favor of checksum checking, as it would be quite automatic and I don't think it would be too much trouble on the plugins side!

It is the authentication part that has always spelled trouble, for users and applications. If only there were some seamless way to do it!!

leamas, 2020-09-17:
Having sound ha, cryptographic hash values like shs2 is really the only requirement for the plugins. If anything needs to be signed it would be the catalog.  However, this is IMHO beyond at least our short-term objectives.

So, a simple hash (checksum) for the plugins should IMHO be enough to close this.

leamas, 2020-10-05:
Here is a design problem: How handle checksums for imported plugins?  The checksum cannot go into the tarball...

rgleason, 2020-10-06:
An authenticated and  secure list somewhere?
plugin   checksum

leamas, 2020-10-06:
But the very idea with the import plugin scenario is that user is disconnected and/or is a developer which makes a quick check.  In either case, requiring an url somewhere doesn't' really fit the bill.

rgleason, 2020-10-06:
So we have to save it manually? and put it in as part of the installation?
What about local builds and using  Import Plugin, I use this for testing all the time.
Dealing with checksum would be a huge PITA in this case.

rgleason, 2020-10-06:
So checksum is going to require manual entry for everyone?

leamas, 2020-10-06:
I think you describe the problem quite well. Now we need solutions.

rgleason, 2020-10-06:
If is is a local build and Import must there be checksum checking?
Is there any way to determine the difference?

bdbcat, 2020-10-08:
I'll go out on a limb and suggest that checksum validation and possibly signature validation should apply only to normal master-catalog-based plugin loading only.
Experienced devs and testers will understand the risk of importing executable code from potentially unverified sources.  Happens all the time in development and testing....

Thoughts?

leamas, 2020-10-08:
hm... yes. Expanding on this, there are three cases(?).

When it comes to master catalog, current code  refuses to load a plugin if the checksum is wrong which should be fine. However, it silently accepts plugins without checksum. From a user perspective, a warning is probably justified in this case (?)

For beta/alpha, the current code does the same. Now, since these exists primarily for devs, it would probably be better just to display warnings for missing or bad checksum, loading the plugin in any case (?)

For the plugin import we probably just can't mandate mandatory checksum testing since the checksum can't live in the tarball. The remaining question is then if we can *offer* users the possibility to check the thing they downloaded in a shabby bar before they use it.

Just my 5 öre...

rgleason, 2020-10-08:
`The remaining question is then if we can offer users the possibility to check the thing they downloaded in a shabby bar before they use it.`
I don't know quite how this would work, but it sounds convenient and I would use it even when importing locally created tarballs when testing.

leamas, 2020-10-09:
First: can we agree on to modify current code like:
  
  1. Display a warning when user loads a plugin without checksum from master catalog
  2. Load all plugins in beta/alpha catalog, but display warnings if checksum is bad or missing.

Or?

leamas, 2020-10-09:
The second issue, verifying an import plugin is after some thinking not that meaningful when just applying a checksum test. The checksum comes the same source as the tarball, and doesn't add much comfort.

What *would* help was if either the catalog or the individual plugins was signed. This would the mean that we need to store public keys for the project in the client, and also have some code to manage (add/remove/update) keys. Probably more than we want to do in this cycle (?), but we could perhaps start preparing by adding some code to manage keys without actually using them.

This is not rocket science, this is what anyone publishing software does. I think it should be a long-time goal to have all binary downloads signed and verified in the client. 

That is, small steps in this direction.

rgleason, 2020-10-09:
Alec, I can't really respond meaningfully here because I don't know quite how this would work.
Public Key stored in the 'Client'  (What is the client?) The "master" catalog or the individual plugin (metadata? tarball?)
Private Key stored somewhere privately
OpenCPN then manages this using the keys.

I wonder if this isn't something like I was talking about, but instead of checksums, it is authentication.
If I have understand correctly, I do think it is a reasonable and good approach.

closed 2021-03-05 by bdbcat