OpenCPN/OpenCPN #2042: RFC: New sort of Debian plugins.

leamas, 2020-09-08:
RFC actually mean Request For Comments. Let's use it that way ;)

The legacy plugins vs the new tarballs has been discussed in depth. For me, this discussion is closed.

However, what if we created another kind of packaged plugin which just contains a tarball and metadata which it installs somewhere under /usr/lib/opencpn? That is, the same tarball which we now download from cloudsmith or elsewhere.

The required changes are minimal: the new file:///usr/lib/opencpn url would work exactly as current plugin urls, perhaps with some feedback in the gui. Basically, we are extending the catalog with local stuff.

Pros:
  - We can use established ubuntu build chains for different arches, versions etc.
  - The deb package handles dependencies.
  - Raises fewer questions about downloading binary content from unknown sources

Cons:
  - Debian (+ derivatives) only.
  - One  more deliverable, we need the tarballs anyway.

This is yet another input from the downstream packaging discussions. Filing for further reference. Looking at the pros/cons, my gut feeling is that the cons wins.

Guts should not always be trusted, though.

rgleason, 2020-09-08:
This sounds simple to me, if it solves user problems.

leamas, 2020-09-14:
After some sleep: this does not make sense, it's a bad idea. Even if we want to benefi from the Debian build chain there must be better ways. And  we can actually bulid for the platforms we care for using the CI build chains we have.

OTOH, I tried. Closing.

closed 2020-09-14 by leamas