OpenCPN/OpenCPN #2047: Support PluginManager for Ubuntu2004/ARM64 (beta)

hreuver0183, 2020-09-27:
Is it possible to start with supplying a limited set of plugins in the beta catalog for Ubuntu 20.04/ARM64?
As far as I can tell ARM64 is supported and Ubuntu 20.04 is supported, only the combination is not.

When the beta catalog starts it is easier to channel al efforts in the ARM64 port.
As far as I know mayor plugins like oesenc should work without problem (since the github modules work fine).

Background is of course that the only stable 64-bit release for the RPI4 currently is Ubuntu 20.04.
(Alternatives of the RPI4 mostly have a working Ubuntu 18.04 port.)

rgleason, 2020-09-27:
Just for information.  This is a system.

Every time these kinds of changes are made, every plugin has to be rebuilt again to get whatever version you are asking for.  It is best to get these kinds of kinks worked out early and it is getting pretty late in the game I think, because the major effort of rebuilding is winding down now, and melding into normal maintenance, as far as I am concerned.  I might not be touching some plugins of Sean's for awhile (pretty soon).

Also it is virtually impossible to support every minor/major version of OS that users seem to want!

leamas, 2020-09-27:
Sorry, arm64 is not not really supported if I understand this right. When it comes to ubuntu, we support x86_64 a. k. a. amd64. And when it comes to arm the primary focus is raspberry and other smaller devices running on armhf.

The lowest hanging fruit would be to release an arm64 ubuntu build, and stop disabling the arm64 flatpak plugins.  That is not to say such a thing is under way.

BTW: what is the Ubuntu/arm64 usecase?

hreuver0183, 2020-09-29:
Arm64 on ubuntu 18.04 is supported via the pluginmanager and that also is usable on debian buster.
My question is also supporting arm64 on on ubuntu 20.04, which would also be usable on debian bullseye.

The way I see it the RPI4 is supported with armhf/32-bit. Yes, it works. But is severely limiting the performance of an already limited system. The usecase would be an OpenPlotter replacement on ARM64, that is not an option now but might be next year (early adapters) or the year after (general public).

Right now OpenCPN works on ubuntu 18.04  (ppa), even the oesenc-plugin works (18.04 => pluginmanager,). However there is no stable distribution for buster/bionic for the RPI4 and therefore no (stable) usecase. (There is a testversion of Buster for RPI4).

The other option is similarly bad. OpenCPN works on ubuntu 20.04 (ppa), the oesenc-plugin works only from github. There is a stable distribution for Focal for the RPI4 (and Bullseye should be released early 2021). But there is no usecase as there are no (legally) available charts since the only source of plugins is github.  With plugins Ubuntu 20.04 could be supported now and next year Bullseye as well.

Currently all software is available, but there is no usecase for ARM64 on RPI4 at all.
Alternatives for the RPI4 (Pine, Odroid) can be used with Ubuntu 18.04 Bionic/Debian Buster but very few people use them. And even there you see the push for Ubuntu 20.04 Focal.

Thus my question is if a policy change for supporting Ubuntu 20.04 / ARM64 is an option.

leamas, 2020-09-29:
Hi,

This is not really a question of policy as much as resources. I also see the change to arm64/aarch64 in the Rpis coming...

As usual questions like these are easily deflected using "patches welcome..." :) To be specific: You could do this yourself, and as long as you limit it to the presumably small number of plugins you are actually using it shouldn't be that much work. The  basic workflow for a plugin:

 - Clone the  sources from github (note: you probably want jon's or Rick's forks rather than the pristine source in many cases).
 - Set up Travis, CircleCI and Cloudsmith according to  https://github.com/leamas/shipdriver_pi/blob/master/CI.md
 - When you have everything running, add your own arm64 buillders on travis or circleci.
 - Create a new ocpn-plugins.xml as described in https://github.com/OpenCPN/plugins/blob/master/TESTING.md
 - Test it locally, either by copying to correct location or making it available on an url; the latter used as "custom url" in the plugin UI.
 - Make a PR against https://github.com/OpenCPN/plugins.

So: I cannot speak for @bdbcat, but as I see it: patches welcome!

rgleason, 2020-09-29:
If this is going to be an inevitable progression for RPI, wouldn't it be better to get it over with now?
But the big problem is resources, as you say.  Sorry for the bellyaching.

bdbcat, 2020-09-29:
Hard to disagree with most of the above.  Certainly we have no "policy" precluding any specific combination of platform and distro.
But the devil is in the resources.
So, as we say, "patches welcome"

hreuver0183, 2020-09-29:
I'll check if I can manage to build a usable package with cloudsmith.
it was on my list anyway.

leamas, 2020-09-29:
Great!  Feel free to ask if you have any problems.

hreuver0183, 2020-10-02:
Results up-to-now:
- I started with oesenc for ubuntu focal/arm64
- I did use travis-ci for arm (like bdbcat) as it seems circle-ci and appveyor won't build for arm.
- build successful after minor changes to .travis.yml and ci/travis-build-ubuntu-gtk3-arm64.sh

Changes can be found in my repository, but it is not tested yet.

Currently I need to push the package to Cloudsmith. I think the structure is ready.

Questions:
- do I need to push the package to <user>/<ocpn-plugin-unstable> or should it be <org>/<ocpn-plugin-unstable>?
(example oesenc_pi suggests <user>)

- where do I put the Cloudsmith keys for Travis-CI (readable in the GitHub sources can't be the solution)?

hreuver0183, 2020-10-02:
A second question, for bdbcat this time, might be off-topic but somehow related:
- Is the RaspiOS Buster package for ARM64 supported? 

As far as I can see now (haven't tested it personally) the PM does not give a list of installable plugins and the bionic plugins don't seem to work. I was interested because earlier there was some discussion over permitting (x86?) Ubuntu packages for use on Debian (Bionic => Buster and Focal => Bullseye).

Related because if the Focal package works, do I need to build an extra package for Bullseys/ARM64?

Did not want to start a new topic now. A yes/no answer would be sufficient here.

leamas, 2020-10-02:
> do I need to push the package to user/ocpn-plugin-unstable or should it be org/ocpn-plugin-unstable?
(example oesenc_pi suggests user)

Use  _user/ocpn-plugin-unstable_ . The basic idea is that each "vendor" is responsible for her urls. 

> where do I put the Cloudsmith keys for Travis-CI (readable in the GitHub sources can't be the solution)?

Certainly not. They should be made available as environment variables in Travis, configured in the travis web UI.  The details could be found in https://github.com/Rasbats/shipdriver_pi/blob/master/CI.md (follow link at bottom).

> Is the RaspiOS Buster package for ARM64 supported?

I know should shut up, but... If you look at ocpn-plugins.xml, as downloaded when you update the catalog or by cloning http://github.com/OpenCPN/plugins  you can do ` grep '<target>' ocpn-plugins.xml | sort | uniq` to get a  list of what we build (i. e., support) today. Here are no arm64 builds.

hreuver0183, 2020-10-02:
I haven't checked for android lately (not my preferred platform).
But nice!

And thanks for your off-topic answer.

rgleason, 2020-10-02:
This is not covered by the links cited above.   Where do I put the CLOUDSMITH_API_KEY?
To enable your building service accounts to deploy to Cloudsmith use your Cloudsmith api key.

`Circleci  - from the Project, pick Project Settings (upper right),  Advanced Settings (left menu) turn on github updates and build forked builld requests.  Then Environment Variables (left menu) and enter name: CLOUDSMITH_API_KEY and value: the number. `

`Travis  -from your Dashboard select the right menu in the plugin line, pick "Settings"  Under General consider turning on build pushed branchs and pull requests,  Under  Environment  put your Cloudsmith key name and number in the proper place pice add or save.
`
`Appveyor - from your Projects  find the popup "Edit Project Settings" at the right of the selected project line.  Pick Environment on the right.  Find the "Environment Variables" and enter your Cloudsmith key.`

To enable your building service accounts to collaborate with github.com you will need to set those up too, but travis and appveyor havve already been done.

leamas, 2020-10-02:
> THis is not covered...

It actually is. If you read the reference, it pointed to the links at the bottom. Here we find  https://docs.travis-ci.com/user/environment-variables/#defining-variables-in-repository-settings

rgleason, 2020-10-02:
I stand corrected

hreuver0183, 2020-10-02:
I could not get it working, so I tried installing the cloudsmith-cli tool on my laptop.

Just for future reference, you have to use the "org" nowadays (the one with the garbage in the name):

`user@host:~$ dd if=/dev/zero of=testbin.5mb count=10000
10000+0 records in
10000+0 records out
5120000 bytes (5,1 MB, 4,9 MiB) copied, 0,021092 s, 243 MB/s
user@host:~$ du -sm testbin.5mb 
5	testbin.5mb
user@host:~$ export CLOUDSMITH_API_KEY=<secret>
user@host:~$ cloudsmith push raw --republish --name testdata-5mb-zeros.bin ci-user-k7d2rpURrpwe/opencpn-plugins-unstable ./testbin.5mb 

Checking raw package upload parameters ... OK
Checking testbin.5mb file upload parameters ... OK
Requesting file upload for testbin.5mb ... OK
Uploading testbin.5mb:  [####################################]  100%
Creating a new raw package ... OK
Created: ci-user-k7d2rpURrpwe/opencpn-plugins-unstable/testbin5mb (B2bGvCC7sViU)

Synchronising testbin5mb:  [#####-------------------------------]   15%  In ProgSynchronising testbin5mb:  [#########---------------------------]   25%  00:00:0Synchronising testbin5mb:  [################--------------------]   45%  00:00:0Synchronising testbin5mb:  [################--------------------]   45%  00:00:0Synchronising testbin5mb:  [###################-----------------]   55%  00:00:0Synchronising testbin5mb:  [###################-----------------]   55%  00:00:0Synchronising testbin5mb:  [###########################---------]   75%  00:00:0Synchronising testbin5mb:  [####################################]  100%  Completed / Fully Synchronised                      

Package synchronised successfully in 6.514413 second(s)!
`

leamas, 2020-10-02:
Well, I certainly don't have to : https://app.circleci.com/pipelines/github/leamas/OpenCPN/1112/workflows/943a3795-129b-4df0-a5cb-1c8a87952f40/jobs/2107  (look under the upload tab).

Perhaps the have a new policy for those late to the party?

hreuver0183, 2020-10-03:
My guess is that is is only for new accounts.

hreuver0183, 2020-10-03:
First: I saw bdbcat added the focal/arm64 plugin to the master.
Thanks for that.

Update of OpenCPN this morning:
- tried installing my own plugin, it failed with "incompatible plugin".
- tried installing the plugin from bdbcat, same error
(The updated master catalog contains no installable plugins).

I applied PR #2052 to OpenCPN (own github build, change 1 line of code):
- tried again installing my own plugin, success (4.2.16.7)
- tried installing the plugin from bdbcat, it failed with "incompatible plugin"  (4.2.16.8)
Since the bdbcat plugin already contains some extra patches, my current patch will not be officially published.

If someone wants to test my obsoleted plugin:
https://cloudsmith.io/~h-reuver-k7d2rpURrpwe/repos/opencpn-plugins-unstable/packages/detail/raw/oesenc-ubuntu-gtk3-arm64-20.04-metadata/4.2.16.7.0.59f7daa/

I might look into the oesenc_pi 4.2.16.8 version. Bdbcat's plugin should work.

leamas, 2020-10-03:
hm...If you look into the logfile there is more info on *why* the plugin is rejected. Start opencpn with `--loglevel debug` , and I'm sure you will find out the exact reason.

hreuver0183, 2020-10-03:
> cat .opencpn/opencpn.log|grep arm64|grep -i oesenc
13:40:32.659   DEBUG PluginHandler.cpp:234 Plugin compatibility check1: oeSENC  OS:ubuntu-arm64  Plugin:android-arm64
13:40:32.659   DEBUG PluginHandler.cpp:291 Plugin compatibility checkFinal REJECTED: oeSENC  PluginOS:android-arm64  PluginVersion: 16
13:40:32.659   DEBUG PluginHandler.cpp:234 Plugin compatibility check1: oeSENC  OS:ubuntu-arm64  Plugin:ubuntu-arm64
13:40:32.659   DEBUG PluginHandler.cpp:291 Plugin compatibility checkFinal REJECTED: oeSENC  PluginOS:ubuntu-arm64  PluginVersion: 18.04
13:40:48.869   DEBUG PluginHandler.cpp:234 Plugin compatibility check1: oeSENC  OS:ubuntu-arm64  Plugin:android-arm64
13:40:48.869   DEBUG PluginHandler.cpp:291 Plugin compatibility checkFinal REJECTED: oeSENC  PluginOS:android-arm64  PluginVersion: 16
13:40:48.869   DEBUG PluginHandler.cpp:234 Plugin compatibility check1: oeSENC  OS:ubuntu-arm64  Plugin:ubuntu-arm64
13:40:48.869   DEBUG PluginHandler.cpp:291 Plugin compatibility checkFinal REJECTED: oeSENC  PluginOS:ubuntu-arm64  PluginVersion: 18.04

leamas, 2020-10-03:
The plugin version is actually a mess. Short story: it comes from the metadata file rather than the compiled data. Sparing you the long story for now...

hreuver0183, 2020-10-03:
@leamas 
I believe that if I'm able to copy the plugin from bdbcat and make a working version, the startpoint is the same and correct.
I can only guess that something went wrong in the docker image.

Changes by copying the travis-build-ubuntu-arm64.sh to travis-build-ubuntu-gtk3-arm64.sh:
 docker run --privileged -d -ti -e "container=docker" \
+      -e "DEBIAN_FRONTEND=noninteractive" \
       -v ~/source_top:/source_top \
And the ubuntu 18.04 image does not install the libwxgtk3.0-gtk3-dev package (I hardly believe the dialog package the dialog package I added will change anything in noninteractive mode.)

That and the CLOUDSMITH_API was all I changed

My plugin:
4:27:47.056   DEBUG PluginHandler.cpp:318 Plugin compatibility checkFinal ACCEPTED: oeSENC  PluginOS:ubuntu-gtk3-arm64  PluginVersion: 20.04
14:27:47.056   DEBUG PluginHandler.cpp:261 Plugin compatibility check1: oeSENC  OS:ubuntu-gtk3-arm64  Plugin:android-arm64
14:27:47.056   DEBUG PluginHandler.cpp:318 Plugin compatibility checkFinal REJECTED: oeSENC  PluginOS:android-arm64  PluginVersion: 16
14:27:47.056   DEBUG PluginHandler.cpp:261 Plugin compatibility check1: oeSENC  OS:ubuntu-gtk3-arm64  Plugin:ubuntu-arm64

But I'll post the complete logs from the original OpenCPN version later.

hreuver0183, 2020-10-03:
Commands:
> sudo apt-get install opencpn   # restoring OpenCPN to the standard distribution
> dpkg -l opencpn
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version          Architecture Description
+++-==============-================-============-==============================>
ii  opencpn        5.2.0.9-0~focal1 arm64        Open Source Chartplotter and M>
> /usr/bin/opencpn --loglevel debug

Command within OpenCPN: options - select plugins - update catalog (no update candidate for my 4.2.16.7 plugin).

[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/5321664/opencpn.log)

leamas, 2020-10-05:
Hi, sorry for delay... 

There is a major mess when it comes to the platform id. In bionic, we had ubuntu and ubuntu-gtk3 . It was necessary since bionic was a transition release supporting both  wx-gtk2 and wx-gtk3.

In a better world we should have dropped the -gtk3 suffix in focal since it does not support gtk2, making the gtk3 suffix useless. However, we didn't and are now stuck using ubuntu-gtk3 for x86_64 until next ABI break.

In the logs the rejection is clear: the host OS is  ubuntu-arm64, and the plugin is ubuntu-gtk3-arm64. This could basically be solved in two ways: by "fixing" the opencpn code to use ubuntu-gtk3-arm64, or changing the plugin to use ubuntu-arm64.

I think we should settle for ubuntu-arm64 for focal (20.04+). It's the Right Thing, and that we made the wrong decision for x86_64 shouldn't make us do the wrong thing also for arm64. This means updating the plugin side somehow. It also means that we don't have to change main opencpn.

When it comes to bionic it's trickier since there is two bionic platforms: ubuntu  and ubuntu-gtk3.  Since this work is aiming forward, we could perhaps just drop this for the moment?

leamas, 2021-04-26:
@hreuver0183: There is now #2178 under way.  The arm64 flatpak ubuntu build is available, we need to fix the build chain which should be possible now when drone, CircleCI and Travis offers arm64 builds. 

Could we close this bug and focus the efforts on #2178?

leamas, 2021-05-06:
I did it the other way around by closing #2178,  which leaves this bug with the overall issue. In short, I think the overall strategy for the arm64 support should be based on flatpak so we can avoid the need to support Debian, Ubuntu, Raspbian and what not on arm64/aarch64.

Current flatpak status: main application is available on the flatpak beta branch, installable usíng

    $ flatpak install --user https://flathub.org/beta-repo/appstream/org.opencpn.OpenCPN.flatpakref

The plugin situation is covered in #2178. Flatpak aarch64 support has been merged into the major plugin communities, but it will take some time before actual plugin builds starts to trickle down into the catalog.

transmitterdan, 2021-05-06:
For the rare users on some "offshoot" Linux variants I agree flatpak is the way forward.

But Raspbian is going to be very widely used IMO. I would be worried about the overhead of flatpak on Raspbian.  OpenCPN performance has decreased significantly on 32-bit Raspbian but I haven't been able to discover why.  It could be that the underlying OpenGL code has been hampered.  But flatpak has a lot of overhead and I worry that Pi users will resist it.  I think we should support Raspbian native packages owing to the large number of Pi users.  Eventually, 90+% of them will move to 64-bit, probably by force.

leamas, 2021-05-06:
>  But flatpak has a lot of overhead 

Performance-wise? Citation needed...

rgleason, 2021-05-06:
Alec isn't the issue more the size as you have mentioned before?  Is it 500mb?  but what is it without building, just installed?

leamas, 2021-05-06:
As I read @transmitterdan's post, he argues that there is a large performance penalty. This statement does indeed IMHO need some citation.

As for the size, the basic package is about 50MB, about the same as other packages. However, it needs a runtime which is about 400MB.

OTOH, the runtime is shared and is used also by other flatpak applications. So, if they already are using a flatpak'ed application, chance is that they already have the runtime in place. Likewise, when updating opencpn there is normally no need for a new runtime.

bdbcat, 2021-05-06:
tdan...
If you know more, please comment on the current state of official arm64 support for rPi.  It seems pretty fragmented.  AFAICT, even the name "Raspbian" has been deprecated.  So what will "normal" users be installing, in your view.

Hakansv, 2021-05-06:
One point: A "normal" Raspbian user may use the Openplotter update routine. We may involve Roberto?

bdbcat, 2021-05-07:
Of course.

transmitterdan, 2021-05-07:
Does open plotter support a 64-bit OS without using a "container"?

@bdbcat ; there is only a native 64-bit Raspbian beta and it isn't easy to get working and is not officially supported. Most 64-bit'ers are using containers such as docker or flatpak. The truly brave build 64-bit Ubuntu 20.04 and are thus self supported. Fragmented is a kind description of the 64-bit Pi landscape. Hell is a better analogy IMO.

There is often something that doesn't work right when using containers. For some it's WiFi, for others a USB comm port issue, etc. And the containers are huge plus keeping them all patched is a PITA. It's like owning a half dozen computers. 

I frankly do not understand the lust for a 64-bit OS on a tiny CPU like the Pi 4 arm chip. There is zero performance improvement AFAICT. In fact, I think it is actually a bit slower than 32-bit but that may be a maturity problem that will get better over time.

Is the Linux kernel going to drop 32-bit support in the near future? Otherwise, what is driving people to 64 bit OS on a tiny arm?

@leamas : my Pi4 with flatpak died shortly after I got it running. The thing won't boot at all so probably an infant mortality event. I'm ordering a couple more to play with.

When it was working, I ran O in flatpak but it wasn't working right. I admit that was a month or so ago and wasn't a fair test due to lack of time B4 the chip died. I'll try to restart performance testing soon.

But O on a "stock" Raspbian Pi4 with a huge 4K monitor is a beautiful thing to watch. People cannot believe the hi-def charts that get churned out of a deck of cards size computer using <15 watts of DC power. Looks much better than any paper chart system IMO.

bdbcat, 2021-05-07:
@transmitterdan 
If I may summarize what I hear:  Despite some quirks and annoyance with flatpak, this seems to be THE way forward for OCPN on 64bit rPi, at least in our limited resource model.  Otherwise, for plugins,  we will be chasing distros and versions indefinitely, for probably a minimal user base in real life.
Performance:  In my testing OCPN on linux x86_64 (flatpak), I find no perceptible loss in performance compared to a native build.  Some aspects seem even snappier under flatpak.  So, without more contrary data, let's try to put the "performance overhead" argument against flatpak to bed.

rgleason, 2021-05-07:
So are you going to stick with 32 bit RPI?

leamas, 2021-05-07:
Taking the liberty to talk for Dave, who said:

> If I may summarize what I hear: Despite some quirks and annoyance with flatpak, this seems to be THE way forward for OCPN on 64bit rPi, at least in our limited resource model.

How on earth can you interpret this as we should not support arm64/aarch64?

rgleason, 2021-05-07:
Thanks. I was reading bdbcat's observations about 64bit on RPI!

transmitterdan, 2021-05-08:
Are plug-ins supposed to work any place flatpak works?

I created a Pi4 system (Arm) running 64-bit Ubuntu Server-Hirsute. OpenCPN worked great in flatpak. But no third party plugins work. PIM will install them but PIM says they are incompatible.

leamas, 2021-05-08:
@rgleason: No. If it's reported as incompatible it's something else

transmitterdan, 2021-05-08:
Thanks. I'll dig in and find the root cause. Should not be hard.

transmitterdan, 2021-05-12:
@leamas , link at bottom of this post is a LOG file with "trace" mode enabled.  I cannot really tell why it is rejecting all 3rd party plugins.  Maybe you can give some hint and I can research more.

Platform: Rasberry Pi 4
OS details:
<pre><font color="#26A269"><b>ubuntu@ubuntu</b></font>:<font color="#12488B"><b>~</b></font>$ cat /etc/os-release 
NAME=&quot;Ubuntu&quot;
VERSION=&quot;21.04 (Hirsute Hippo)&quot;
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME=&quot;Ubuntu 21.04&quot;
VERSION_ID=&quot;21.04&quot;
HOME_URL=&quot;https://www.ubuntu.com/&quot;
SUPPORT_URL=&quot;https://help.ubuntu.com/&quot;
BUG_REPORT_URL=&quot;https://bugs.launchpad.net/ubuntu/&quot;
PRIVACY_POLICY_URL=&quot;https://www.ubuntu.com/legal/terms-and-policies/privacy-policy&quot;
VERSION_CODENAME=hirsute
UBUNTU_CODENAME=hirsute
</pre>

This is the command line used to run:
```
flatpak run --share=ipc org.opencpn.OpenCPN --loglevel=trace
```

Note that O will not run without the --share=ipc flag set for flatpak.  I suspect it should be configured in the package so flatpak knows this flag is needed.

[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/6466047/opencpn.log)

rgleason, 2021-05-12:
Are these related issues?   Perhaps not.
https://github.com/rgleason/LogbookKonni_pi/issues/25
https://github.com/rgleason/LogbookKonni_pi/issues/24

leamas, 2021-05-12:
it's odd...  I currently don't have anything to test on. However, when I look into the logs the first thing which strikes me is that a large number of plugins are accepted:

       $ grep ACCEPTED opencpn.log |awk '{print  $8}' | sort | uniq | wc
       33

The problem seems to be at another level. Snippet from the logfile:

    07:00:42.357 MESSAGE pluginmanager.cpp:1174 Checking plugin candidate: /home/ubuntu/.var/app/org.opencpn.OpenCPN/lib/opencpn/libvdr_pi.so
    07:00:42.357 MESSAGE pluginmanager.cpp:2124 Plugin is compatible by elf library scan: true
    07:00:42.357 MESSAGE pluginmanager.cpp:2232 PlugInManager: Loading PlugIn: /home/ubuntu/.var/app/org.opencpn.OpenCPN/lib/opencpn/libvdr_pi.so
    07:00:42.357   ERROR dlunix.cpp:162 /home/ubuntu/.var/app/org.opencpn.OpenCPN/lib/opencpn/libvdr_pi.so: cannot open shared object file: No such file or directory
    07:00:42.358 MESSAGE pluginmanager.cpp:2291    PlugInManager: Cannot load library: /home/ubuntu/.var/app/org.opencpn.OpenCPN/lib/opencpn/libvdr_pi.so

There are more snippets like these. So the first question becomes: Is this and other plugin libs available in 
`/home/ubuntu/.var/app/org.opencpn.OpenCPN/lib/opencpn/` ?

BTW: The username *ubuntu* is, well, somewhat odd (?)

leamas, 2021-05-12:
Hm... if the file, for example libvdr_pi.so exists at the expected location we need to check it's dynamic linkage. 

First: start a shell in the flatpak sandbox: ` flatpak run --command=bash --devel org.opencpn.OpenCPN`. This will start an interactive shell in the sandbox. From here, please do _ldd~/.var/app/org.opencpn.OpenCPN/lib/opencpn/libvdr_pi.so_ and post the resulits

transmitterdan, 2021-05-12:
Maybe this result points to a problem?

<pre><font color="#26A269"><b>ubuntu@ubuntu</b></font>:<font color="#12488B"><b>~</b></font>$ flatpak run --command=bash --devel org.opencpn.OpenCPN
<font color="#C01C28"><b>error: </b></font>runtime/org.freedesktop.Sdk/aarch64/18.08 not installed
<font color="#26A269"><b>ubuntu@ubuntu</b></font>:<font color="#12488B"><b>~</b></font>$ 
</pre>

transmitterdan, 2021-05-12:
@leamas 

Here are the contents of the plugins lib folder:

<pre><font color="#26A269"><b>ubuntu@ubuntu</b></font>:<font color="#12488B"><b>~</b></font>$ ls /home/ubuntu/.var/app/org.opencpn.OpenCPN/lib/opencpn/ -l
total 6124
-rwxr-xr-x 1 ubuntu ubuntu 3023616 Dec 31  1969 <font color="#26A269"><b>libaisradar_pi.so</b></font>
-rwxr-xr-x 1 ubuntu ubuntu 2910008 Dec 31  1969 <font color="#26A269"><b>libocpn_draw_pi.so</b></font>
-rwxr-xr-x 1 ubuntu ubuntu  331216 Dec 31  1969 <font color="#26A269"><b>libvdr_pi.so</b></font>
<font color="#26A269"><b>ubuntu@ubuntu</b></font>:<font color="#12488B"><b>~</b></font>$</pre>

And the result of ldd:

<pre><font color="#26A269"><b>ubuntu@ubuntu</b></font>:<font color="#12488B"><b>~</b></font>$ ldd ~/.var/app/org.opencpn.OpenCPN/lib/opencpn/libvdr_pi.so            
	not a dynamic executable
<font color="#26A269"><b>ubuntu@ubuntu</b></font>:<font color="#12488B"><b>~</b></font>$ 
</pre>

rgleason, 2021-05-13:
Please see  https://github.com/rgleason/LogbookKonni_pi/issues/25#issuecomment-840208642

leamas, 2021-05-13:
@transmitterdan : So, the issue is that while for example libvdr_pi.so exists, the log says that according to openpcn it does not exist.

What *really* bugs me is ` error: runtime/org.freedesktop.Sdk/aarch64/18.08 not installed` . What does  _flatpak list_  tell us on your box?

@rgleason: This is for now about tdan's problems. We need to resolve this before trying to understand if it's related to "your" issues.

transmitterdan, 2021-05-13:
<pre><font color="#26A269"><b>ubuntu@ubuntu</b></font>:<font color="#12488B"><b>~</b></font>$ flatpak list
<b>Name          Application ID               Version         Branch Installation</b>
Freedesktop … org.freedesktop.Platform     18.08.39        18.08  user
html5-codecs  …sktop.Platform.html5-codecs                 18.08  user
Greybird GTK… org.gtk.Gtk3theme.Greybird                   3.22   user
OpenCPN       org.opencpn.OpenCPN          5.2.4-0+6b314e6 stable user
<font color="#26A269"><b>ubuntu@ubuntu</b></font>:<font color="#12488B"><b>~</b></font>$ 
</pre>

Remember that OpenCPN also said that the shared object file passed the "elf" test.  So the file "existed" earlier in the process.

leamas, 2021-05-13:
You have an unsupported build from the stable branch. This is probably not the culprit, but as as starter please install and run the build from the beta branch, which is the only thing which is supposed to work:

     $ flatpak install --user https://flathub.org/beta-repo/appstream/org.opencpn.OpenCPN.flatpakref
     $  flatpak run org.opencpn.OpenCPN//beta

>  Remember that OpenCPN also said that the shared object file passed the "elf" test. So the file "existed" earlier in the process.

Indeed. IIRC one can get the same message if the library has some unresolved dependencies. ldd should be able to show if there is such a problem. However;

    $ flatpak run --command=bash --devel org.opencpn.OpenCPN
    error: runtime/org.freedesktop.Sdk/aarch64/18.08 not installed

You need to have the sdk (i. e., the System Development Kit) installed to debug. But before diving into this, update opencpn to the beta branch as of above. After doing this, install the correct SDK:

    $ flatpak install --user flathub org.freedesktop.Sdk//20.08 

You should then be able to run the shell and ldd command as I wrote [above](https://github.com/OpenCPN/OpenCPN/issues/2047#issuecomment-840037229). 

The ubuntu variant of ldd seems old and broken. But even if it was in a better shape, it wouldn't  be  useful when outside the sandbox

transmitterdan, 2021-05-13:
I was able to install the beta build and the Sdk from flathub.

However, when I run the beta build O will not install any plugins from catalog.  Log file attached.

[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/6472394/opencpn.log)

rgleason, 2021-05-13:
I believe this is a very old version of VDR that I did for bdbcat following oesenc CI, when you first got PIM implemented, and it badly needs an update using ShipDriver CMake/CI.  It also is not using getpPrivateApplicationDataLocation so when it is recording there is likely a problem!

ubuntu@ubuntu:~$ ldd ~/.var/app/org.opencpn.OpenCPN/lib/opencpn/libvdr_pi.so            
	not a dynamic executable

transmitterdan, 2021-05-14:
It is not only VDR.  All plugins in the catalog fail.  There is something simple going on I think.

leamas, 2021-05-14:
Sorry for late reply. Occasionally I enjoy the illusion of having a life...

In the present logs from the beta branch the typical snippet is

     07:11:10.376   DEBUG PluginHandler.cpp:174 Plugin: init: abi: flatpak-x86_64, abi_version: 18.08, major ver: 18
     07:11:10.376   DEBUG PluginHandler.cpp:198 Host: init: abi: flatpak-aarch64, abi_version: 20.08, major ver: 20
     07:11:10.376   DEBUG PluginHandler.cpp:343 Plugin compatibility check Final: REJECTED: VDR

This basically says that the host is flatpak-aarch64 on 20.08 whereas the plugin is flatpak-aarch64 on version 18.08. Although they are both flatpak-aarch64 the comparison fails since they have different versions.

That the plugins currently are built for 18.08 is in many cases just because changes in the latest master branches has yet not been published. This includes testplugin, shipdriver, oesenc and radar  which all have unpublished changes. For testplugin and shipdriver, it also affects those plugins which are using these templates.

For those using the shipdriver templates (shipdriver, radar and oesenc) it's possible to make a local build of the plugin to test. Clone the sources, look into INSTALL.md for build deps and how to build the plugin. It's not that complicated. The resulting tarball can be imported. 

The same for testplugin is possible, but more complicated and as fas as I know  not documented (?).

Without builiding the plugin, it's just to wait until they publish their changes. :(

bdbcat, 2021-05-14:
Hmm...
Here we go on the ubuntu release version merry-go-round again.
Having to track ubuntu releases (4 times per year) for flatpak plugins somewhat obviates the advantage of flatpak, does it not?

Is there such a thing as flatpak-aarch64/18.08?  And may we insist that this is the flatpak-$ARCH} version "officially" supported by OCPN?  Or should we step up to a later LTS, like 20.04?  Nothing special about 20.08, is there?

leamas, 2021-05-14:
The version 18.08 or 20.08 is the version of the flatpak runtime. Think of it as the operating system version. 20.08 is the LTS version, there is no such thing as 20.04.

Their idea is to release a supported version every two year, similar to Ubuntu LTS.  However, we could probably skip some versions, effectively having a four year cycle. There is **no** need to track changes four times a year.

OTOH, at some point we will need to update 18.08. Given this, it would IMHO be wrong to release a new platform on an already outdated OS. We should thus use 20.08 on aarch64

bdbcat, 2021-05-14:
OK, sensible.  Flatpak LTS is one "step" later than canonical ubuntu LTS.
So, we roll out the flatpak-aarch64 plugins as dev resources permit.
Thanks for the 'splaining.

transmitterdan, 2021-05-14:
Alec is right.  We just have to keep the main O build and plugin build on the same flatpak release.  Here is the OS info for the flatpak runtime O is using for 64-bit ARM.

<pre><font color="#26A269"><b>transmitterdan@ubuntu</b></font>:<font color="#12488B"><b>~/OpenCPN/build</b></font>$ cat /etc/os-release 
NAME=Freedesktop.org
VERSION=&quot;20.08.8 (Flatpak runtime)&quot;
VERSION_ID=20.08.8
ID=org.freedesktop.platform
PRETTY_NAME=&quot;Freedesktop.org 20.08.8 (Flatpak runtime)&quot;
BUG_REPORT_URL=https://gitlab.com/freedesktop-sdk/freedesktop-sdk/issues
<font color="#26A269"><b>transmitterdan@ubuntu</b></font>:<font color="#12488B"><b>~/OpenCPN/build</b></font>$ 
</pre>

This is independent of the underlying native OS.  We just have to build plugins for the same runtime O uses and all should be ok.  Not saying it is easy but it is manageable.

[31 more replies]

closed 2021-05-31 by hreuver0183