OpenCPN/OpenCPN #2049: MacOS builder: multiple problems.

leamas, 2020-10-01:
The MacOS circleci builder is too slow and also needs some general love.

  - It's too slow, build times around 25 minutes.
  - It uses the Xcode 10.3 level, which depends on python2 which is about to be removed from homebrew (upstream EOL 2021-01-01).
  - It does not declare dependencies to libraries used in the package
  - As a consequence, there are "truly awful" hacks to handle some of these libs.
  - The bundle permissions problem is another hack instead of the correct usage of BU_CHMOD_BUNDLE_ITEMS

.

leamas, 2020-10-05:
Not really about mac, buit still about ci: We need to authenticate our docker pulls: https://circleci.com/docs/2.0/private-images from November, 1.

leamas, 2020-10-05:
The  circleći email: 

Hi there,

On November 1st, Docker Hub will begin limiting anonymous image pulls. We want to make sure you know how you might be impacted and what you can do to avoid interruptions to your workflow.

Adding Docker authentication to your pipeline config is the easiest way to avoid any service disruptions. If you use the Docker executor or pull Docker images when using the machine executor on CircleCI, we encourage you to authenticate. Because the anonymous API rate limits are based on IP addresses, they will impact CircleCI cloud customers. Authenticated users get higher per-user rate limits, regardless of IP.

We are currently working on a partnership with Docker to minimize the impact of this change for our users and will share more details as we get them.

For more information or to leave a question for us, please head over to Discuss or contact support.

Thanks and happy building,

- The CircleCI team

bdbcat, 2020-10-06:
@leamas 
"We need to authenticate our docker pulls:"
And how do we do this?  I think we are only using private docker images for some rPI builds.  I recall building/pushing those two images.  The rest are public, aren't they?

leamas, 2020-10-06:
The info is in the link: : https://circleci.com/docs/2.0/private-images

Basically:
    - Set up a docker account (normal free registration).
    - Add the password to the environment, similar to CLOUDSMITH_API_KEY
    - Add username and name of environment variable to .circleci/config

As I said, the details are in the link. As I understand it, docker.io will limit all anonymous downloads, be the images public or not.

rgleason, 2020-10-06:
Is this going to affect plugins building MacOS?

leamas, 2020-10-06:
> Is this going to affect plugins building MacOS?

Yes

> Will I need an opensource docker account?

Yes, at least an account

rgleason, 2020-10-16:
A "Docker Hub" account?   (cloud type?) https://hub.docker.com/

PR #2050 (2020-12-09): Macos builder: Performance , bugfixes.
Basically fixes for problems described in #2049 

XCode is updated to 11.0, modern enough to avoid the python2 dep (but still not that modern that it exposes CPack bugs when using for example 11.3).

The dependencies are cached, and the build now completes in about 15 minutes instead of 25. Still not "fast", but better. The cache ins invalidated when ci/generic-build-macos.sh is changed.

Libraries used are properly linked to the executable so that CPack can pick them up.

Remove various hacks for library inclusion and chmod.

EDIT: Drop the travis macos  [... cut]
closed 2020-12-09 by bdbcat