OpenCPN/OpenCPN #2051: ARM build failed in pluginmanager

Hakansv, 2020-10-03:
Today's master update works fine on Win.
One build error on my RPi3 Stretch though:

[ 41%] Building CXX object CMakeFiles/opencpn.dir/src/PluginHandler.cpp.o
/home/pi/builds/OpenCPN/src/PluginHandler.cpp: In constructor ‘CompatOs::CompatOs()’:
/home/pi/builds/OpenCPN/src/PluginHandler.cpp:189:8: error: ‘os_detail’ was not declared in this scope
     if(os_detail->osd_ID.size())
        ^~~~~~~~~
/home/pi/builds/OpenCPN/src/PluginHandler.cpp:191:8: error: ‘os_detail’ was not declared in this scope
     if(os_detail->osd_version.size())
        ^~~~~~~~~
CMakeFiles/opencpn.dir/build.make:1817: recipe for target 'CMakeFiles/opencpn.dir/src/PluginHandler.cpp.o' failed
make[2]: *** [CMakeFiles/opencpn.dir/src/PluginHandler.cpp.o] Error 1
CMakeFiles/Makefile2:267: recipe for target 'CMakeFiles/opencpn.dir/all' failed
make[1]: *** [CMakeFiles/opencpn.dir/all] Error 2
Makefile:152: recipe for target 'all' failed
make: *** [all] Error 2

hreuver0183, 2020-10-03:
I was also strugling at that part.

Maybe my PR also works for you?
Fix broken compile for arm64 (check on plugin arch) #2052

Or maybe you can think of a better solution?

Hakansv, 2020-10-03:
Solved by #2052
Thanks

closed 2020-10-03 by Hakansv