OpenCPN/OpenCPN #2053: Compilation error with main branch

mgrouch, 2020-10-06:
/home/pi/OCPN/opencpn/src/glChartCanvas.cpp: In member function ‘void glChartCanvas::RenderMBTilesOverlay(ViewPort&)’:
/home/pi/OCPN/opencpn/src/glChartCanvas.cpp:4951:78: warning: overflow in conversion from ‘double’ to ‘long unsigned int’ changes value from ‘5.0e+9’ to ‘4294967295’ [-Woverflow]
             if(!ChartData->CheckAnyCanvasExclusiveTileGroup() || (tileSize > 5e9)){
                                                                              ^~~
make[2]: *** [CMakeFiles/opencpn.dir/build.make:1613: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o] Error 1

leamas, 2020-10-06:
hm... I  cannot reproduce using:
  - Current master @ a44756712e
  - Steps: `rm -rf build; mkdir build; cd build; cmake ..;  make `
  - gcc version: 

          $ gcc --version
          gcc (GCC) 10.2.1 20200723 (Red Hat 10.2.1-1)
  - OS: Fedora 32.

So, what's the difference?

mgrouch, 2020-10-06:
On Raspberry pi 4 with 64 bit kernel

build wxWidgets 3.1.4 like this:

sudo apt-get -y install build-essential cmake gettext git-core libgtk-3-dev libgtk-3-0

git clone -b v3.1.4 --recurse-submodules https://github.com/wxWidgets/wxWidgets.git
cd wxWidgets || exit 255

mkdir build-gtk
cd build-gtk || exit 255

../configure --with-gtk=3 --with-libpng=builtin --with-libjpeg=builtin --with-libtiff=builtin --with-regex=builtin \
 --with-zlib=builtin --with-expat=builtin --enable-unicode --enable-aui --with-opengl \
 LDFLAGS="-static-libgcc -static-libstdc++"
make -sj5
sudo make install
sudo ldconfig

then build OpenCPN against it:

# Build wxWidgets before
# See ../wxWidgets-rpi/

sudo apt-get -y install build-essential cmake gettext git-core gpsd gpsd-clients libgps-dev \
 libglu1-mesa-dev libgtk-3-dev libbz2-dev libtinyxml-dev libportaudio2 portaudio19-dev libcurl4-openssl-dev \
 libexpat1-dev libcairo2-dev libarchive-dev liblzma-dev libexif-dev libelf-dev libsqlite3-dev \
 bc bison flex libssl-dev python3 ddd htop

git clone https://github.com/opencpn/opencpn.git

cd opencpn && mkdir build
cd build || exit 255

cmake -DOCPN_BUNDLE_TCDATA=ON -DOCPN_BUNDLE_GSHHS=ON -DOCPN_USE_GL=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo \
 -DOCPN_USE_CRASHREPORT=ON -DOCPN_ENABLE_PORTAUDIO=OFF -DOCPN_ENABLE_SYSTEM_CMD_SOUND=ON -DOCPN_FORCE_GTK3=ON ..

make -sj5

sudo make install

mgrouch, 2020-10-06:
```
pi@raspberrypi:~/OCPN/opencpn/build $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
pi@raspberrypi:~/OCPN/opencpn/build $ gcc --version
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0

```

bdbcat, 2020-10-07:
Should be corrected by 5079214fe12424ac0ebed0a596cccb6acba54ace

closed 2020-10-07 by mgrouch